RIOT-OS/RIOT #10364: kw2xrf: Implement CSMA in driver

gebart, 2018-11-12:
_Originally posted by @PeterKietzmann in https://github.com/RIOT-OS/RIOT/pull/10355_

>Well, we're all short in time but we (HAW Hamburg) have certain interest in that radio so it might be us to do the follow-up. @gebart, @bergzand could you please open a feature request issue?

The kw2xrf driver does not support NETOPT_CSMA, but it does support NETOPT_AUTOCCA instead (and is currently the only driver which supports that flag)

The kw41z radio also does not support CSMA in hardware but only automatic CCA before TX (i.e. no automatic backoff and retry). The kw41zrf driver (#7107) implements CSMA exponential backoff and retry in software but utilizing the autocca hardware support. The same method could be used in the kw2xrf driver.

Whether the implementations can be shared between the two drivers or if it is better to reimplement the same behavior is left as an exercise to the reader.

gebart, 2018-11-12:
Some pointers towards the kw41zrf implementation:

TX initialization:
https://github.com/gebart/RIOT/blob/pr/kw41zrf/drivers/kw41zrf/kw41zrf_netdev.c#L100-L174

CCA feedback using the CCAIRQ flag:
https://github.com/gebart/RIOT/blob/pr/kw41zrf/drivers/kw41zrf/kw41zrf_netdev.c#L961-L1000

Ilmu011, 2019-08-26:
I am working on this right now and there is one thing I don't understand;
https://github.com/gebart/RIOT/blob/pr/kw41zrf/drivers/kw41zrf/kw41zrf_netdev.c#L47-L51

```
#define KW41ZRF_CCA_TIME               8	//128 us  Ref Manual Page 2146
#define KW41ZRF_SHR_PHY_TIME          12	//192 us
#define KW41ZRF_PER_BYTE_TIME          2	//32 us
#define KW41ZRF_ACK_WAIT_TIME         54	//864 us
#define KW41ZRF_CSMA_UNIT_TIME        20	//320 us 
```

Where are these timing parameters coming from? I have perused the MKW41Z/31Z/21Z Reference Manual but were not able to find them except `KW41ZRF_CCA_TIME`, which is explicitly stated as 128 us. 
https://www.nxp.com/files-static/32bit/doc/ref_manual/MKW41Z512RM.pdf#page=2146

Furthermore, why are these timing parameters even a thing? These parameters are currently used to start a timer, which is supposed to indicate whether the acknowledge frame in a TR-sequence has timed out. The timer is set BEFORE issuing the operation to hardware, using these timing parameters to estimate, when the TX operation will be complete and adding `KW41ZRF_ACK_WAIT_TIME` on top of that.
https://github.com/gebart/RIOT/blob/pr/kw41zrf/drivers/kw41zrf/kw41zrf_netdev.c#L125-L132

Why isn't the timer just set to `KW41ZRF_ACK_WAIT_TIME` AFTER the hardware has signaled us, that the TX operation is complete?

PeterKietzmann, 2019-08-26:
> Where are these timing parameters coming from?

These parameters result from the 802.15.4 specification. For example: 802.15.4 has a data rate of 250kbit/s. How long does it take to transmit one bit? 1 bit / 250kbit/s = 4 us and (1bit x 8) / 250kbps/s = 32us for one byte respectively. How does this relate to `#define KW41ZRF_PER_BYTE_TIME          2	//32 us`? I **assmue** the timer that is used within that driver runs at 62.5 kHz. Then, one tick takes16us and two ticks take 32us as mentioned next to the define.

PeterKietzmann, 2019-08-26:
> Furthermore, why are these timing parameters even a thing?

I don't understand. But in any way, please note that the driver that you are referring to is WIP and not part of the official RIOT master. 

> Why isn't the timer just set to KW41ZRF_ACK_WAIT_TIME AFTER the hardware has signaled us, that the TX operation is complete?

I'm not familiar with the implementation in that detail. Your concerns seem reasonable to me. Maybe @benemorius can shed light on the as he indicated to take over #7107.

Ilmu011, 2019-08-26:
> I **assmue** the timer that is used within that driver runs at 62.5 kHz. Then, one tick takes16us and two ticks take 32us as mentioned next to the define.

That's how it is. I checked it before.

https://github.com/gebart/RIOT/blob/pr/kw41zrf/drivers/kw41zrf/kw41zrf.c#L177

Ilmu011, 2019-09-03:
Another thing; kw2xrf supports `NETOPT_AUTOCCA` and kw41zrf supports `NETOPT_CSMA`, despite both of them doing the exact same thing. Both use automatic CCA before TX operations, and in both cases that's exactly what the respective NETOPTs in both drivers enable. The only difference is, that kw41zrf additionally does exponential backoff.

CSMA emulation via usage of automatic CCA is exactly what [NETOPT_AUTOCCA](https://riot-os.org/api/group__net__netopt.html#gga19e30424c1ab107c9c84dc0cb29d9906a3c84d6406a0e0c6929e15db79b3dfc8b) defines and should be supported here.

> Perform channel clear assessment before transmitting
> 
> This may be a hardware feature of the given transceiver, or might be otherwise implemented in software. If the device supports CSMA this option will enable CSMA with a certain set of parameters to emulate the desired behaviour.

Actually, according to the documentation, it is even WRONG to support [NETOPT_CSMA](https://riot-os.org/api/group__net__netopt.html#gga19e30424c1ab107c9c84dc0cb29d9906a566a086347f022dbd8962485bd245a10) here;

> If the device supports CSMA in hardware, this option enables it with default parameters. For further configuration refer to the other NETOPT_CSMA_* options.

Neither kw2xrf nor kw41zrf support CSMA in hardware. None should support `NETOPT_CSMA`, but should support `NETOPT_AUTOCCA` instead.

The advantage of supporting `NETOPT_CSMA` is that the driver can make use of the additional options for `NETOPT_CSMA`, like `NETOPT_CSMA_RETRIES`, `NETOPT_CSMA_MAXBE` or `NETOPT_CSMA_MINBE`, but from a specification standpoint, it's just wrong.

Ilmu011, 2019-09-03:
For efficiency's sake, for now I'll just throw out `NETOPT_AUTOCCA` from kw2xrf and implement support for NETOPT_CSMA, similar to kw41zrf, but something should be done about this in the future.

benemorius, 2019-09-12:
I was puzzled at first by the timer usage too. These radios have a tighter integration with software than others I've used. The use of the radio timer isn't just to fire callbacks in software. It's actually used to trigger events in the radio hardware like timing out an ACK wait for example.

In other words:
> Why isn't the timer just set to `KW41ZRF_ACK_WAIT_TIME` AFTER the hardware has signaled us, that the TX operation is complete?

Because the hardware waits to signal us the outcome of the TX operation until it has received an ACK or not, and the timer is what causes it to stop waiting for a missed ACK and signal us. At least that's the case with kw41zrf. It looks like kw2xrf does as you say where the hardware signals TX complete before waiting for ACK and only then the driver starts an ACK timer. It's possible that both radios could support either paradigm and the difference in the two drivers is just developers' preference.

My interpretation of NETOPT_AUTOCCA is that the clear channel assessment should be performed but NOT the retries. Meaning if CCA succeeds then the packet is transmitted but if it fails then the radio driver returns a medium-busy failure rather than retrying.

I interpret NETOPT_CSMA to mean NETOPT_AUTOCCA plus retries. And you're right the documentation specifically says that it must be hardware CSMA support which is interesting. Is that because we have (or intend to have (or used to indend to have)) software CSMA retries implemented in an upper layer? And maybe it's getting implemented in netdev drivers because the upper layer implementation is missing so far? I haven't looked in to that.

benemorius, 2019-09-13:
By the way there's a bug in the #7107 CSMA implementation that prevents any retransmissions from actually happening if the initial CCA fails.

After `_isr_event_seq_t_ccairq()` calls `kw41zrf_tx_exec()` to retransmit the last frame, `_isr_event_seq_tr()` aborts the sequencer prematurely by calling `kw41zrf_set_sequence(dev, dev->idle_seq)` while handling `ZLL_IRQSTS_SEQIRQ_MASK`.

But with that fixed I think it works as intended.

Ilmu011, 2019-10-27:
@PeterKietzmann 
I opened #12589 regarding this issue.

miri64, 2020-07-03:
Should also be fixed by @jia200x lower-layer rework (see #13943)

PeterKietzmann, 2020-07-06:
The kinetis hardware provides dedicated timers for CSMA. Although a SubMAC might solve the issue already using AutoCCA, a CSMA driver is missing which might reduce load.

Stopkaa, 2026-09-17:
@jia200x this issue can be closed; submac takes care of csma-ca

miri64, 2026-09-17:
> The kinetis hardware provides dedicated timers for CSMA. Although a SubMAC might solve the issue already using AutoCCA, a CSMA driver is missing which might reduce load.

@Stopkaa I think this point still stands.

jia200x, 2026-09-17:
Note that such timers are just "regular" 16-us-resolution timers, which are intended to be used for MAC duties (CSMA-CA, ACK management, etc.).

Given that, I don't think implementing CSMA-CA from scratch in the driver will make it more efficient than using the current SubMAC, as the CSMA-CA code remains the same.

A different story occurs with the CC2538 (where we manually implement CSMA/CA using the strobe processor), but there the device acts as a co-processor. In such a case, the CPU is offloaded.

We could indeed leverage such radio timers for the SubMAC, though, which currently uses the high-level API (ztimer), which has some overhead at this level. Also, counting in multiples of 16 us is handy, as it's the O-QPSK symbol duration, which converts the whole thing into just counting symbols.

Therefore, I propose to focus the work on adding more HW acceleration to the current SubMAC rather than re-implementing features in the driver.