RIOT-OS/RIOT #10365: Use after free in gnrc_sock_udp sock_udp_recv

crest42, 2018-11-12:
#### 
Sometimes the sock_udp_recv function in the implementation provided by gnrc_sock_udp returns unexpected values (-12736 or -1 for example). This is probably due to a use after free in this functions return statement. I will provide a pull request for more details.

Affected Code:
-----------------------------
    gnrc_pktbuf_release(pkt);
    return (int)pkt->size;
-----------------------------

#### Steps to reproduce the issue
Read from a udp socket using gnrc_sock_udp implementation with sock_udp_recv and check if the return values is < 0 and none of the expected errors. Only happens when a certain amount of packets is send to the reading host. I wasn't able to reproduce with running valgrind yet. 

I use native arch with a tap network setup as described in https://github.com/RIOT-OS/RIOT/wiki/Virtual-riot-network

#### Expected results
sock_udp_recv should either return a value >= 0 or any of -EADDRNOTAVAIL, -ENOMEM, -EAGAIN, ,EINVAL, -ENOBUFS, -ENOMEM, -EPROTO or -ETIMEDOUT

#### Actual results
Somethimes the function returns random values < 0 or values > 0 which does not match the actual packet size.

#### Versions
Operating System Environment
-----------------------------
       Operating System: "Arch Linux" 
                 Kernel: Linux 4.18.16-arch1-1-ARCH x86_64 unknown

Installed compiler toolchains
-----------------------------
             native gcc: gcc (GCC) 8.2.1 20180831
      arm-none-eabi-gcc: arm-none-eabi-gcc (Arch Repository) 8.2.0
                avr-gcc: avr-gcc (GCC) 8.2.0
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
                  clang: clang version 7.0.0 (tags/RELEASE_700/final)

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "3.0.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
               avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                  cmake: cmake version 3.12.4
               cppcheck: missing
                doxygen: 1.8.14
                 flake8: missing
                    git: git version 2.19.1
                openocd: Open On-Chip Debugger 0.10.0
                 python: Python 3.7.1
                python2: Python 2.7.15
                python3: Python 3.7.1
             coccinelle: missing

miri64, 2018-11-12:
Confirmed. 

> I will provide a pull request for more details.

Thanks! It would be great if you would fix `gnrc_sock_ip` as well then:

https://github.com/RIOT-OS/RIOT/blob/98f42eeea8e04b0e2109aa77e04d51f566f60bee/sys/net/gnrc/sock/ip/gnrc_sock_ip.c#L123-L124

crest42, 2018-11-12:
Thanks for the fast answer. I referenced the relevant PR.

PR #10366 (2018-11-12): gnrc_sock_[udp|ip]: read return value for _recv after release

### Contribution description

Sometimes the sock_udp_recv function in the implementation provided by gnrc_sock_udp returns unexpected values (-12736 or -1 for example). This is probably due to a use after free in this functions return statement. This PR fixes the bug for me.

### Testing procedure
"ALL TESTS SUCCESSFUL" for tests/gnrc_sock_ip and tests/gnrc_sock_udp as well as my application which triggered the error before.

### Issues/PRs references
Fixes #10365
PR #10374 (2018-11-12): gnrc_sock_[udp|ip]: read return value for _recv after release [backport 2018.10]
# Backport of #10366

### Contribution description

Sometimes the sock_udp_recv function in the implementation provided by gnrc_sock_udp returns unexpected values (-12736 or -1 for example). This is probably due to a use after free in this functions return statement. This PR fixes the bug for me.

### Testing procedure
"ALL TESTS SUCCESSFUL" for tests/gnrc_sock_ip and tests/gnrc_sock_udp as well as my application which triggered the error before.

### Issues/PRs references
Fixes #10365
closed 2018-11-12 by miri64