RIOT-OS/RIOT #10370: gomach: Resetting netif with cli doesn't return

bergzand, 2018-11-12:
#### Description

A reset of the network interface via `ifconfig 7 set state reset` never returns.

#### Steps to reproduce the issue

Flash the `examples/gnrc_networking_mac` on a samr21-xpro
send `ifconfig 7 set state reset`.

#### Expected results

The network interface is reset

#### Actual results

The node stops functioning

#### Versions

```
Installed compiler toolchains
-----------------------------
             native gcc: gcc (Gentoo 6.4.0-r1 p1.3) 6.4.0
      arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]              
                avr-gcc: avr-gcc (Gentoo 4.9.4 p1.0, pie-0.6.4) 4.9.4
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
                  clang: clang version 6.0.1 (tags/RELEASE_601/final)

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "3.0.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
               avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                  cmake: cmake version 3.9.6
               cppcheck: Cppcheck 1.81
                doxygen: 1.8.14
                 flake8: 3.5.0 (mccabe: 0.6.1, pycodestyle: 2.3.1, pyflakes: 1.6.0) CPython 3.6.5 on Linux                                                                        
                    git: git version 2.18.1
                openocd: Open On-Chip Debugger 0.10.0
                 python: Python 3.6.5
                python2: Python 2.7.15
                python3: Python 3.6.5
             coccinelle: missing
```

miri64, 2020-07-03:
Might be related to #9656. We should see if this is fixed, after #9656 is fixed.

maribu, 2025-12-23:
This issue has been fixed in https://github.com/RIOT-OS/RIOT/pull/21600