RIOT-OS/RIOT #10373: drivers/cc110x: Rewrite of the driver for CC110X transceivers

maribu, 2018-11-12:
#### Description

The current `cc110x` has bit-rotten a bit and lacks some documentation and features. Also the driver was written in a pre-GNRC era, so GNRC support was added on top of it as a separate layer, which adds some overhead. Those issues are best addressed with a rewrite of the driver with the GNRC in mind.

#### Roadmap

- [ ] Rewrite of the base driver https://github.com/RIOT-OS/RIOT/pull/10340
- [ ] Implementation of CCA before transmission as pseudo-module `cc110x_cca`
    - depends on https://github.com/RIOT-OS/RIOT/pull/10340
    - A working proof of concept is here: https://github.com/maribu/RIOT/tree/cc110x_cca
    - Is implemented as pseudo-module, as default ALOHA used less ROM and is fine for many scenarios
- [ ] Access to the temperature sensor of the transceiver via SAUL as pseudo-module `cc110x_temperature`
    - depends on https://github.com/RIOT-OS/RIOT/pull/10340
    - will only work if GDO0 is connected to pin that can be used as analog input
- [ ] Debug utility as RIOT shell command as pseudo-module `cc110x_shell_command`
    - depends on https://github.com/RIOT-OS/RIOT/pull/10340
    - During development I used a small shell command to access the low level details of the driver. This was extremely useful for debugging, so I want to polish it a bit and create a PR to make future debugging easier
- [ ] Allow setting default TX power via `cc110x_params_t`
    - depends on https://github.com/RIOT-OS/RIOT/pull/10340
- [ ] Add pesudo-module `cc110x_autocalibration` to perform automatic re-calibrations of the frequency generator
    - depends on https://github.com/RIOT-OS/RIOT/pull/10340
    - In some scenarios automatic re-calibration of the frequency generator is required, e.g. when temperature is not constant (e.g. outdoor) or the supply voltage changes over time (e.g. when directly connected to a battery)
    - This module will perform re-calibration in configurable intervals, e.g. every 10 minutes

PeterKietzmann, 2019-11-12:
#10340 got merged a while ago. @maribu I'm gonna close the issue now. In case of leftovers, I suggest to either open a new and more precise feature request issue or reopen this one with an updated description

maribu, 2019-11-12:
> reopen this one with an updated description

done