RIOT-OS/RIOT #10375: New board suggestion: stm32WB

andrea-nisti, 2018-11-12:
## Description
I would like to use the stm32WB MCUs series but I haven't found the compatibility. How hard could be to port this series? I think that this is a pretty good architecture for IOT applications having the embedded connectivity tools.

### Useful links
[STM32WB](https://www.st.com/en/microcontrollers/stm32-wireless-mcus.html?querycriteria=productId=SC2156)

Thanks!

MrKevinWeiss, 2019-01-03:
@andrea-nisti It looks like an interesting chip.
I am not sure how to answer that question since I could not find the reference manual or any way to purchase the chip or module.  It also appears that it uses both an M4 and M0 which may take some effort to port.

If ST releases the reference manual or if you are able to find it, please post a link and I or someone can spend some time looking through it to see how many common block it has with other ST boards.

andrea-nisti, 2019-01-07:
@MrKevinWeiss I discovered that this chip is still under preview, not really sure when available for us mortals. Anyway, I really think this could be an awesome solution for embedded IOT applications and as soon as possible I will play with it. 

Through the STM MCU Finder I downloaded the datasheet, you can find it at [this](https://1drv.ms/b/s!Avq0vFmfOUSQwyzRsq1jTGWe4vcw) link

MrKevinWeiss, 2019-01-07:
Ya, it is very fresh but very cool.  Thanks for the datasheet but I think the reference manual is what I would need.  Everyone has their own naming conventions.  What I would be interested in would be the register map, then I can compare which register functions match with what we already have.

I guess we just have to patient until it is released!

andrea-nisti, 2019-01-07:
Oops, my bad. Yes apparently ref manual it is still not available, we will wait. Moreover, I am going to visit ST in Milan pretty soon as I have a meeting with a business manager (i think) there. I will certainly ask when I could lay my hands on that chip and keep you updated.

andrea-nisti, 2019-01-08:
For the curious people around, ST is organizing a workshop exclusively on STM32WB before its launch on the market. Check [this](https://www.st.com/content/st_com/en/about/events/events.html/stm32wb-hands-on-workshop-2018.html) link for reservations and dates, it's free!
I am going to this workshop in Milan on Feb 13th but there are many dates and locations around europe

Sending a couple of RIOT members could be a good idea, I'm still too noob in this field ;)

MrKevinWeiss, 2019-08-12:
I think we can close it.  I have a board and maybe the datasheet is out but I have no time in the near future to work on it and seeing as it is two cpus (I think) it may be quite difficult.

closed 2019-08-12 by MrKevinWeiss