RIOT-OS/RIOT #10388: gnrc_sock_udp: Possible Race condition on buffer cpy

crest42, 2018-11-14:
#### Description
When using gnrc_sock_udp in my application after a certain amount of hosts (and thus a certain amount of packets/s on a single host), the buffer content returned by sock_udp_recv is filled with garbage. My message format is defined to be 4 bytes + 4 bytes + 4 bytes + 4 bytes (message type, app src id, app dst id, length of the remainder data).

As an example: A message of type 11 always sends a static sized buffer with size 32 and thus the message "header" would look like this: 0x0000000b 0x00000001 0x00000002 0x00000020. (Second and third 4 bytes can differ)

Sometimes if many hosts are joining the network it happens very seldom that a single host instead read something like: 0x0000000b 0xfe80aaf0 0x00000000 0x00afbb20

Two intresting obeservations here:

1. The first 4 bytes always are correct. 
2. The second 4 bytes always contains fe80something (Maybe i am just paranoid but this looks like the beginning of a link local ipv6 address)

#### Steps to reproduce the issue
Reading content from a udp socket and sending a lots (TBD) of messages. Testing the read buffer for correctness

#### Expected results

#### Actual results

#### Versions

crest42, 2018-11-14:
Woops hit enter accidentally.

closed 2018-11-14 by crest42

miri64, 2018-11-19:
Taken up again in #10389. @crest42 you can edit issues after open ;-)