RIOT-OS/RIOT #10395: stm32f103xx: I2C has no internal pullups and af config

MrKevinWeiss, 2018-11-15:
#### Description
When using the nucleo-f103rb the standard configuration is i2c0 on PB8 and PB9 the AFIO periph clock must be enabled 
`RCC->APB2ENR |= 0x00000001;` 
and the pin must be remapped to those pins 
`AFIO->MAPR |= AFIO_MAPR_I2C1_REMAP;`
Also due to the hardware constraints it seems like the internal pullup cannot be enabled since the pins must be set to `GPIO_AF_OUT_OD`.

In the [reference manual](https://www.st.com/content/ccc/resource/technical/document/reference_manual/59/b9/ba/7f/11/af/43/d5/CD00171190.pdf/files/CD00171190.pdf/jcr:content/translations/en.CD00171190.pdf) it seems like it is impossible to do so (in section 9.1.9 Alternate function configuration it states that 'The weak pull-up and pull-down resistors are disabled.')

I am not sure if anything can be done about requiring external pullups.

#### Steps to reproduce the issue
Try to use the i2c0 on any nucleo103xx on PB8 and PB9.

#### Expected results
It works.

#### Actual results
It doesn't work and will hang

MrKevinWeiss, 2018-11-15:
@aabadie, @haukepetersen, or @dylad, I would like your input on how to cleanly add an remap config and if you know any way to try to activate the pullup.

maribu, 2018-11-28:
The same issue affects the blue pill (and clones). Using [PB6 and PB7 instead](https://github.com/RIOT-OS/RIOT/pull/10466) is my proposed solution in that case, because than no remapping of the first I2C interface is required. Would that be an option for the Nucleo as well?

MrKevinWeiss, 2018-11-28:
Ya but then you would have to use some morpho pins _(I think)_ and not the nice labeled/standard sda and scl.

When I find I can try to figure something out, I just didn't want it forgotten.

PR #10607 (2018-12-14): cpu/stm32f1: Fix i2c for stm32f1 based boards
### Contribution description
This was discovered by the HiL testing.

This PR adds remapping if PB8 or PB9 are being used for i2c.  Previously the remapping didn't occur and would cause errors when trying to use this configuration.  This should only effect stm32F1XX based boards.  It also updated documentation to indicate pullup resistors are needed.  With this something like the nucleo-f103rb should now work.

### Testing procedure
Try to use the i2c on a nucleo-f103rb in master and with this PR.  It should at least respond (though the cpu impleme [... cut]
closed 2018-12-14 by aabadie