RIOT-OS/RIOT #10399: core: merge `msg` into `mbox`

miri64, 2018-11-15:
From my understanding there is some shared functionality between the asynchronous behavior of `core_msg` and `core_mbox`: messages are put into a queue by *n* threads and gotten out *n* threads. In case of `core_msg` *n* is however one.

On one hand side we might get some code size benefits in merging the two, on the other we would put some more complexity into the asynchronous behavior of `core_msg`, since it isn't required to handle more than one reception thread.

A good compromise in my opinion would be to provide an new `mbox` implementation that assumes only one reader and base the old `mbox` and `msg` implementation on it. I started something into that direction in [this branch](https://github.com/miri64/RIOT/tree/core/enh/merge-msg-into-mbox). But I

1. did not port `msg` yet 
2. am not sure if this https://github.com/RIOT-OS/RIOT/blob/1631d2961a56fa1066da2a4966922a86291323d0/core/mbox_base.c#L63-L66 and that https://github.com/RIOT-OS/RIOT/blob/1631d2961a56fa1066da2a4966922a86291323d0/core/mbox_base.c#L92-L96 is still possible, as both `_mbox_base_put()` and `mbox_base_get()` would be called by the wrapper implementations of `mbox` and `msg`, both might have their own `irqstate` stored (and thus it might not properly return to the right IRQ state in both `_mbox_base_wait()` and `_mbox_base_notify()`.

kaspar030, 2018-11-15:
I had an msg implementation that used mbox below. I didn't PR it because plain msg was simply faster. Whatever you come up with, try bench_msg() on real hardware.

miri64, 2018-11-15:
@kaspar030 any thoughts on 2.?

miri64, 2018-12-18:
Ping @kaspar030?

closed 2019-09-10 by stale[bot]