RIOT-OS/RIOT #10419: gnrc_icmpv6_error: Able to bounce up to 64 ICMPv6 error messages between 2 instances

bergzand, 2018-11-16:
#### Description

It is possible to cause a node to transmit up to 64 ICMPv6 destination unreachable messages. Other ICMPv6 error messages are probably also possible.

This took a bit of time to reproduce after discovering it.

#### Steps to reproduce the issue

I was able to trigger this when using two native instances in combination with a Linux host using scapy.
 1. Start 2 native instances.
 2. Assign an global address to the first node, e.g. `fd02::5/64`
 3. wait a bit for the router solicitations to stabilize.
 4. send a packet with scapy to the second instance (the one that got the address from the router advertisement of the first instance):
```
sendp( Ether(dst=DST_HWADDR) / IPv6(src="fd02::1", dst=DST_ADDR) / UDP(), iface="tapbr0")
```

With
`DST_HWADDR` = the link layer address of the second instance.
`DST_ADDR` = The globally unique IPv6 address of the second instance (`fd02::something`)
`fd02::1` = An **"non-existing"** address on the subnet.

#### Expected results

Nothing is send, as neighbour discovery will fail.

#### Actual results

64 destination unreachable messages are bounced between two instances, each with the all zeros IPv6 address as destination (`::`). I suspect that the messages are bounced between the two hosts as the mac addresses are switched between each frame.

See attached pcapng for capture.

[multi_icmpv6_error.zip](https://github.com/RIOT-OS/RIOT/files/2590702/multi_icmpv6_error.zip)

@miri64 Please let me know if you're able to reproduce, it took me quite some time before I was able to trigger it on purpose.

bergzand, 2018-11-16:
I suspect that the issue here has something to do with the instances responding to the unspecified IPv6 address (`::`), I'm also sometimes able to trigger the issue by crafting one of the ICMPv6 error packets with HW address source and destination set to the link layer addresses of the two instances and the IPv6 src address to the matching source IPv6 address and `dst` to `"::"`.

bergzand, 2018-11-16:
It could be that this issue is caused by two separate issues
 - The instance responding to the unspecified address `"::"`
 - The instance responding with the unspecified address as destination address when neighbour discovery fails.

Not completely certain that these two issues are the actual problem here though.

miri64, 2018-11-17:
Yepp, I can reproduce. The problem is two-fold:

1) Why send to `::` at all? Any IPv6 packet entering the send operation of the IPv6 module should be silently dropped. However, this is just a symptomal fix for this issue, but still a valid one
2) What actually is happening is that the system you are sending the packet too first is trying to send a "port unreachable" packet to `fd02::1` causing such packet to be generated that has the unspecified address set as source. *That* is thrown into the neighbor discovery. Before #8594 the packet was just silently dropped when that failed, now it tries sends an `address unreachable` message to signal that. Problem: since we need an interface (which we only get through neighbor discovery) to determine the source address, we basically get into a cyclic dependency here, so the source address is still unspecified at this point. Solution: don't send error messages to (and don't even build error messages for) IPv6 addresses originating from unspecified (in 99% of the cases this is the local node anyway that is generating those and for the other 1% it still doesn't make sense and isn't in fact allowed to reply to unspecified).

I have some local fixes for both, but I won't push them until I have looked over them with less sleepy eyes ;-) (so tomorrow at the earliest ;-))

miri64, 2018-11-17:
(if anyone is wondering like me, why it's exactly 64 error messages: it's because the packet is actually forwarded between them (because the node thinks the unspecified address is the default route) and thus decreasing the hop limit with every bounce).

PR #10421 (2018-11-17): gnrc_icmpv6_error / gnrc_ipv6: fixes for unspecified address
### Contribution description
This fixes the two bugs regarding the unspecified address I described in https://github.com/RIOT-OS/RIOT/issues/10419#issuecomment-439573538

1. Don't build and send ICMPv6 error messages destined to the unspecified address (::).
2. Don't send (and try to resolve) IPv6 messages addressed to the unspecified address.

### Testing procedure
Repeat the steps to reproduce in https://github.com/RIOT-OS/RIOT/issues/10419. The expected results (except for probing neighbor solicitations which are legitimate) shoul [... cut]
closed 2018-11-17 by bergzand