RIOT-OS/RIOT #10436: gnrc_netapi_set: argument void *data should be const

gebart, 2018-11-20:
#### Description

The public API for the gnrc_netapi_set function require a non-const `void *data` argument.

This API prohibits constructs such as:
```c
        static const netopt_state_t state_sleep = NETOPT_STATE_SLEEP;
        int res = gnrc_netapi_set(netif->pid, NETOPT_STATE, 0, &state_sleep, sizeof(state_sleep))
        if (res < 0) {
            LOG_ERROR("gnrc_netapi_set(%d, NETOPT_STATE_SLEEP) failed: %d\n", netif->pid, res);
        }
```
(`state_sleep` must be made not const for the code to compile)

### Useful links

gnrc_netapi_set implementation:

https://github.com/RIOT-OS/RIOT/blob/e0e02c04c8a7a65d41fd5ac605f01c6f75e3604a/sys/net/gnrc/netapi/gnrc_netapi.c#L165-L170

declaration:

https://github.com/RIOT-OS/RIOT/blob/e0e02c04c8a7a65d41fd5ac605f01c6f75e3604a/sys/include/net/gnrc/netapi.h#L191-L207

miri64, 2018-11-20:
Since this uses the same struct as `gnrc_netapi_get()` for the message communication, this might be difficult.

miri64, 2018-11-20:
I mean, we can make the the parameter `const` and then cast, but then I'm not sure how valuable the change is.

miri64, 2018-11-20:
> Since this uses the same struct as `gnrc_netapi_get()` for the message communication, this might be difficult.

I am referring to this struct and the referenced member in particular btw:

https://github.com/RIOT-OS/RIOT/blob/e0e02c04c8a7a65d41fd5ac605f01c6f75e3604a/sys/include/net/gnrc/netapi.h#L102

gebart, 2018-11-20:
@miri64 

> Since this uses the same struct as `gnrc_netapi_get()` for the message communication, this might be difficult.

yes, there are several ways to work around this, but not a single obviously correct way to fix it.

Maybe it would be best to do what you said and make the parameter const, then cast const void *data to void * inside gnrc_netapi_set. This would IMO improve the user facing API since it would let users write constructs like the one in the top post without any casting. The data buffer should never be modified by the set functions in the drivers and gnrc_netif, any such writes should be considered a bug IMO.

miri64, 2018-11-20:
> The data buffer should never be modified by the set functions in the drivers and gnrc_netif, any such writes should be considered a bug IMO.

Will do. Also found some optimizations for the helper functions, but need to evaluate first, before I post those.

gebart, 2018-11-20:
Another solution would be to use a union of void * and const void * for the internal message struct, but that approach has few benefits and could lead to hard to read implementation code

miri64, 2018-11-20:
> Another solution would be to use a union of void * and const void * for the internal message struct, but that approach has few benefits and could lead to hard to read implementation code

Agreed. Also would require, that `GET`/`SET` had somewhat diverging helper functions which could lead to an increase in codesize.

miri64, 2018-11-20:
See #10438.

PR #10438 (2018-11-20): gnrc_netapi: declare `data` for set function as const
### Contribution description
Addresses #10436 as discussed there. This adds a few by to the GNRC codesize on some platforms though (tested with `nrf52dk`).

### Testing procedure
`ifconfig` in `gnrc_networking` still be able to modify any option without crash.

### Issues/PRs references
Resolves #10436
closed 2018-11-20 by miri64