RIOT-OS/RIOT #10442: Problems in current Lora implementation

fesselk, 2018-11-20:
I found a Problem in the current LoRa-Implemetation

According to the LoRa specs https://lora-alliance.org/resource-hub/lorawantm-specification-v11 Paragraph 3.3.1 the Receive Windows after Transmitting should account for a 20 microseconds (us or µs) delay inaccuracy. The comment in current master:pkg/semtech-loramac/contrib/semtech_loramac_timer.c contains a comment starting at line 63
seems wrong to me ( it is stating milliseconds). 

See my comment in #10180 

This might have caused #10352.

If i try to debug this and enable debugging at some places like master:drivers/sx127x/sx127x_getset.c
it behaves different since the debug messages are printed before the state is changed and influence the timing which is critical (is this a Heisen-Bug?) 

There is another Problem:
The first RX window is is just about 55ms, it is closed by the DIO1 interrupt (I don't know why) which makes the 50ms even more critical ( if some one sleeps 22 ms short offset it seems to be very safe (hit the actual receive window in the close center) but only as long as u do not enable to many debug messages ) 
The second RX window is much longer (why?)

'''
this was a join request 
XX:28,877 - INFO # [sx127x] Change state: TX
XX:28,942 - INFO # [sx127x] Change state: IDLE
XX:28,944 - INFO # [sx127x] Change state: IDLE
XX:28,947 - INFO # [sx127x] Change state: IDLE
XX:28,951 - INFO # [semtech-loramac] Transmission completed
XX:30,520 - INFO # [semtech-loramac] MAC timer timeout
XX:31,971 - INFO # [semtech-loramac] MAC timer timeout
XX:32,965 - INFO # [semtech-loramac] MAC timer timeout
XX:33,958 - INFO # [semtech-loramac] MAC timer timeout
XX:34,951 - INFO # [semtech-loramac] MAC timer timeout
XX:35,046 - INFO # [semtech-loramac] MAC timer timeout
first rx window ( a littele delayed for debugging )
XX:35,051 - INFO # [sx127x] Change state: RX
XX:35,103 - INFO # [sx127x] Change state: IDLE
closed by DIO1  52ms 
XX:35,106 - INFO # [semtech-loramac] RX timer timeout
XX:35,109 - INFO # [sx127x] Change state: IDLE
XX:35,945 - INFO # [semtech-loramac] MAC timer timeout
XX:36,087 - INFO # [semtech-loramac] MAC timer timeout
second  rx window  (it has some extra delay Why ( should have been 1 second after rxwindow 1 started ) 
XX:36,092 - INFO # [sx127x] Change state: RX
XX:36,488 - INFO # [sx127x] Change state: IDLE
closed much later (why?)
'''

fesselk, 2018-11-21:
If the SX127x waits a maximum of 25ms, the offset should be less than that (afaik there was a 22 in some versions), and the reason should be stated in the comment (its not the LoRa Standard but the Inprecision of xtimer and time to compensate for extra work (debugging) to be done).

The second one received no accept, but there might have been something on air that the receiver detected.

>> Are you still facing the issues in #10352?
Not with an shorter offset (50ms ist just wrong) 1ms with debugging disabled worked but failed when i enabled debugging, I think  Riot should try to match the center of the receive window and the actual answer. maybe somthing like '''-13*<Symboltime of curret SF>'''

kaspar030, 2018-11-21:
I'm confused. Are you guys talking about microseconds or milliseconds (us or ms)?

fesselk, 2018-11-21:
milliseconds sice i talk about the xtimer corection offset

fesselk, 2018-11-21:
the offset that the standard recomment plays no role in setting up the xtimer  ( the comment is wrong)

kaspar030, 2018-11-21:
> milliseconds sice i talk about the xtimer corection offset

If xtimer has inaccuracies in the millisecond range, that is a bug in xtimer.

jia200x, 2018-11-21:
In any case, I don't think the correction should go [here](https://github.com/RIOT-OS/RIOT/blob/master/pkg/semtech-loramac/contrib/semtech_loramac_timer.c#L68). That's the global timer of LoRaWAN, so ALL LoRaWAN stuff start 50 millis before! I will try to patch this as soon as I find some time

fesselk, 2018-11-21:
no not 50ms (the window by the chip is 25ms as You told me) so u cannot start the window 50ms early

fesselk, 2018-11-21:
and since the window depends on the symbol time the offset might have to depend on it too

jia200x, 2018-11-21:
> no not 50ms (the window by the chip is 25ms as You told me) so u cannot start the window 50ms early

I meant, the current implementation is handling this wrongly. This correction should be done in the MAC layer, exactly where the RX window is open and not in the global timers

> no not 50ms (the window by the chip is 25ms as You told me) so u cannot start the window 50ms early

The stack already does this (check [symbTimeout](https://github.com/RIOT-OS/RIOT/blob/master/pkg/semtech-loramac/contrib/semtech_loramac_radio.c#L99)). The SX127x driver waits 25 symbol times, not 25 millis. With SF12, 25 symbols is around 800 ms. 

The problem is likely to be related to RX windows being open at the wrong time. I will investigate further

fesselk, 2018-11-21:
if at SF7 the window is just 25ms starting the receive 50 ms early is too early

jia200x, 2018-11-21:
> if at SF7 the window is just 25ms starting the receive 50 ms early is too early

Just to clarify, I'm not discussing about the value. I agree is too early and the fact this should be handled better

jia200x, 2018-11-21:
@aabadie what's your opinion?

fesselk, 2018-11-21:
seems i exactly hit this case in my bug #10352
in https://github.com/RIOT-OS/RIOT/blob/master/pkg/semtech-loramac/contrib/semtech_loramac_radio.c#L99 the 25 is doubled to 50 than the xtimer was a little urgent and so th chip could not detect its minimum of 4 symbols to find the begin of the frame that was send on time

fesselk, 2018-11-21:
I just took a look at an other log, is it the case the the 2nd RX window has to uses an other SF?

I saw SF7 for the first window an SF12 for the 2nd, wihch led to the different RX Window length.

jia200x, 2018-11-21:
> I saw SF7 for the first window an SF12 for the 2nd, wihch led to the different RX Window length.

Yes, that's true. They are usually in different SF

fesselk, 2018-11-21:
I was wrong some comments before: the offset should account just for the inprecision of xtimer (and something else is happening on the cpu (like debugging)) part. But the window-timeout (for detection) should be shrunken for higher SF, The RX-window-timeout  should just reach 5 Symbol times (4 is minimum for detection, 1st one might be cut so take 5) over the actual RX-Window starting point (1 and 2 or 5 and 6 seconds after TX Finished)

Since the Inprecission of the Station is limmited by the LoRA standard to just 25µs it can be ignored, 

We may account for the speed of light at SF7 and extend by another Symbol (300km (not sure about the current LoRarecord) take 1ms)

closed 2019-09-10 by stale[bot]