RIOT-OS/RIOT #10443: Unable to find mutex reference: pthread in packages

Citrullin, 2018-11-21:
I added a [package to RIOT](https://github.com/Citrullin/RIOT/blob/iota_new_implementation/pkg/iota-wallet/Makefile) which clones [this git repo](https://github.com/embedded-iota/micro-light-wallet/tree/micro_wallet).
[I added posix and pthread as depedency](https://github.com/Citrullin/RIOT/blob/iota_new_implementation/pkg/iota-wallet/Makefile.dep).
I also implemented an [example in RIOT](https://github.com/Citrullin/RIOT/tree/iota_new_implementation/examples/iota_transaction_node) to use this package. 
But when I compile it, the compiler is not able to find the mutex reference.

```
error: implicit declaration of function 'pthread_mutex_lock'; did you mean 'pthread_atfork'? [-Werror=implicit-function-declaration]
     pthread_mutex_lock(&iota_wallet_bundle_essence_mutex);
```

miri64, 2018-11-21:
Uff, I never implemented a packet using Cmake, so I have to hot-potato this one: @kaspar030, @cgundogan, @cladmi?

Citrullin, 2018-11-21:
@josephnoir Do I understand you correct? In both cases I need to adjust the CMakeList in my library to get around this?

josephnoir, 2018-11-21:
What @cladmi wrote sounds right. You only need to touch the CMakeLists.txt if it doesn't offer a way to accept additional flags. If the `iota-wallet` cmake does have such a way you only need to adjust the [package Makefile](https://github.com/Citrullin/RIOT/blob/iota_new_implementation/pkg/iota-wallet/Makefile) you posted. I want to say there are flags cmake simply accepts per default unless they are explicitly overwritten, might be `CMAKE_C_FLAGS`.

Citrullin, 2018-11-22:
@cladmi @josephnoir Huge thanks. So, I know where my problem is placed at.
I changed that and pushed the changes. So, it passes all the RIOT flags to cmake. Interestingly I got a new issue. A really interesting one. 

```
Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: /usr/bin/arm-none-eabi-gcc 
Build flags: -Werror;-Wno-implicit-fallthrough;-mno-thumb-interwork;-mcpu=cortex-m3;-mlittle-endian;-mthumb;-mfloat-abi=soft;-ffunction-sections;-fdata-sections;-fno-builtin;-fshort-enums;-ggdb;-g3;-Os;-std=c99;-fno-delete-null-pointer-checks;-fdiagnostics-color;-Wstrict-prototypes;-Wold-style-definition;-fno-common;-Wall;-Wextra;-Wformat=2;-Wformat-overflow;-Wformat-truncation;-Wmissing-include-dirs;-include;'/home/citrullin/git/riot_libs/examples/iota_transaction_node/bin/bluepill/riotbuild/riotbuild.h'
Id flags:  

The output was:
1
[01m[Kcc1:[m[K [01;31m[Kfatal error: [m[K'/home/citrullin/git/riot_libs/examples/iota_transaction_node/bin/bluepill/riotbuild/riotbuild.h': No such file or directory
compilation terminated.
```
The interesting part is that this file actually exists. Could it be that the file is generated after cmake is trying to apply -include for it?

Besides the No such file error, there is also this error in CMakeError.log
```
Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: /usr/bin/arm-none-eabi-gcc 
Build flags: 
Id flags:  

The output was:
1
/usr/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/lib/libc.a(lib_a-exit.o): In function `exit':
exit.c:(.text.exit+0x2c): undefined reference to `_exit'
collect2: error: ld returned 1 exit status
```

Citrullin, 2018-11-23:
@cladmi Nevermind, my mistake. I just added ```-CMAKE_C_FLAGS="$(RIOT_CFLAGS)"``` instead of ```-DCMAKE_C_FLAGS="$(RIOT_CFLAGS)" ```. So, it is now able to resolve the references. But, of course, another issue appears. :) The RIOT header files are conflicting with the arm-none-eabi header files. 
E.g.
```
/home/citrullin/git/riot_libs/sys/posix/pthread/include/pthread_mutex.h:33:17: error: conflicting types for 'pthread_mutex_t'
 typedef mutex_t pthread_mutex_t;
                 ^~~~~~~~~~~~~~~
In file included from /usr/arm-none-eabi/include/sys/types.h:239:0,
                 from /usr/arm-none-eabi/include/stdio.h:61,
                 from /home/citrullin/git/riot_libs/examples/iota_transaction_node/bin/pkg/bluepill/iota-wallet/src/iota/common.h:4,
                 from /home/citrullin/git/riot_libs/examples/iota_transaction_node/bin/pkg/bluepill/iota-wallet/src/iota/addresses.c:2:
/usr/arm-none-eabi/include/sys/_pthreadtypes.h:154:20: note: previous declaration of 'pthread_mutex_t' was here
 typedef __uint32_t pthread_mutex_t;      /* identify a mutex */
                    ^~~~~~~~~~~~~~~
```

Citrullin, 2018-11-24:
@josephnoir I thought so too. Maybe I simply have the wrong compiler, since this looks really like a linux target. I targeting a board:
```
BOARD ?= bluepill
```

josephnoir, 2018-11-26:
There are pthread examples in RIOT. Did you check how the include paths differ when compiling those?

cladmi, 2018-11-27:
I just thought about something, could you try reproducing the same error when done from a module directly using RIOT build system. Because as there is `cmake` in the way, it is harder to debug what comes from what and if the issue is on our side or on the cmake integration.

Citrullin, 2018-11-29:
@cladmi [I implemented a Makefile version.](https://github.com/Citrullin/RIOT/tree/iota_new_implementation/pkg/iota-wallet) So, you can check this out. Still the same issue.

frodak17, 2018-12-01:
Hi, I was trying to answer Citrullin's question on [StackOverflow](https://stackoverflow.com/questions/53461598/cmake-for-embedded-remove-linux-posix-headers/).

@josephnoir @cladmi  I don't think the issue is either with cmake or necessarily with RIOT.  I think the problem is with `arm-none-eabi` and `newlib`.

As far as I could tell whenever you include a header file from `newlib` like `stdio.h` it will eventually include `sys/types.h`, `sys/_pthreadtypes.h` and `sys/features.h`.  `sys/_pthreadtypes.h` had a guard and only defined pthread types if `_POSIX_THREADS` was defined.  But this was changed in `newlib` commit `0c081aa62c08c6a00bbacf629e2232dfdd8ba24d` on  Mon Jun 19 2017.  The new guard check is now `#if defined(_POSIX_THREADS) || __POSIX_VISIBLE >= 199506`.  Unfortunately `sys/features.h` will  set `__POSIX_VISIBLE` to 200809 by default.  You can avoid this problem by compiling with strict ANSI mode enabled like using `-std=c99` or `-D__STRICT_ANSI__`. The problem will come back if the source code does anything like `#define _GNU_SOURCE` or anything else that might trigger POSIX mode before including a library header file or trying to use gnu extension `-std=gnu99`.

Anyways the pkg being developed doesn't pass in the `-std=c99` to `iota-wallet` via `xcompile-toolchain.cmake` correctly.  So when `iota-wallet` compiles `newlib` and RIOT have the pthread compiler error issue.

The quickest workaround is to simply update the compiler header files to not use `sys/_pthreadtypes.h` but that isn't really a long term solution. 

I don't know if RIOT has rules stating that strict ANSI is a must because of these kinds of issues.

frodak17, 2018-12-01:
Also the easiest way to demonstrate this issue is:
RIOT/tests/pthread$ CFLAGS=-std=gnu99 make BOARD=bluepill

frodak17, 2018-12-01:
@Citrullin please accept my answer on StackOverflow.  Let me know here if you need any more help.

frodak17, 2018-12-01:
@Citrullin please accept my answer on StackOverflow if you found it useful. Let me know here if you need any more help.

closed 2018-12-03 by Citrullin

cladmi, 2018-12-03:
@frodak17 thank for the info regarding `std=gnu99` I see that we are using it for some cpus in RIOT so may cause issue if they use newlib.

frodak17, 2018-12-03:
@cladmi Personally I think it is an issue with `newlib`.  Sometimes it can be built to provide `pthreads`.  But it is always exposing it's pthreads header files even if this wasn't actually provided as part of newlib.

cladmi, 2018-12-03:
@frodak17 the issue on our side, is also that we overwrite some of the posix headers: https://github.com/RIOT-OS/RIOT/tree/master/sys/posix/pthread/include but not all of them

frodak17, 2018-12-03:
@cladmi Ok.  I just think it is strange that `newlib` would have a header file for something it doesn't seem  to implement when used in the case of `arm-none-eabi`.

Citrullin, 2018-12-19:
@frodak17 I think the question is more: Is it part of the C11 standard? I mean the C11 standard specifies threading, but not as pthreads. If it is against the standard: I totally agree with you, it shouldn't be part of the header files then. But if it is in the standard, it should also be part of newlib.