RIOT-OS/RIOT #10444: Emcute Example does not work with Atmel SAM R21

scarabsoft, 2018-11-21:
#### Description

I am not able to connect from an Atmel SAM R21 to the mosquito.rsmb on my local machine. I followed the steps in [https://github.com/RIOT-OS/RIOT/tree/master/examples/emcute_mqttsn](example).

#### Steps to reproduce the issue

1.) I created an tap device with
`$ sudo ip tuntap add tap0 mode tap user ${USER}`
`$ sudo ip addr add 2001:0db8::/64 dev tap0 `
` $ sudo ip link set tap0 up`

2.) Added some stuff into the makefile

DEVELHELP ?= 1
QUIET ?= 1
TAP ?= tap0
IPV6_PREFIX ?= 2001:0db8::/64
ifeq (native,$(BOARD))
TERMDEPS += uhcpd-daemon
.PHONY: uhcpd-daemon
uhcpd-daemon: host-tools
	$(RIOTTOOLS)/uhcpd/bin/uhcpd $(TAP) $(IPV6_PREFIX) &
TERMPROG ?= sudo sh $(RIOTTOOLS)/ethos/start_network.sh
TERMFLAGS ?= $(PORT) $(TAP) $(IPV6_PREFIX)
TERMDEPS += host-tools
endif
include $(RIOTBASE)/Makefile.include
.PHONY: host-tools
host-tools:
	$(Q)env -u CC -u CFLAGS make -C $(RIOTTOOLS)
ifneq (,$(filter cc110x,$(USEMODULE)))          # radio is cc110x sub-GHz
  DEFAULT_CHANNEL ?= 0
  CFLAGS += -DCC110X_DEFAULT_CHANNEL=$(DEFAULT_CHANNEL)
else
  ifneq (,$(filter at86rf212b,$(USEMODULE)))    # radio is IEEE 802.15.4 sub-GHz
    DEFAULT_CHANNEL ?= 5
    CFLAGS += -DIEEE802154_DEFAULT_SUBGHZ_CHANNEL=$(DEFAULT_CHANNEL)
  else                                          # radio is IEEE 802.15.4 2.4 GHz
    DEFAULT_CHANNEL ?= 26
    CFLAGS += -DIEEE802154_DEFAULT_CHANNEL=$(DEFAULT_CHANNEL)
  endif
endif

3.) started mqtt broker with this config

`# add some debug output`
`trace_output protocol`
`# listen for MQTT-SN traffic on UDP port 1885`
`listener 1885 INADDR_ANY  mqtts`
`  ipv6 true`

4.) try to connect after ping to the broker

sudo sh /home/ddymke/repository/riot/dist/tools/ethos/start_network.sh /dev/ttyACM0 tap0 2001:0db8::/64
ioctl(TUNSETIFF): Device or resource busy
net.ipv6.conf.tap0.forwarding = 1
net.ipv6.conf.tap0.accept_ra = 0
----> ethos: sending hello.
----> ethos: activating serial pass through.
----> ethos: hello received
gnrc_uhcpc: Using 8 as border interface and 7 as wireless interface.
uhcp_client(): sending REQ...
udp_sendto(): sendto(): Cannot assign requested address
main(): This is RIOT! (Version: 2018.10-RC1-206-g455cd)
MQTT-SN example application

Type 'help' to get started. Have a look at the README.md for moreinformation.
> ----> ethos: hello received
gnrc_uhcpc: Using 8 as border interface and 7 as wireless interface.
uhcp_client(): sending REQ...
main(): This is RIOT! (Version: 2018.10-RC1-206-g455cd)
MQTT-SN example application

Type 'help' to get started. Have a look at the README.md for moreinformation.
got packet from fe80::4e8:64ff:fe0a:fe6c port 35620
uhcp: push from fe80::4e8:64ff:fe0a:fe6c:35620 prefix=2001:db8::/64
gnrc_uhcpc: uhcp_handle_prefix(): configured new prefix 2001:db8::7b7c:3c3e:f334:64fa/64

> ping6 2001:0db8::%8
ping6 2001:0db8::%8
ping timeout
12 bytes from 2001:db8::: id=83 seq=2 hop limit=64 time = 12.655 ms
12 bytes from 2001:db8::: id=83 seq=3 hop limit=64 time = 12.640 ms
--- 2001:0db8:: ping statistics ---
3 packets transmitted, 2 received, 34% packet loss, time 3.0648698 s
rtt min/avg/max = 12.640/12.647/12.655 ms

> con 2001:0db8:: 1885
con 2001:0db8:: 1885
uhcp_client(): no reply received
uhcp_client(): sending REQ...
got packet from fe80::94e5:f5ff:fea9:997d port 53933
uhcp: push from fe80::94e5:f5ff:fea9:997d:53933 prefix=2001:db8::/64
gnrc_uhcpc: uhcp_handle_prefix(): configured new prefix 2001:db8::7b7c:3c3e:f334:64fa/64

#### Expected results

I expect the client connecting to the broker successfully.

#### Actual results

I does not connect. The log of the MQTT-SN Broker

$ ./broker_mqtts config.conf 
20181121 084425.010 CWNAN9999I Really Small Message Broker
20181121 084425.010 CWNAN9998I Part of Project Mosquitto in Eclipse
(http://projects.eclipse.org/projects/technology.mosquitto)
20181121 084425.010 CWNAN0049I Configuration file name is config.conf
20181121 084425.010 CWNAN0053I Version 1.3.0.2, Nov  5 2018 15:11:15
20181121 084425.010 CWNAN0054I Features included: bridge MQTTS 
20181121 084425.010 CWNAN9993I Authors: Ian Craggs (icraggs@uk.ibm.com), Nicholas O'Leary
20181121 084425.010 CWNAN0300I MQTT-S protocol starting, listening on port 1885

20181121 084727.928 3 fe80::2:1883  <- MQTT-S CONNECT cleansession: 0
20181121 084727.928 -74199360 fe80::2:1883 myid -> MQTT-S CONNACK returncode 0 (0)
20181121 084727.928 CWNAN0000I Client connected to udp port 1885 from myid (fe80::2:1883)
20181121 084742.955 3 fe80::2:1883 myid <- MQTT-S CONNECT cleansession: 0
20181121 084742.955 CWNAN0034I Duplicate connection attempt received for client identifier "myid" from address fe80::2:1883, ending oldest connection
20181121 084742.956 -74199360 fe80::2:1883 myid -> MQTT-S CONNACK returncode 0 (0)
20181121 084742.956 CWNAN0000I Client connected to udp port 1885 from myid (fe80::2:1883)
20181121 084757.972 3 fe80::2:1883 myid <- MQTT-S CONNECT cleansession: 0
20181121 084757.983 CWNAN0034I Duplicate connection attempt received for client identifier "myid" from address fe80::2:1883, ending oldest connection
20181121 084757.983 -74199360 fe80::2:1883 myid -> MQTT-S CONNACK returncode 0 (0)
20181121 084757.983 CWNAN0000I Client connected to udp port 1885 from myid (fe80::2:1883)

So it seems that there is a connection between both devices. Futhermore I can see the UDP pakets via wireshark. And I am able to use the mosquito.rsmb broker with the [https://github.com/njh/mqtt-sn-tools](mqtt-sn-tools). So I am pretty sure that there is something wrong with the emcute client

#### Versions

$  ./dist/tools/ci/print_toolchain_versions.sh

Operating System Environment
-----------------------------
       Operating System: "Linux Mint" "19 (Tara)"
                 Kernel: Linux 4.15.0-39-generic x86_64 x86_64

Installed compiler toolchains
-----------------------------
             native gcc: gcc (Ubuntu 7.3.0-27ubuntu1~18.04) 7.3.0
      arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q3-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
                avr-gcc: missing
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
                  clang: clang version 6.0.0-1ubuntu2 (tags/RELEASE_600/final)

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "3.0.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
               avr-libc: missing (missing)

Installed development tools
---------------------------
                  cmake: cmake version 3.10.2
               cppcheck: missing
                doxygen: 1.8.13
                 flake8: missing
                    git: git version 2.17.1
                   make: GNU Make 4.1
                openocd: Open On-Chip Debugger 0.9.0-dev-gbdd8312f (2018-10-23-17:22)
                 python: Python 3.6.5 :: Anaconda, Inc.
                python2: Python 2.7.15rc1
                python3: Python 3.6.5 :: Anaconda, Inc.
             coccinelle: missing

PeterKietzmann, 2018-11-21:
Could you try to set ``CFLAGS += -DGNRC_IPV6_NIB_CONF_SLAAC=1`` in the application Makefile? See description in #9522 as a reference.

scarabsoft, 2018-11-21:
Thanks for your response. I did a $ sudo make clean flash term and can see the following in the mosquitto.rsmb log file

20181121 094245.760 3 fe80::265:1eff:fea7:46e3:1883 myid <- MQTT-S CONNECT cleansession: 0
20181121 094245.760 CWNAN0034I Duplicate connection attempt received for client identifier "myid" from address fe80::265:1eff:fea7:46e3:1883, ending oldest connection
20181121 094245.760 505839296 fe80::265:1eff:fea7:46e3:1883 myid -> MQTT-S CONNACK returncode 0 (0)
20181121 094245.760 CWNAN0000I Client connected to udp port 1885 from myid (fe80::265:1eff:fea7:46e3:1883)
20181121 094300.775 3 fe80::265:1eff:fea7:46e3:1883 myid <- MQTT-S CONNECT cleansession: 0
20181121 094300.792 CWNAN0034I Duplicate connection attempt received for client identifier "myid" from address fe80::265:1eff:fea7:46e3:1883, ending oldest connection
20181121 094300.792 505839296 fe80::265:1eff:fea7:46e3:1883 myid -> MQTT-S CONNACK returncode 0 (0)
20181121 094300.792 CWNAN0000I Client connected to udp port 1885 from myid (fe80::265:1eff:fea7:46e3:1883)
20181121 094315.824 3 fe80::265:1eff:fea7:46e3:1883 myid <- MQTT-S CONNECT cleansession: 0
20181121 094315.824 CWNAN0034I Duplicate connection attempt received for client identifier "myid" from address fe80::265:1eff:fea7:46e3:1883, ending oldest connection
20181121 094315.824 505839296 fe80::265:1eff:fea7:46e3:1883 myid -> MQTT-S CONNACK returncode 0 (0)
20181121 094315.824 CWNAN0000I Client connected to udp port 1885 from myid (fe80::265:1eff:fea7:46e3:1883)

I am using the latest version of riot os (git-hash: 455cdcd00c071a0a5dfc6e4397467717995111f9)

PeterKietzmann, 2018-11-21:
> I did a $ sudo make clean flash term 

After adding the above proposed `CFLAG`? Unfortunately the log indicates that the client still uses the link local (source) IPv6 address to connect. (i) mosquitto.rsmb has known issues with handling link local addresses and (ii) the RIOT node should use its global IPv6 address. Can you paste the output of `ifconfig` here before you send the connection request?

> Duplicate connection attempt received for client identifier "myid" 

Did you also restart the broker after trying to connect a second time with "myid"?

PeterKietzmann, 2018-11-30:
@scarabsoft any news?

PeterKietzmann, 2018-12-13:
ping @scarabsoft

PeterKietzmann, 2018-12-18:
The provided [mosquitto.rsmb log](https://github.com/RIOT-OS/RIOT/issues/10444#issuecomment-440582896) indicates that link local IPv6 addresses were used with which the broker has know issues. Furthermore the issue looks pretty like the issue explained in #9522. Thus, I expect `CFLAGS += -DGNRC_IPV6_NIB_CONF_SLAAC=1` to fix the problem. I'm closing this issue due to unresponsiveness. @scarabsoft please reopen if I am mistaken.

scarabsoft, 2018-12-19:
@PeterKietzmann sorry for the late response - I was in vacation without any technical device..

Side node: I changed my ipv6 from 2001:0db8 --> 2001:db8

Well I applied the flag to the Makefile and my if config looks like:

main(): This is RIOT! (Version: 2018.10-RC1-206-g455cd)
MQTT-SN example application

Type 'help' to get started. Have a look at the README.md for moreinformation.
got packet from fe80::305e:baff:fe40:d40d port 33507
uhcp: push from fe80::305e:baff:fe40:d40d:33507 prefix=2001:db8::/64
gnrc_uhcpc: uhcp_handle_prefix(): configured new prefix 2001:db8::7b76:3154:d8ab:6c3e/64
> ifconfig
ifconfig
Iface  7  HWaddr: 6C:3E  Channel: 26  Page: 0  NID: 0x23
          Long HWaddr: 79:76:31:54:D8:AB:6C:3E 
           TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4 
          AUTOACK  ACK_REQ  CSMA  MTU:1280  HL:64  RTR  
          RTR_ADV  IPHC  
          Source address length: 8
          Link type: wireless
          inet6 addr: fe80::7b76:3154:d8ab:6c3e  scope: local  VAL
          inet6 addr: 2001:db8::7b76:3154:d8ab:6c3e  scope: global  VAL
          inet6 group: ff02::2
          inet6 group: ff02::1
          inet6 group: ff02::1:ffab:6c3e
          
Iface  8  HWaddr: 00:0C:FF:E2:7B:63 
          MTU:1500  HL:64  RTR  
          RTR_ADV  Source address length: 6
          Link type: wired
          inet6 addr: fe80::20c:ffff:fee2:7b63  scope: local  VAL
          inet6 addr: fe80::2  scope: local  VAL
          inet6 group: ff02::2
          inet6 group: ff02::1
          inet6 group: ff02::1:ffe2:7b63
          inet6 group: ff02::1:ff00:2
          

but I am still not able to connect to the broker

./broker_mqtts config.conf 
20181219 085323.320 CWNAN9999I Really Small Message Broker
20181219 085323.321 CWNAN9998I Part of Project Mosquitto in Eclipse
(http://projects.eclipse.org/projects/technology.mosquitto)
20181219 085323.321 CWNAN0049I Configuration file name is config.conf
20181219 085323.321 CWNAN0053I Version 1.3.0.2, Nov  5 2018 15:11:15
20181219 085323.321 CWNAN0054I Features included: bridge MQTTS 
20181219 085323.321 CWNAN9993I Authors: Ian Craggs (icraggs@uk.ibm.com), Nicholas O'Leary
20181219 085323.321 CWNAN0300I MQTT-S protocol starting, listening on port 1885
20181219 090005.085 3 fe80::20c:ffff:fee2:7b63:1883  <- MQTT-S CONNECT cleansession: 0
20181219 090005.085 -1947124032 fe80::20c:ffff:fee2:7b63:1883 myid -> MQTT-S CONNACK returncode 0 (0)
20181219 090005.085 CWNAN0000I Client connected to udp port 1885 from myid (fe80::20c:ffff:fee2:7b63:1883)
20181219 090020.101 3 fe80::20c:ffff:fee2:7b63:1883 myid <- MQTT-S CONNECT cleansession: 0
20181219 090020.116 CWNAN0034I Duplicate connection attempt received for client identifier "myid" from address fe80::20c:ffff:fee2:7b63:1883, ending oldest connection
20181219 090020.116 -1947124032 fe80::20c:ffff:fee2:7b63:1883 myid -> MQTT-S CONNACK returncode 0 (0)
20181219 090020.116 CWNAN0000I Client connected to udp port 1885 from myid (fe80::20c:ffff:fee2:7b63:1883)
20181219 090035.132 3 fe80::20c:ffff:fee2:7b63:1883 myid <- MQTT-S CONNECT cleansession: 0
20181219 090035.144 CWNAN0034I Duplicate connection attempt received for client identifier "myid" from address fe80::20c:ffff:fee2:7b63:1883, ending oldest connection
20181219 090035.144 -1947124032 fe80::20c:ffff:fee2:7b63:1883 myid -> MQTT-S CONNACK returncode 0 (0)
20181219 090035.144 CWNAN0000I Client connected to udp port 1885 from myid (fe80::20c:ffff:fee2:7b63:1883)
20181219 090050.173 3 fe80::20c:ffff:fee2:7b63:1883 myid <- MQTT-S CONNECT cleansession: 0
20181219 090050.173 CWNAN0034I Duplicate connection attempt received for client identifier "myid" from address fe80::20c:ffff:fee2:7b63:1883, ending oldest connection
20181219 090050.173 -1947124032 fe80::20c:ffff:fee2:7b63:1883 myid -> MQTT-S CONNACK returncode 0 (0)
20181219 090050.173 CWNAN0000I Client connected to udp port 1885 from myid (fe80::20c:ffff:fee2:7b63:1883)
20181219 090058.564 3 fe80::20c:ffff:fee2:7b63:1883 myid -> MQTT-S PINGRESP (0)

PeterKietzmann, 2018-12-19:
The output you posted is from`ifconfig` on the samr21-xpro with ethernet/ethos?

On the wired interface there's no global address. @miri64 ideas?

scarabsoft, 2018-12-19:
Yes, I guess so. Its connected via USB

miri64, 2020-07-03:
AFAIK `2001:db8::` is not a legal IPv6 address, because it is missing a non-zero suffix. @ghost can you maybe try `2001:db8::1`? Sorry for _very_ late response, somehow this PR was not on my radar...

closed 2021-07-08 by stale[bot]