RIOT-OS/RIOT #10468: Tinycryt upstream rewrote history in master branch

cladmi, 2018-11-26:
#### Description

Murdock recently fails to build packages related to `tinycrypt` as the reference commit was removed from the upstream master branch:

https://ci.riot-os.org/RIOT-OS/RIOT/9819/ae4cecd14c36f7bf68a5357727e5874822bebe60/output/compile/tests/pkg_tinycrypt/native:gnu.txt

The commit was this one https://github.com/01org/tinycrypt/commit/3ea1a609e7aff9f2d8d13803e1076b7a8e551804 and as you can see, it does not reference any branch.

The new commit is this one: https://github.com/01org/tinycrypt/commit/6a22712bebbf2fc60d9fc6192dddefd5ad1933e3

Thanks to the new github feature to show forced push we can easily see here that the master branch was force-pushed:

https://github.com/01org/tinycrypt/pull/35

Also attached as an image.

[image]

#### Handling of this

Both commits can be compared and contain the same content, so could be replaced one by another.
I will do a PR to update to the new reference as the content stayed the same.

However, what is the state of keeping a package in RIOT, even more a cryptographic library, where upstream rewrites history in master ? 11month after being merged…

#### Versions

https://github.com/RIOT-OS/RIOT/blob/ce8815f4d1af13ef21528c8140a226b3b24143d0/pkg/tinycrypt/Makefile

cladmi, 2018-11-26:
@haukepetersen could be of your interest as you added the package.

kaspar030, 2018-11-26:
> However, what is the state of keeping a package in RIOT, even more a cryptographic library, where upstream rewrites history in master ?

If we can rule out malicious intent, we should open an issue with them and let them know about the implications of a force push.

miri64, 2018-11-26:
Mh... A force push does not remove a commit from GitHub (evident by the link provided by @cladmi by #10470).up until now I understood that that was the cause of Murdock not being able to fetch that commit. However, since the commit is not removed this might require us to rethink our fetching strategy for `pkg` as the commit still could have bin fetched afaik.

BTW, history rewrites can happen (even in a crypto lib) for completely legitimate reasons, e. g. if secrets were accidentally exposed (which can happens to the best of us, even crypto wizards ;))

kaspar030, 2018-11-27:
> However, since the commit is not removed this might require us to rethink our fetching strategy for `pkg` as the commit still could have bin fetched afaik.

I'm not sure about that. If the commit is not referenced by anything in that repository, it will become garbage collected eventually and it won't be sent to new clones. Github having a copy somewhere is similar to most of our build workers having cached the commit.

Also the main reason our fetching strategy is based on a commit hash is that the hash perfectly describes the resulting data. In git, that includes git's metadata, so it is possible that the same file data ends up having different hashes. But that is a feature.

> BTW, history rewrites can happen (even in a crypto lib) for completely legitimate reasons, e. g. if secrets were accidentally exposed (which can happens to the best of us, even crypto wizards ;))

+1. IMO this just should be handled differently, especially by crypto wizards. ;)

miri64, 2018-11-27:
> I'm not sure about that. If the commit is not referenced by anything in that repository, it will become garbage collected eventually and it won't be sent to new clones. Github having a copy somewhere is similar to most of our build workers having cached the commit.

You might be right with that.

> […] In git, that includes git's metadata, so it is possible that the same file data ends up having different hashes. But that is a feature.

??!? A blob (git's representation of file's content) always has the same hash (as long as the content stays the same). Blob hashes are used with file names (and other metadata) when constructing tree objects, so the tree object (or the commit object pointing to it) could be different. But file *data* should always have the same hash (though that would be hard to check out at commit level).

kaspar030, 2018-11-27:
> But file _data_ should always have the same hash (though that would be hard to check out at commit level).

Yeah, sorry, I meant what was happening here: exactly the same file data might get pointed to by different *commit* hashes because of differing git metadata.

miri64, 2018-11-27:
It probably does cause everyone pain that wants to pull from their repo in standard config.

jcarrano, 2018-12-05:
I attach their response to @kaspar030's email so that people who are not on the maintainers list can know what happened.

>Subject: Re: [riot-maintainer] tinycrypt force pushes
>Date: Mon, 3 Dec 2018 01:57:29 +0000
>From: Misoczki, Rafael <rafael.misoczki@intel.com>
>
>Hi Kaspar,
>
>Thank you for your feedback. I'm really glad that you took the time to write this note. 
>
>Indeed, I recently force pushed a few commits. What happened is that I accidentally pushed a few things using my local git user that did not match the one in GitHub officially maintaining the tinycrypt project. To avoid such kind of inconsistency, I force pushed the changes once more using the correct (publicly known) git user. 
>
>I was unaware this would cause problems to your project... I'm sorry for that. But thanks to your message I will now make sure no more force pushes are made on tinycrypt moving forward. 
>
>Also, please let me take this opportunity to say that you folks from RIOT OS are very welcome to send us any comments and/or suggestions related to our project. RIOT OS is a very interesting project and we want to make sure you folks keep using tinycrypt moving forward. 
>
>Thanks, and have a great week!
>
>Cheers,
>Rafael

Can we close this?

closed 2019-07-08 by cladmi