RIOT-OS/RIOT #10488: core/thread: mutex failure to block inside ISRs

ZetaR60, 2018-11-28:
#### Description
On systems that use an interrupt to do a thread context switch ```mutex_lock(mutex)``` does not block as expected when inside an ISR and the mutex is already locked.

#### Steps to reproduce the issue
On a system such as ```esp8266```, issue an ```i2c_acquire(dev)``` or similar while inside of an ISR and while the same device is already acquired. I expect that this can be reproduced on other systems that use an interrupt to do a thread context switch.

#### Expected results
The system yields to another thread until the mutex is unblocked.

#### Actual results
The thread yield is put off until after the ISR completes, and the ISR's access collides with whatever is already using the bus.

#### Versions
Board was an ```esp8266-esp-12x```. Initial behavior found by @gschorcht in #10430, but I am following up on what is causing the issue.

#### References
#10430 Adds support for an I2C-based GPIO expander, and utilizing this device to generate a pin interrupt that alters the state of another expander pin causes some strange behavior. See @gschorcht  's explanation of the behavior [here](https://github.com/RIOT-OS/RIOT/pull/10430#issuecomment-441556234).

ZetaR60, 2018-11-28:
Note: I initially thought that this does not affect systems with non-interrupt context switches such as ATmega, but I see that ATmega also only schedules the switch until the end of the ISR.

gschorcht, 2018-11-28:
In short, in the case described in #10430, the problem was simply that an I2C write access to one device causes an interrupt on another device while the I2C write access is not yet completed. However, the ISR requires I2C read access to the second device. If both devices are on the same I2C bus, I2C read access will either fail o interfere with the I2C write access.

I'm not really sure whether the behavior is really wrong and that it is related to the interrupt based context switch.

As I understand what exactly happens is:

- thread x has the mutex and is interrupted
- the ISR tries to get the mutex by calling `mutex_lock`, however the mutex is locked
- since `mutex_lock` is blocking, the mutex tries to add the calling thread (still thread x) to the mutex queue
- then the mutex calls 'thread_yield_higher', which simply sets 'sched_context_switch_request' for many architectures when 'thread_yield_higher' is called in an ISR.
- that is, the mutex simply returns without a context switch to the ISR

Even if  `thread_yield_higher` would do a context switch immediately, as long as there is no thread with higher priority, it would also continue to execute the ISR in the context of the interrupted thread.

From my point of view, an ISR must not call blocking functions such as `mutex_lock`. Therefore, operation like I2C, SPI, ... that rely on blocking mutexes should not be executed at all in an ISR.

gschorcht, 2018-11-28:
@gebart 
> This is why we don't allow drivers to use SPI or i2c from inside ISRs

Yes, that's exactly what I said.

On the other hand, design rules of RIOT are also to keep the number of threads small and to give each thread another priority. Creating a thread for each different device driver makes it difficult to meet these design rules. Especially the right choice of the priority might become a problem on architectures that do not allow to increase the default number of priorities due resource problems.

In the device driver in #10430, I used a workaround that works for this driver, which let return the ISR immediatly if the driver is already within an I2C access and to handle the pending event by the driver when the I2C access has been finished. But this works only for this driver and is probably no general solution.

In any case, it should be documented anywhere very clear, that drivers or anything else must not access I2C or SPI. Maybe it is documented and I just missed it.

gschorcht, 2018-11-29:
@miri64 I prefer to continue the discussion of a handler for driver here to avoid to discuss something that is out of topic in #10496. 

In #10496 you said
> Mh... I'm not sure #10488 can be solved in other ways. Most of our drivers are based on SPI or I2C and we didn't face any problems so far with them.

Sure, I've written some sensor drivers that all use I2C or SPI. They all also work with interrupts as long as they only inform the application that an interrupt has occurred, and the application then asynchronously accesses the sensor in the application thread.

The problem occurs when the driver would have to access I2C or SPI in the ISR to perform its function. The only option I see is to return immediately from the ISR and to have a thread that is handling interrupts asynchronously. That's exactly how the netdev drivers work. But having a thread for any sensor driver ... So grouping them and having only one handling thread might also work sensor drivers.

miri64, 2018-11-29:
Mh, but I don't see how multiple threads per device (as we have currently) pose a problem with that.

gschorcht, 2018-11-29:
> #10496 is just about network device drivers. What we do for sensor drivers is unrelated from that.

I know. Sorry, my mistake, I just thought that having one thread per device driver would already dramatically increase the number of threads as the number of sensors increases. If this is not a problem, we can just do it that way.

ZetaR60, 2018-11-29:
Sorry if my initial issue seemed kind of dumb. For some reason, it was not obvious to me that you wouldn't be able to block within an ISR.

What about creating a singular max priority thread to handle most complex tasks triggered by interrupts? The thread would have a queue for callbacks usable by anything that needs to be triggered by an interrupt but uses functions that are not legal for use within an ISR.

ZetaR60, 2018-11-30:
I have edited the title and description to more accurately reflect what is being discussed.

ZetaR60, 2018-11-30:
> The disadvantage would be that the interrupt events are handled in the order of the handler queue, which determines the priority and possibly causes unnecessary blocks.

With a slight queue lookup overhead, a numerical priority could be added to lessen this problem. I assume that anything needing a really fast response would be executed directly in the ISR, so the overhead might not be much of an issue.

> For example, if you have interrupt events from two different drivers. One driver needs access to I2C and the other driver requires access to SPI or another I2C bus. In this case, the I2C access of the first driver must be completed before the interrupt event of the second driver can be handled, even if it does not interfere. That is, the first driver blocks the seconds driver, although it would not be necessary.

Wouldn't one get blocked no matter how you executed it on a single-core system, just because you can't be simultaneously executing code to use each bus? I suppose in principle you could run a little bit of another thread while waiting for a response, but SPI and I2C aren't set up that way in RIOT. Something at some point must decide the order of events.

gschorcht, 2018-11-30:
> With a slight queue lookup overhead, a numerical priority could be added to lessen this problem. I assume that anything needing a really fast response would be executed directly in the ISR, so the overhead might not be much of an issue.

The overhead is probably not a problem, just the latency caused by the sequential handling of interrupt events. While there are certainly no special latency requirements for data-ready interrupts of a gas sensor that only provides data once every second, the interrupt events in GPIO expanders must be handled immediately. So they should be prioritized.

> Wouldn't one get blocked no matter how you executed it on a single-core system, just because you can't be simultaneously executing code to use each bus? I suppose in principle you could run a little [...]

If the `periph/i2c` and `periph/spi` implementations use hardware implementations for SPI and I2C, they could use`xtimer_usleep` to be blocked for a certain time before they polls the status of a transfer. In that case another thread could use another serial interfaces also on single core system. I don't exactly, but I would guess most implementations are simply busy waiting to poll the status.

gschorcht, 2018-11-30:
@ZetaR60 While thinking about a solution, it became clear that the workaround as used in #10430 doesn't really help because the driver is simply marked as busy to protect himself from his own interrupt events. But, if another device uses `I2C` and the GPIO expander driver gets an interrupt because a pin has changed its level, this would interfere with I2C access.

Therefore, I have to change the driver, probably to let it in run in a separate thread for the moment.

gschorcht, 2018-12-05:
@ZetaR60 I made a first attempt in PR #10555 to solve the problem described above.

closed 2019-09-10 by stale[bot]