RIOT-OS/RIOT #10489: How define an interrupt handler for dma stream?

bluephoton, 2018-11-28:
I'm brand new to this, so likely my ignorance. I couldn't find any info on how to define ISR, looking at the code it appears that some are defined with attribute weak. So I though I add same name in my file and things should work. I did this in my **.cpp** file
``` cpp
void isr_dma1_stream2(void)
{
   // code
}
```

unfortunately when DMA transfer complete interrupt triggers I get RIOT kernel panic, dummy handler message and cpu was halted. Which means interrupt vector is not pointing to my handler!

Could you please point me to an example on how to do this?

gebart, 2018-11-28:
You write that you put this in a .cpp file, did you remember to use `extern "C" { ... }` for the ISR function?
The default ISRs are C language functions, so the replacements need to be declared with C linkage too.

Try placing this at the top of your .cpp file where the ISR is defined.
```c
extern "C" {
void isr_dma1_stream2(void);
}
```

bluephoton, 2018-11-28:
Fantastic! That worked, thanks a lot @gebart !

closed 2018-11-28 by bluephoton