RIOT-OS/RIOT #10496: [RFC] gnrc_netif: Group multiple devices/connections into single thread?

miri64, 2018-11-28:
https://github.com/RIOT-OS/RIOT/issues/10488#issuecomment-442362850 and some discussions with @haukepetersen regarding Bluetooth LE got me thinking: How sensible is it to have one thread per interface? Especially with emerging new boards which have multiple network devices or with BLE where each established connection/pairing can (or even should) be abstracted as an interface, this sounds like a RAM explosion to happen. Also syncing between two threads when a multi-capable radio like the `nrf52840` is switching between modes would also be quite a hassle.

In a short discussion with @haukepetersen we came up with the following draft for a concept:

- Interfaces are (GNRC-internally) not identified by a PID anymore, but the `gnrc_netif_t` pointer (taking discussions around a 64-bit support into account, we need to come up with an alternative solution for that, since `demux_ctx` - see below - only has 32-bit...).
   - ⇒ The PID member of `gnrc_netif_t` is replaced with a suitable identifier for user space identification (e.g. a running number).
- `GNRC_NETTYPE_NETIF` is redefined from being `-1` to a value `> GNRC_NETIF_UNDEF`. This has the result, that `GNRC_NETTYPE_NETIF` would [become a subscribable type via `gnrc_netreg`](http://doc.riot-os.org/group__net__gnrc__netreg.html#gafb6a9d0b840aae07c3c1617932835ef1).
- `gnrc_netif_t` get's a `gnrc_netreg_t` member which the corresponding handler thread uses to subscribe itself to the (nettype, [demux_ctx](http://doc.riot-os.org/structgnrc__netreg__entry.html#a0fe3f1311abb5730b17f0d86fe97bf14)) tuple (`GNRC_NETTYPE_NETIF`, interface identifier).
- If someone wants to send to the interface, they set the interface identifier in the [GNRC netif header](http://doc.riot-os.org/group__net__gnrc__netif__hdr.html) of the packet and use [`gnrc_netapi_dispatch_send()`](http://doc.riot-os.org/group__net__gnrc__netapi.html#ga2b5b0f662061aa332cc1cbb9c145ee07) to send it to (`GNRC_NETTYPE_NETIF`, interface identifier). This way only the corresponding handler thread for the network interface is receiving the packet.
- The corresponding handler thread of the network interface uses the interface identifier in the GNRC netif header to access the interface and uses its `gnrc_netif_ops_t::send` operation to pass it on to the network device (or alternative sending approaches ;-)).
- Grouping of interfaces could e.g. be done by MAC protocol ⇒ No mac protocol is one thread, [contikimac](https://github.com/RIOT-OS/RIOT/pull/9478)  another, etc.

This has the added benefit, that `gnrc_netif` would act much more like other GNRC modules, as it currently bypasses the abstraction layer provided by `gnrc_netreg` and only allows `gnrc_netapi` to directly use PIDs for addressing.

Another benefit is that we could get finally a proper loopback interface without much overhead, with `gnrc_ipv6`'s thread being the handler for that interface (and its interface identifier being the `NULL` pointer ;-)).

Note that this just contradicts one concept in https://github.com/RIOT-OS/RIOT/issues/9903: that the `gnrc_netif_t` variable would be stored on stack. If the user supplies the interface struct on `_create()`, a `GNRC_NETIF_NUMOF`-based approach is still possible, even necessary. So #9903 actually should be a step towards the goal of the issue of this discussion.

Also note, that with this approach multiple interface handler threads are still possible.

I don't know how much this interferes with #7736 and its idea to allocate the netdev's layers in the network interface thread. :-/

gschorcht, 2018-11-29:
Good idea. One question that came to mind was: how should the GNRC interfaces be prioritized? Is it even necessary to prioritize them. Or are the interrupts of different GNRC interfaces simply handled in the order in which they were created?

miri64, 2018-11-29:
> how should the GNRC interfaces be prioritized?

Since `gnrc_netapi` doesn't have any way to priorities except for thread priority, they won't.

> Is it even necessary to prioritize them.

Currently they all are handled in the same priority level.

> Or are the interrupts of different GNRC interfaces simply handled in the order in which they occured?

They should be, because they are just dropped in the message queue (or alternatively fired as event when #9326 is merged). With the message queue approach it might happen, that the queue overruns and interrupts are dropped.

gschorcht, 2018-11-29:
> Since gnrc_netapi doesn't have any way to priorities except for thread priority, they won't.

I guess that the only prerequisite is that the handler thread for the network interfaces has a higher priority than the threads for upper layers.

miri64, 2018-11-29:
Yepp

gschorcht, 2018-11-29:
This concept of having only one handler thread might also be a solution for the problem in #10488 for other drivers that require access to SPI or I2C.

miri64, 2019-04-27:
@bergzand 

> I don't know how much this interferes with #7736 and its idea to allocate the netdev's layers in the network interface thread. :-/

Any words on that?

jia200x, 2019-11-12:
this goes in the same direction of #12688