RIOT-OS/RIOT #10510: xtimer_set_msg: crash when using same message for 2 timers

jia200x, 2018-11-29:
#### Description
I've noticed this pattern gives (sometimes) a stack overflow:
```
xtimer_t timer1, timer2;
msg_t msg;
...
xtimer_set_msg(&timer1, 1000000, &msg, target_pid);
xtimer_set_msg(&timer2, 2000000, &msg, target_pid);

/** 
After 2 seconds:

Stack pointer corrupted, reset to top of stack
Misc
EXC_RET: 0xfffffff1
*/
```

The reason is because of [this line](https://github.com/RIOT-OS/RIOT/blob/master/sys/xtimer/xtimer.c#L150).

Setting the target PID in sender PID has the following effect:

1. `xtimer_set_msg(&timer2, 2000000, &msg, target_pid)` sets `msg->sender_pid` to the thread pid (e.g 3)
2. timer1 triggers. [`_callback_msg`](https://github.com/RIOT-OS/RIOT/blob/master/sys/xtimer/xtimer.c#L138) calls `msg_send_int` with `target_pid`=`msg->sender_pid` (3 in this case)
3. `msg_send_int` sets `sender_pid` to KERNEL_PID_ISR (33)
4. timer2 triggers. Calls `msg_send_int` with `target_pid`=`msg->sender_pid` (**now it's KERNEL_PID_ISR!**)

Then, the second message is never delivered and the node might crash. 

#### Steps to reproduce the issue

Try this snippet:
```
#define SENDER_PRIO         (THREAD_PRIORITY_MAIN - 1)

static kernel_pid_t sender_pid;
static char sender_stack[THREAD_STACKSIZE_MAIN / 2];

static void *sender(void *arg)
{
    (void)arg;

    msg_t msg;
    msg_t msg_queue[8];
    msg_init_queue(msg_queue, 8);

    while (1) {
        msg_receive(&msg);
        puts("Received message!");
    }

    /* this should never be reached */
    return NULL;
}

int main(void)
{
    sender_pid = thread_create(sender_stack, sizeof(sender_stack),
                               SENDER_PRIO, 0, sender, NULL, "sender");
    printf("Wait 5 secs\n");
    xtimer_sleep(5);

    xtimer_t timer1, timer2;
    msg_t msg;
    msg.type = 0;
    xtimer_set_msg(&timer1, 1000000, &msg, sender_pid);
    xtimer_set_msg(&timer2, 3000000, &msg, sender_pid);

    while(1);

    return 0;
}
```

it might or not crash, but at least it's possible to see only one message is received.
#### Expected results
Both messages are received and there's no crash during the execution.

#### Actual results
Message is not delivered or the node crashes.

#### Versions
Tested on `b-l072z-lrwan1`. It should be board independent though

jia200x, 2018-11-29:
how to proceed?
It should at least be stated in the documentation that's not possible to do this pattern IMO, but also think it's a bug.

kaspar030, 2018-11-29:
> It should at least be stated in the documentation that's not possible to do this pattern IMO, 

Yes!

>but also think it's a bug.

Not really, it is a design decision. The sender_pid field is used as state variable. When a message is only used for one timer, that works just fine. The alternative to using sender_pid is having an extra state variable that's passed to xtimer_set_message().

That's not possible to implement without an API change, and frankly, I don't think it looks nice. I propose updating the documentation...

benpicco, 2019-08-09:
Yes please document this. This behavior is valid, but not obvious and without knowing about the internals, the result is surprising.

miri64, 2020-07-03:
@jia200x @kaspar030 @benpicco Was this ever documented?

maribu, 2022-09-16:
With xtimer on its way out and ztimer being used instead now every, I'd say this is a wontfix.

If anyone disagrees, please reopen.

closed 2022-09-16 by maribu