RIOT-OS/RIOT #10531: ESP-now: ESP_NOW_MAX_SIZE < IPv6 minimum MTU?

miri64, 2018-11-30:
While browsing through the ESP-now code I noticed that the maximum frame size is 250. IPv6 however requires [for a *minimum* Maximum Transition Unit (MTU) of 1280 byte][RFC8200]. If a protocol does not support such frame sizes an adaption layer (like 6LoWPAN) is required. Is there such a thing for ESP-now? Can we want to use 6Lo for it? Or am I reading the code of it wrong?

[RFC8200]: https://tools.ietf.org/html/rfc8200#section-5

miri64, 2018-11-30:
(or do we just live with the fact that ESP-now, a proprietary protocol itself, is not compliant with the IPv6 standard)

miri64, 2018-12-01:
Ok there might be some misunderstanding: *GNRC* supports any frame length (as long as it fits in the packet buffer). The *IPv6 specification* asks for a minimum maximum frame length 1280B. Why it might be not a good idea to have less was discussed in length here https://github.com/RIOT-OS/RIOT/pull/10162. But the main reason I think is well summarized in https://github.com/RIOT-OS/RIOT/pull/10162#issuecomment-441455974.

6LoWPAN (or 6Lo to take into account that it isn't used just for IEEE 802.15.4 aka WPAN anymore) is an adaptation layer between IPv6 and the link layer that allows link layers with lesser frame sizes to send IPv6 anyway (by providing hop-wise fragmentation and header compression).

> If not, we need fragmentation/reassembly functionality in esp-now netdev driver.

Or we could just use 6Lo once #10521 is merged.

miri64, 2018-12-01:
Yes. Maybe also have some search if some preliminary work exists for that... Sadly, since ESP can also be resolved to [**E**ncapsulating **S**ecurity **P**ayload](https://tools.ietf.org/html/rfc4303) (part of IPsec) Googling "IPv6 over ESP-now" is pretty annoying ^^.

gschorcht, 2018-12-01:
I would guess that there is nothing we could reuse. Using ESP-NOW for IPv6 is probably not of interrest for ESP8266 and ESP32 users. Reason is that ESP8266 SDK comes with AP-based WiFi and lwIP stack for IPv4. ESP32 ESP-IDF comes also with AP-based WiFi and lwIP stack, but for IPv4 and IPv6. For meshed networks, Espressif provides ESP-MESH, at least for ESP32. But I think most ESP users are happy with AP-based WiFi and IPv4.

The problem is, that all ESP WiFi libraries are proprietary and completely undocumented. Until now, I was only able to get ESP-NOW working. That's the reason why I only provided a driver for ESP-NOW. Although, you will also find a netdev driver for AP-based WiFi in `cpu/esp32/esp-wifi`, it is experimental. It gets a global unicast address with 2000::/3 prefix, provided that there is a IPv6 capable router in the network, but it disconnects after 1 minute for any reason. It is on my ToDo list, to investigate the problem. I was hoping to find a student who would support me. Unfortunately, I have not been lucky so far.

TimoRoth, 2018-12-07:
I've been looking at the issue for a bit now, and with the current 6Lo changes, it should be possible to use 6Lo over ESP-NOW.
The way I'd try to work on this would be by changing the esp-now netdev/netif code to something similar to that of the cc110x driver. It would not use a raw netif anymore, but its own one, which can then get the l2 dst address from the gnrc_netif_hdr_t instead of depending on being able to decode the IPv6 packet being transported.
Is that the right way to get this working? If so, I'll try to modify the esp32 esp-now module according to that idea.

gschorcht, 2018-12-07:
@TimoRoth Thanks for investigating it. I would be very happy if you could contribute a working approach. But @miri64 should make a statement whether this would be the right way. I would rely on her statement.

miri64, 2018-12-10:
What @TimoRoth is saying is correct. The `gnrc_netif_raw` implementation was intended for link-layers without any addresses (e.g. SLIP) or were access to the link-layer address is not necessary. If the link-layer handling can't be done by one of the others (currenlty `gnrc_netif_ethernet` and `gnrc_netif_ieee802154`) a custom implementation needs to be provided (as done by @TimoRoth in #10581)

PR #10581 (2018-12-29): esp_now: avoid parsing IPv6 header in netdev driver
### Contribution description

Due to the very low MTU of ESP-NOW, it cannot use normal IPv6, which needs a way larger minimum packet size(See #10531).
These patches make ESP-NOW capable of being used with 6Lo instead. In turn they get rid of IPv6 header parsing inside of the netdev driver, which seemed very out of place there.
It provides a custom netif driver instead, that gets the regular l2 destination address from the netif header.

### Testing procedure

Two esp32 boards with the gnrc_networking should still be able to communicate, altho [... cut]
closed 2018-12-29 by miri64