RIOT-OS/RIOT #10547: Makefile/flasher: is `preflash` still relevant?

jcarrano, 2018-12-05:
#### Description

`preflash` is a command that is (sometimes) run before flashing. Now, if it is _always_ run before flashing, then it makes more sense to make it part of the flashing procedure itself.

For the sake of simplicity, I would like to remove this target. The only places where it is used in the ESP32 port (which I'm working to fix with @cladmi ) and in the bossa tool.

In the latter, it is used for setting the TTY mode, but that is better done in the flash script itself.

In the ESP32 port, the presence of PREFLASH seems to have introduced more confusion and led to a sub-optimal makefile.

kaspar030, 2018-12-05:
+1 for removing once not used anymore

cladmi, 2018-12-05:
For the way it is used, having `preflash` a separate target is really misleading. Because the `stty` thing requires it to be done a few seconds before `flashing` which is nothing that should be guaranteed by `make`.

However, for some cases, if the `flasher` is an existing binary, I would not want to wrap it as a shell script to duplicate both its argument parsing and the preflash arguments too, to be able to do a `preflash` procedure.
For that case, I would see it more as a command executed within the `flash` target.

jcarrano, 2018-12-05:
After doing #10548 I think this may be handled by having a canned recipe that does the flashing and allowing this recipe to be changed.

jcarrano, 2018-12-06:
> allowing this recipe to be changed.

But for that we need newer Make than 3.8, see #10554 .

closed 2019-09-10 by stale[bot]