RIOT-OS/RIOT #10559: I2C API write_regs does not fit implmentation

MrKevinWeiss, 2018-12-06:
#### Description
The API says that it is a repeated start write but it is a continuous write (writes 1 or 2 bytes with I2C_NOSTOP then the length I2C_NOSTART).

#### Steps to reproduce the issue
Read the documentation in drivers/include/periph/i2c.h and the generic implementation on drivers/periph_common/i2c.c.  You could also scope it but that is more difficult than reading a bit.

#### Expected results
To select either full write a continuous read or send a repeated start.

#### Actual results
Differs in code documentation

#### Versions
N/A

MrKevinWeiss, 2018-12-06:
@dylad @haukepetersen @aabadie 
It would be nice to get some feedback.

It is very hard to support the API with all hardware since some hardware requires knowledge of the full frame (stm i2c_1.c) in order to work or actions before the final byte is sent (stm i2c_2.c).  The issue I am dealing with now is that I can either send write_bytes and expect a repeated start or stop (ie. RELOAD = 0) or I can expect more bytes to be send with a no start (ie. RELOAD = 1) but if we want to allow splitting up the frame we just can't support both.

Therefore I think that if write_regs supports repeated starts then the write_bytes should support continuous writes (ie RELOAD = 1) and would not allow repeated starts on the low level (or vice versa).

The other suggestion from @jia200x is to use something to build a set of commands and only execute at the end, I don't know how this will effect the API and binary size.

dylad, 2018-12-06:
@MrKevinWeiss Indeed, the doc does not match the implementation.

> Therefore I think that if write_regs supports repeated starts then the write_bytes should support continuous writes (ie RELOAD = 1) and would not allow repeated starts on the low level (or vice versa).

Well some I2C IP doesn't support continuous write and we're force to use repeated start (i.e NRF5x).

IMHO, I think we should just agree on continuous writes or repeated start for the common driver. If a specific hardware cannot use this implementation then we fallback to a specific implementation by removing `PERIPH_I2C_NEED_X_Y` stuff.

MrKevinWeiss, 2018-12-06:
lis2dh, bmp180, isl29125, tmp006 sensors all use continuous read.
Those were just the first ones that I saw.  I cannot find a repeated start write yet and seeing as how that has been the previous operation maybe it would be good to simply update the documentation to support that.

> Well some I2C IP doesn't support continuous write and we're force to use repeated start (i.e NRF5x).

Ya it looks like a workaround was made to prevent repeated start.

MrKevinWeiss, 2018-12-06:
Can we agree that we provide:
* repeated start read register, no support for NOSTART and NOSTOP
* continuous write register, no support for NOSTART and NOSTOP
* write bytes will try to allow repeated start over frame splitting
_for example sending write_bytes+NOSTOP, write_bytes+NOSTOP|NOSTART, write_bytes+NOSTOP|NOSTART, ..., write_bytes+NOSTART should be the first to drop support if possible_
**The preferred operation to support would be write_bytes+NOSTOP, write_bytes (with start and stop)**
* read bytes will allow frame splitting over repeated starts since it is already supported in the read regs

MrKevinWeiss, 2018-12-06:
Also I would like clarification on some error codes,
 `-ENXIO When no devices respond on the address sent on the bus` should only be if the device nacked not the data
`-EIO When slave device doesn't ACK the byte` only when a data nack occurs
`-EAGAIN When a lost bus arbitration occurs` For arbitration loss and bus errors too

Can we use EINVAL if a nostart is sent without the bus being busy?

MrKevinWeiss, 2018-12-06:
AND... when releasing the bus some checks for if it is busy occurs...  They don't really do anything.  Should we not release if the bus is still busy (return -1) or should we just continue to try to release (and in some cases powerdown), in which case there shouldn't be a check at all because it doesn't make a difference.

MrKevinWeiss, 2018-12-06:
AAAAANNNNDDDD if errors such as arbitration loss or incorrect state when trying to start occur should the master try to send a stop?

MrKevinWeiss, 2019-05-28:
Fixed with #10578

closed 2019-05-28 by MrKevinWeiss