RIOT-OS/RIOT #10566: [CTF-tracker] List of identified compile time configurations

jia200x, 2018-12-06:
# Description

This issue is a tracker of the compile time configurations.
After some offline discussions with some RIOT maintainers we came to the conclusion it¡s a good idea to first **identify these configurations and tag them with a Doxygen `@configurations` group**. This would give a huge kick start to further work related to compile time configurations.

Here's a list of identified configurations by @miri64 and me. We used lots of `grep` and scripts, so we politely request to RIOT kernel developers to:

1. Help identifying which configurations doesn't make sense here (there are some macros that are not configurations, or some others that doesn't make sense to be exposed to the user). Also, add missing configurations
2. (WIP) When the `@configurations` label is defined, use this tracker to add it to all configurations.

# Identified configurations

## boards
- [ ] **boards/atmega**
    - [ ] [CPU_ATMEGA_CLK_SCALE_INIT](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/boards/common/atmega/board.c#L27) (boards/common/atmega/board.c:27)

## cpu
- [ ] **cpu/cc26x0**
    - [ ] [HF_CLOCK_SOURCE](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/cc26x0/cpu.c#L25) (cpu/cc26x0/cpu.c:25)
    - [ ] [LF_CLOCK_SOURCE](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/cc26x0/cpu.c#L28) (cpu/cc26x0/cpu.c:28)

- [ ] **cpu/cortexm_common**
    - [ ] [SRAM_BASE](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/cortexm_common/vectors_cortexm.c#L39) (cpu/cortexm_common/vectors_cortexm.c:39)

- [ ] **cpu/efm32**
    - [ ] [CMU_HFXOINIT](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/efm32/cpu.c#L34) (cpu/efm32/cpu.c:34)
    - [ ] [CMU_LFXOINIT](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/efm32/cpu.c#L37) (cpu/efm32/cpu.c:37)
    - [ ] [EMU_DCDCINIT](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/efm32/cpu.c#L44) (cpu/efm32/cpu.c:44)
    - [ ] [EMU_EM23INIT](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/efm32/cpu.c#L47) (cpu/efm32/cpu.c:47)
    - [ ] [EMU_EM4INIT](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/efm32/cpu.c#L50) (cpu/efm32/cpu.c:50)

- [ ] **cpu/esp32**
    - [ ] [CAN_TX](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/esp32/esp-can/esp_can.c#L46) (cpu/esp32/esp-can/esp_can.c:46)
    - [ ] [CAN_RX](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/esp32/esp-can/esp_can.c#L49) (cpu/esp32/esp-can/esp_can.c:49)
    - [ ] [CAN_CLK_OUT_DIV](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/esp32/esp-can/esp_can.c#L53) (cpu/esp32/esp-can/esp_can.c:53)
    - [ ] [CAN_DEFAULT_BITRATE](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/esp32/esp-can/esp_can.c#L58) (cpu/esp32/esp-can/esp_can.c:58)
    - [ ] [ESP_CAN_DEV_BASE_PRIORITY](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/esp32/esp-can/esp_can.c#L66) (cpu/esp32/esp-can/esp_can.c:66)
    - [ ] [CONFIG_WIFI_STA_SSID](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/esp32/esp-now/esp_now_netdev.c#L303) (cpu/esp32/esp-now/esp_now_netdev.c:303)
    - [ ] [CONFIG_WIFI_STA_PASSWORD](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/esp32/esp-now/esp_now_netdev.c#L306) (cpu/esp32/esp-now/esp_now_netdev.c:306)
    - [ ] [CONFIG_WIFI_STA_CHANNEL](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/esp32/esp-now/esp_now_netdev.c#L309) (cpu/esp32/esp-now/esp_now_netdev.c:309)
    - [ ] [CONFIG_WIFI_STA_SSID](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/esp32/esp-wifi/esp_wifi_netdev.c#L126) (cpu/esp32/esp-wifi/esp_wifi_netdev.c:126)
    - [ ] [CONFIG_WIFI_STA_PASSWORD](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/esp32/esp-wifi/esp_wifi_netdev.c#L129) (cpu/esp32/esp-wifi/esp_wifi_netdev.c:129)
    - [ ] [CONFIG_WIFI_STA_CHANNEL](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/esp32/esp-wifi/esp_wifi_netdev.c#L132) (cpu/esp32/esp-wifi/esp_wifi_netdev.c:132)
    - [ ] [MAX](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/esp32/include/esp_common.h#L100) (cpu/esp32/include/esp_common.h:100)
    - [ ] [USEC_PER_MSEC](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/esp32/include/esp_common.h#L115) (cpu/esp32/include/esp_common.h:115)
    - [ ] [MSEC_PER_SEC](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/esp32/include/esp_common.h#L119) (cpu/esp32/include/esp_common.h:119)
    - [ ] [ICACHE_RAM_ATTR](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/esp32/include/esp_common.h#L47) (cpu/esp32/include/esp_common.h:47)
    - [ ] [MIN](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/esp32/include/esp_common.h#L95) (cpu/esp32/include/esp_common.h:95)
    - [ ] [configASSERT](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/esp32/include/freertos/FreeRTOS.h#L25) (cpu/esp32/include/freertos/FreeRTOS.h:25)
    - [ ] [CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/esp32/include/sdk_conf.h#L37) (cpu/esp32/include/sdk_conf.h:37)
    - [ ] [CONFIG_CONSOLE_UART_BAUDRATE](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/esp32/include/sdk_conf.h#L49) (cpu/esp32/include/sdk_conf.h:49)
    - [ ] [CONFIG_LOG_DEFAULT_LEVEL](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/esp32/include/sdk_conf.h#L56) (cpu/esp32/include/sdk_conf.h:56)
    - [ ] [CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/esp32/include/sdk_conf.h#L66) (cpu/esp32/include/sdk_conf.h:66)
    - [ ] [CONFIG_ESP32_XTAL_FREQ](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/esp32/include/sdk_conf.h#L76) (cpu/esp32/include/sdk_conf.h:76)
    - [ ] [P_tmpdir](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/esp32/include/stdio.h#L142) (cpu/esp32/include/stdio.h:142)
    - [ ] [SEEK_SET](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/esp32/include/stdio.h#L146) (cpu/esp32/include/stdio.h:146)
    - [ ] [SEEK_CUR](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/esp32/include/stdio.h#L149) (cpu/esp32/include/stdio.h:149)
    - [ ] [SEEK_END](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/esp32/include/stdio.h#L152) (cpu/esp32/include/stdio.h:152)

- [ ] **cpu/esp8266**
    - [ ] [ICACHE_RAM_ATTR](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/esp8266/include/common.h#L44) (cpu/esp8266/include/common.h:44)
    - [ ] [ESP8266_CPU_FREQUENCY](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/esp8266/include/user_config.h#L28) (cpu/esp8266/include/user_config.h:28)

- [ ] **cpu/native**
    - [ ] [ASYNC_READ_NUMOF](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/native/include/async_read.h#L29) (cpu/native/include/async_read.h:29)
    - [ ] [TRACE_SIZE](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/native/include/trace.h#L35) (cpu/native/include/trace.h:35)

- [ ] **cpu/nrf5x_common**
    - [ ] [NRFMIN_PAYLOAD_MAX](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/nrf5x_common/include/nrfmin.h#L104) (cpu/nrf5x_common/include/nrfmin.h:104)
    - [ ] [NRFMIN_GNRC_THREAD_PRIO](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/nrf5x_common/radio/nrfmin/nrfmin_gnrc.c#L35) (cpu/nrf5x_common/radio/nrfmin/nrfmin_gnrc.c:35)
[... 461 lines cut ...]
    - [ ] [LORAMAC_DEFAULT_ADR](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/sys/include/net/loramac.h#L163) (sys/include/net/loramac.h:163)
[... cut ...]
    - [ ] [SKALD_INTERVAL](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/sys/include/net/skald.h#L61) (sys/include/net/skald.h:61)
    - [ ] [SKALD_ADV_CHAN](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/sys/include/net/skald.h#L68) (sys/include/net/skald.h:68)

- [ ] **sys/net**
    - [ ] [SOCK_MBOX_SIZE](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/sys/net/gnrc/sock/include/sock_types.h#L40) (sys/net/gnrc/sock/include/sock_types.h:40)

miri64, 2018-12-06:
> *
> * [ ]  **sys/posix**
>
>   * [ ]  [INADDRSZ](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/sys/posix/include/arpa/inet.h#L38) (sys/posix/include/arpa/inet.h:38)
>   * [...]

I think none of those should be configurable, in the sense we aim to in this issue (exposing them via a `config` doxygen group):
- `INADDRSZ` is always 4 byte. I think the `#ifndef` exists here in case it is provided by the libc.
- Dito for `IN6ADDRSZ` (which is always 16 byte).
- `HAVE_MALLOC_H` typically is provided by the build system, so it can rather be seen as a module than a configuration value.

miri64, 2018-12-06:
> *
> * [ ]  **cpu/cortexm_common**
>   
>   * [ ]  [SRAM_BASE](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/cpu/cortexm_common/vectors_cortexm.c#L39) (cpu/cortexm_common/vectors_cortexm.c:39)

Those are defined unconditionally in the vendor files ;-)

miri64, 2018-12-06:
(I guess we find occurences of the in many macros in `cpu/` ;-))

jia200x, 2018-12-17:
(updated with #10617 )

jia200x, 2018-12-17:
I wonder why GNRC_IPV6_STATIC_LLADDR was not listed here :O

jia200x, 2018-12-18:
added LORA configurations to this list. They were not parse by the script

jia200x, 2018-12-18:
@RIOT-OS/maintainers the `@config` group was already added in #10571. We politely ask request for your help on:
1. Add missing configurations to this tracker
2. Strike configurations that doesn't make sense
3. Pick one or more groups of these tracker and open PRs to add those configuration macros to the `@config` group. You can base e.g on #10617.

Thanks you all! ;)

miri64, 2018-12-18:
> I wonder why GNRC_IPV6_STATIC_LLADDR was not listed here :O

Because it is not `#ifdef`'d but only included over `defined(DOXYGEN)` (so there is no default value).

jia200x, 2018-12-18:
> Because it is not #ifdef'd but only included over defined(DOXYGEN) (so there is no default value).

I see

miri64, 2018-12-18:
Also please: if you introduce new configuration macros, we would be very happy if you group them under the new `config` group ;-).

jia200x, 2018-12-18:
should `sys/auto_init` be configurable? Messing with those PRIOs can break several modules I guess

miri64, 2018-12-18:
A person configuring that should know what their doing, but it should be configurable ;-).

kb2ma, 2018-12-28:
Reviewed gcoap section -- looks good! Added sections for coap and nanocoap in the list above. The items I added are not set up presently for external override (they are just #define, not #ifndef/#define/#endif). However, I'll do that in the PR where I add the config group tags.

jia200x, 2019-01-02:
thanks @kb2ma for the review!

jia200x, 2019-01-02:
there are some options missing in loramac. I will add them in a follow-up PR. Unchecked the `loramac` checkbox to reflect this ;)

miri64, 2019-01-03:
> * [ ]  **sys/net/netif**
>   
>   * [ ]  [NETIF_NAMELENMAX](https://github.com/RIOT-OS/RIOT/blob/050e06f20f845d57eaa4fed9e6077f57de1a1cb4/sys/include/net/netif.h#L45) (sys/include/net/netif.h:45)

Not sure this sure this should be configurable. In the end, the one supposed to provide the (human readable) name is the network stack, so increasing or decreasing the length of this makes no sense (or at least implementations very complicated.

kb2ma, 2019-01-03:
Thanks for shepherding the CoAP PR's for this, @miri64!

gschorcht, 2019-01-09:
When I started to add the compile time configurations of my CPU ports and drivers to the doc group, I was wondering whether it makes sense to structure the `config` group, for example similar to the existing structure `boards`, `cpu`, `drivers`, `kernel` ... Try to imagine if all modules configs are arranged directly below the top entry. The list will grow up to something that is not readable anymore.

miri64, 2019-04-17:
Continued a bit further with #11415.

fjmolinas, 2019-05-21:
@jia200x @miri64 Should I edit your original message and post #11561 or just post it in a comment?