RIOT-OS/RIOT #10572: Systemtime xtimer or alternative

fesselk, 2018-12-07:
I am currently searching for some kind of system time that can be or is all ready fast forwarded after a deep sleep (maybe there is some thing like this).
xtimer seems to be used as a system timer but it is stopped in deep sleep.

Background: LoRa ( other radios as well) needs to track their usage since xtimer is stopped in deep sleep the usage is wrong

jcarrano, 2018-12-07:
Which is the precision needed? Is a rtc-like timer running at 32768 Hz (30µs period) enough?

fesselk, 2018-12-07:
i think i would but i am not sure if every board has an rtc (i think some arduinos have non) ( or is there a mapped rtc which can run via a real rtc or a xtimer)

jcarrano, 2018-12-07:
There is [RTT](http://doc.riot-os.org/group__drivers__periph__rtt.html#details).

When I said RTC-like, I did not mean and RTC only, but rather a low power / low frequency counter that can keep ticking in low-power mode. I think RTT should fill that role. The problems are that:

1. periph_rtt is a low-level interface.
2. It is not guaranteed that it will keep running in deep sleep (and probably does not in many boards).

Also, I should mention that some CPUs may require some sort of external HW (like a 32k crystal) for this sort of timers to work. For example the TM4C (aka LM4F) family is such an example.

If xtimer was powered by one such timer, then your problem could be solved. Currently, that's hard because xtimer needs a 1MHz source:

1. The kind of HW timers that run in sleep modes do not run so fast.
2. Even if they could, overflows would be very frequent, the CPU would have to wake up to handle them and it would defeat the purpose of sleeping to save power.

Correct me if I'm wrong, but I understand that this is an issue because LoRa or other radios active during sleep periods, right?

vac, 2019-03-30:
`xtimer` is a nice "tool" that can control multiple processes in an easy, readable way.
From my experience on samd21/samr21 it is possible to:
* switch xtimer to 32768Hz,
* make xtimer work in deep sleep, properly.

But... while I was using different OS and board, I think it is impossible to create one, generic timer solution that will work perfectly on every hardware.
**You should make a decision after you will know your hardware capabilities.**
There is always like balancing between timer resolution and power consumption.

I.e. while `xtimer` is somehow a great solution it is not very reliable on samd21 as reading it's value takes a relatively very long time (on this particular platform). Note that xtimer reads timer value, internally.

Not quite sure about it but I think I've seen that `GoMacH protocol` use `RTT` for superframes synchronization etc (you may take a look at its sources:  sys\net\gnrc\link_layer\gomach\gomach.c). As RTT is much simpler it is easier to make not complex, but reliable solution on it.

closed 2019-11-01 by stale[bot]