RIOT-OS/RIOT #10575: Free RISC-V Development boards

Citrullin, 2018-12-08:
Now also available outside of the USA: 
https://twitter.com/vegaboardriscv/status/1070765378064633856

The order form is available here:
http://open-isa.org/order/

Zephyr already supports RISC-V processors and even the vega board:
https://twitter.com/ZephyrIoT/status/1070393626709159942

gebart, 2018-12-08:
Interesting board.
It looks like #7107 might work with this SoC too. I hope I manage to get hold of one of these boards to do some experiments on.

aabadie, 2018-12-08:
> I hope I manage to get hold of one of these boards to do some experiments on.

+1 here. Looking at the specs, this board is a perfect candidate for joining the other boards in IoT-LAB (Saclay site)

gebart, 2018-12-08:
@aabadie I was about to @ mention you to ask if you were going to have some of these in the room where all the mixed vendor boards are ;)

dylad, 2018-12-08:
Wow free boards for real ? Seems really interesting to me !

Josar, 2018-12-18:
seems to be updated now for outside US https://open-isa.org/order/outsideus/

gebart, 2018-12-21:
I propose we update this issue into a tracker for the porting effort for the Vega board.

Citrullin, 2018-12-21:
Maybe [this](https://github.com/zephyrproject-rtos/zephyr/pull/11853) is helpful for the implementation

gebart, 2018-12-21:
I started outlining a few tasks for the Vega port in the top post

gebart, 2018-12-21:
My approach to this will be to attempt to get the Cortex-M4 core to run RIOT first, since that should be very similar to simply adding one more Kinetis CPU variant. I believe this MCU may possibly be a close relative to the not yet released K32W0x with extra RISC-V cores.

advancedwebdeveloper, 2019-12-05:
I have the evaluation board, too. I see that mingw examples is a nightmare - no support for Powershell. I wonder if MSVC's cross-compilation would allow to compile Cortex-M4 targeted examples.

dylad, 2019-12-05:
@advancedwebdeveloper 
For Windows, if you just want to cross-compile your examples maybe WSL or WSL2 is a good candidate. But I have no out of the box solution for the flashing part.

advancedwebdeveloper, 2019-12-11:
@dylad , I am on Windows 10 32bit. I do have a Windows 10 64bit machine, nearby - but currently concentrating on 32bit OS.
And nobody answered about Microsoft's support for ARM compiler (MSVC's backend). I heard that it supports 64bit ARM (aarch64).
I thought I could rely more on Powershell scripts. Anyone interested to bootstrap that?
I have created a repo https://github.com/advancedwebdeveloper/rv32m1_sdk_riscv - need to change the license config. We could concentrate our efforts there

dylad, 2019-12-11:
IIRC, you can setup custom toolchain in Visual Studio, so you can download the ARM toolchain for x86 (32bits) [here](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads) and then setup your toolchain in y project configuration.
Regarding powershell scripts I cannot help you unfortunately.

dylad, 2019-12-11:
(hit the wrong button sorry...)

advancedwebdeveloper, 2019-12-23:
I suspect that SDKs for https://docs.microsoft.com/en-us/previous-versions/windows/embedded/jj919345%28v%3dwinembedded.80%29 and https://developer.microsoft.com/en-us/windows/iot could be able to offer compilers, for Cortex M4 processor of Vega board.

closed 2026-04-05 by crasbe