RIOT-OS/RIOT #10577: RFC: Define `PSEUDOMODULES` in `Makefile.include` of drivers

gschorcht, 2018-12-09:
## Description

Drivers often define variants of the supported devices as pseudo-modules. To define these pseudo-modules, file `$ (RIOTBASE)/makefiles/pseudomodules.mk` has to be extended at the moment. This seems to be a sub-optimal solution for the following reasons:

1. The `pseudomodules.mk` file grows more and more and becomes confusing.
2. The number of defined psseudo-modules increases even if the corresponding driver modules are not used.
2. The `$(RIOTBASE)/makefiles` directory does not seem to be the right place for driver stuff.

## Quesion

 Would it be reasonable and of course possible to include a `$RIOTBASE/drivers/<driver>/Makefile.include` in which the pseudomodules a driver supports are defined?

miri64, 2018-12-10:
I think the idea is to even go father so that every module is able to define its own pseudo-(sub-)modules.

jcarrano, 2018-12-10:
Yes and Yes.

One of the goals for the build system reworks I'm doing with @cladmi is to slowly move to local declarations only. This means no stuff in sys/Makefile.*, drivers/Makefile.*, etc.

For submodules, I had a brief discussion offline where I propose that we change from `module_submodule` to `module/submodule`. In the "easy cases" where `submodule` is a file (or directory) under `module` this would obviate the need for stuff like:
```
ifneq (,$(filter module_submodule,$(USEMODULE))
   DIRS+=submodule
endif
```

gschorcht, 2018-12-10:
> One of the goals for the build system reworks I'm doing with @cladmi is to slowly move to local declarations only. This means no stuff in sys/Makefile._, drivers/Makefile._, etc.

This would be really great.

gschorcht, 2019-08-10:
@jcarrano @cladmi Can we close the issue. Will it become part of the changes of the build system?

gschorcht, 2020-02-11:
@aabadie @fjmolinas According to `CODEOWNERS.md` in PR #13289 it seems that you are responsible for the make build system :wink: Is there something like that discussed in this issue on the ToDo-List?

fjmolinas, 2021-01-14:
> @aabadie @fjmolinas According to `CODEOWNERS.md` in PR #13289 it seems that you are responsible for the make build system wink Is there something like that discussed in this issue on the ToDo-List?

I'm not really the build system maintainer, but that being said declaring `PSEUDOMODULES` in `$RIOTBASE/drivers/<driver>/Makefile.include` as long as `PSEUDOMODULES` are declarative there would be no issue with that approach. Its definitively possible.

> Would it be reasonable and of course possible to include a $RIOTBASE/drivers/<driver>/Makefile.include in which the pseudomodules a driver supports are defined?

So Yes, possible, and yes reasonable as long as kept declarative.