RIOT-OS/RIOT #10598: edbg: not compiling on OS X

kYc0o, 2018-12-12:
#### Description

The `edbg` tool for flashing atmel devices (in my case samr21-xpro) doesn't compile anymore in my OS X machine. It's probably due to the merging of #10570.
#### Steps to reproduce the issue

Just try to flash a samr21-xpro (or any other edbg device) on OS X. It fails during the compile time.
#### Expected results

`edbg` is compiled and flashes the targets.
#### Actual results

```console
[INFO] edbg binary not found - building it from source now
CC= CFLAGS= make -C /Users/facosta/git/RIOT-OS/RIOT/dist/tools/edbg
rm -Rf /Users/facosta/git/RIOT-OS/RIOT/dist/tools/edbg/bin
mkdir -p /Users/facosta/git/RIOT-OS/RIOT/dist/tools/edbg/bin
/Users/facosta/git/RIOT-OS/RIOT/dist/tools/git/git-cache clone "https://github.com/ataradov/edbg" "807d948cc8a664ade3e9446b4937aa54268afcb2" "/Users/facosta/git/RIOT-OS/RIOT/dist/tools/edbg/bin"
Cloning into '/Users/facosta/git/RIOT-OS/RIOT/dist/tools/edbg/bin'...
.
.
.
make[3]: Entering directory '/Users/facosta/git/RIOT-OS/RIOT/dist/tools/edbg/bin/hidapi/mac'
  CC       hid.lo
  CCLD     libhidapi.la
clang: warning: argument unused during compilation: '-pthread'
clang: warning: argument unused during compilation: '-pthread'
ar: `u' modifier ignored since `D' is the default (see `U')
make[3]: Leaving directory '/Users/facosta/git/RIOT-OS/RIOT/dist/tools/edbg/bin/hidapi/mac'
Making all in hidtest
make[3]: Entering directory '/Users/facosta/git/RIOT-OS/RIOT/dist/tools/edbg/bin/hidapi/hidtest'
  CXX      hidtest.o
  CXXLD    hidtest
clang: warning: argument unused during compilation: '-pthread'
make[3]: Leaving directory '/Users/facosta/git/RIOT-OS/RIOT/dist/tools/edbg/bin/hidapi/hidtest'
make[3]: Entering directory '/Users/facosta/git/RIOT-OS/RIOT/dist/tools/edbg/bin/hidapi'
make[3]: Nothing to be done for 'all-am'.
make[3]: Leaving directory '/Users/facosta/git/RIOT-OS/RIOT/dist/tools/edbg/bin/hidapi'
make[2]: Leaving directory '/Users/facosta/git/RIOT-OS/RIOT/dist/tools/edbg/bin/hidapi'
make[1]: Leaving directory '/Users/facosta/git/RIOT-OS/RIOT/dist/tools/edbg/bin/hidapi'
gcc -Ihidapi/hidapi -W -Wall -Wextra -O2 -std=gnu11 dap.c edbg.c target.c target_atmel_cm0p.c target_atmel_cm3.c target_atmel_cm4.c target_atmel_cm7.c target_atmel_cm4v2.c target_mchp_cm23.c  dbg_mac.c hidapi/mac/.libs/libhidapi.a -framework IOKit -framework CoreFoundation -o edbg
target_atmel_cm0p.c:113:7: warning: missing field 'name' initializer [-Wmissing-field-initializers]
  { 0 },
      ^
1 warning generated.
target_atmel_cm4v2.c:99:7: warning: missing field 'name' initializer [-Wmissing-field-initializers]
  { 0 },
      ^
1 warning generated.
target_mchp_cm23.c:134:7: warning: missing field 'name' initializer [-Wmissing-field-initializers]
  { 0 },
      ^
1 warning generated.
ld: warning: ignoring file hidapi/mac/.libs/libhidapi.a, file was built for archive which is not the architecture being linked (x86_64): hidapi/mac/.libs/libhidapi.a
Undefined symbols for architecture x86_64:
  "_hid_close", referenced from:
      _dbg_close in dbg_mac-c17578.o
  "_hid_enumerate", referenced from:
      _dbg_enumerate in dbg_mac-c17578.o
  "_hid_error", referenced from:
      _dbg_dap_cmd in dbg_mac-c17578.o
  "_hid_free_enumeration", referenced from:
      _dbg_enumerate in dbg_mac-c17578.o
  "_hid_init", referenced from:
      _dbg_enumerate in dbg_mac-c17578.o
  "_hid_open", referenced from:
      _dbg_open in dbg_mac-c17578.o
  "_hid_read", referenced from:
      _dbg_dap_cmd in dbg_mac-c17578.o
  "_hid_write", referenced from:
      _dbg_dap_cmd in dbg_mac-c17578.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [Makefile:46: edbg] Error 1
make: Leaving directory '/Users/facosta/git/RIOT-OS/RIOT/dist/tools/edbg/bin'
make[1]: *** [Makefile:13: all] Error 2
make: *** [/Users/facosta/git/RIOT-OS/RIOT/makefiles/tools/targets.inc.mk:17: /Users/facosta/git/RIOT-OS/RIOT/dist/tools/edbg/edbg] Error 2
```
#### Versions

```
Operating System Environment
-----------------------------
       Operating System: Mac OS X 10.11.6
                 Kernel: Darwin 15.6.0 x86_64 i386

Installed compiler toolchains
-----------------------------
             native gcc: Apple LLVM version 8.0.0 (clang-800.0.42.1)
      arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
                avr-gcc: avr-gcc (GCC) 8.2.0
       mips-mti-elf-gcc: missing
             msp430-gcc: msp430-gcc (GCC) 4.7.0 20120322 (mspgcc dev 20120911)
   riscv-none-embed-gcc: missing
                  clang: Apple LLVM version 8.0.0 (clang-800.0.42.1)

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "3.0.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
               avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                  cmake: cmake version 3.11.4
               cppcheck: Cppcheck 1.85
                doxygen: missing
                 flake8: missing
                    git: git version 2.19.2
                   make: GNU Make 4.2.1
                openocd: Open On-Chip Debugger 0.10.0+dev-00290-g5a98ff78 (2018-01-30-19:53)
                 python: Python 2.7.10
                python2: missing
                python3: Python 3.7.1
             coccinelle: spatch version 1.0.6 compiled with OCaml version 4.05.0
```

jcarrano, 2018-12-12:
RIOT is using the latest edbg master. Please report the bug to edbg upstream too.

kYc0o, 2018-12-12:
Actually I'm investigating further and there's maybe something related to my local setup. I'll report later the results.

360disrupt, 2018-12-14:
@kYc0o i#m having the same problem I first reinstalled those packages and then uninstalled them but I still get the same err.

Board: `samr21-xpro`
Mac OS 10.14.1 (Mojave)

```
39748	    508	   6044	  46300	   b4dc	/Users/360/Projekte/1-NODEJS/RIOT/examples/default/bin/samr21-xpro/default.elf
[INFO] edbg binary not found - building it from source now
CC= CFLAGS= make -C /Users/360/Projekte/1-NODEJS/RIOT/dist/tools/edbg
env -i PATH=/Users/360/.nvm/versions/node/v11.3.0/bin:/Library/Internet Plug-Ins/JavaAppletPlugin.plugin/Contents/Home/bin:/usr/local/gcc_arm/gcc-arm-none-eabi-6-2017-q2-update/bin/:/usr/local/bin/go:/usr/local/bin/subl:/usr/local/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/usr/local/go/bin TERM=xterm-256color "/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/360/Projekte/1-NODEJS/RIOT/dist/tools/edbg/bin
env: Plug-Ins/JavaAppletPlugin.plugin/Contents/Home/bin:/usr/local/gcc_arm/gcc-arm-none-eabi-6-2017-q2-update/bin/:/usr/local/bin/go:/usr/local/bin/subl:/usr/local/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/usr/local/go/bin: No such file or directory
make[1]: *** [all] Error 127
make: *** [/Users/360/Projekte/1-NODEJS/RIOT/dist/tools/edbg/edbg] Error 2
MasterG:default MasterG$ 
```

360disrupt, 2018-12-17:
My current solution:

- Dl EDBG from `git clone https://github.com/ataradov/edbg && cd edbg`
- `make all`
- export EDBG (`export EDBG=pathToBin` in my `.bash_profile`) or set when compiling with riot: `make flash BOARD=... EDBG=...`