RIOT-OS/RIOT #10614: DEBUG stops shell if used in i2c init

MrKevinWeiss, 2018-12-14:
#### Description
For stm32f4 and stm32f1 based cpu, having debug enabled and writing something in the init seems to prevent from interacting.

#### Steps to reproduce the issue
Enable debug and print in the i2c init of a stm32f4 or f1 board.

#### Expected results
See the output of the debug print
#### Actual results
Nothing gets printed and trying to write to the shell yields no result.

Mentioned in #10608

aabadie, 2018-12-14:
The problem is that stdio_init is called by the newlib intialization [here](https://github.com/RIOT-OS/RIOT/blob/master/sys/newlib_syscalls_default/syscalls.c#L72) and this is done after `board_init` [here](https://github.com/RIOT-OS/RIOT/blob/master/cpu/cortexm_common/vectors_cortexm.c#L129). And all peripherals are initialized in `board_init` by a call to periph_init.

For testing purpose, I think I changed that to call stdio_init at the beginning of periph_init and it was working. But this may have other undesirable effects.

MrKevinWeiss, 2018-12-14:
Sounds like that could be it.  You want to fix it or shall I?

aabadie, 2018-12-14:
> You want to fix it or shall I?

Please go ahead :)

MrKevinWeiss, 2018-12-14:
The nucleo-f303re doesn't print the init but at least doesn't lockup.  Would it be sufficient to attempt to prevent lockup? 

I don't want to be responsible for "undesirable effects" and don't think I have the understanding to ensure nothing else gets effected.

aabadie, 2018-12-14:
Maybe @gebart, @haukepetersen @kaspar030 will have a good opinion on that problem ?

gebart, 2018-12-14:
I do not have much knowledge of the STM32 periph drivers, but some general thoughts regarding the uart initialization:
- It should not be an error to call uart_init again after it has been called once. Especially since we do not have an API for changing the RX callback after initialization.
- Having a check in uart_write to prevent hard faults if called before init is a good safety net. For these situations where you would like to have some debug output but some debug calls are made before stdio is initialized causing a crash. (this is what #10615 basically does)

kaspar030, 2018-12-14:
> And all peripherals are initialized in `board_init` by a call to periph_init. So in the end, you have all peripherals (I2C, etc) but UART initialized before stdio.

We should maybe move periph_init into auto_init, or into main_trampoline before the auto_init call.

PR #10615 (2018-12-19): cpu/stm32_common/uart: Prevent uart from sending if not initialized
### Contribution description
Due to the stdio getting called after periph_init the uart may send before initialized.
This adds a simple check so the uart does not get into a locked-up state.

### Testing procedure

Use an stm32F1, F2, F4, or L0
Enable debug in tests/periph_i2c/main.c
Write a DEBUG message in the init.
`BOARD=<selected board from above> make flash term -C tests/periph_i2c/`
`help` 
you should see a result, on master you won't and it will get locked up.

### Issues/PRs references
fixes #10614
discussed in #10608
closed 2018-12-19 by leandrolanzieri