RIOT-OS/RIOT #10618: Can't start application on STM32F3

Carton32, 2018-12-14:
Hi everyone,

I have a custom board with a radio transceiver, a relay, a sensor for measuring current consumption values and some voltage transformers (230VAC to 12VDC, 5VDC and 3.3VDC). The board has either a STM32F303RB or a STM32F302RB. I have the same board for development which has a Nucleo-F303RE instead of a soldered microcontroller. The production boards don't have any serial output so I use the Nucleo to test other sensors and to have a serial output.

I'm programming the production boards with a ST-Link V2 through a JTAG port. The design of the production boards is used since two years with a "bare-metal" app.

I use for `CPU_MODEL` either `stm32f303rb` or `stm32f302rb`. I copied the file of the MCD Application team `stm32f302xc.h` and adapted it for RIOT as it's not included by default. The support for STM32F303RB is already done.

On the setup with the Nucleo-F303RE, my application runs fine without any issue in every cases. On the other hand, on the production boards with soldered microcontroller, the application can run only after the flash with the ST-Link V2. The application restarts itself after pressing the reset button on the board. **But, once I unplug the power and replug it, I encounter a hard fault and it get stuck at the breakpoint l.337 of vectors_cortexm.c**.

For the moment, I can't give any other information. I'm not a master with GDB and I have no serial communication with the production boards.

Has anyone ever seen this problem ?

If anyone has some advice about how to debug the app in this situation, I would be happy.

Thanks for your help

kaspar030, 2018-12-14:
This issue tracker is meant for issues within RIOT's code base, not for usage trouble. Next time, try the mailing list "devel@riot-os.org".

> **But, once I unplug the power and replug it, I encounter a hard fault and it get stuck at the breakpoint l.337 of vectors_cortexm.c**.

Can you, at that point, in GDB, try "print pc"? That might point to the address where the hard fault happens.

Better yet, try to get into gdb, reset using "monitor reset halt", set a breakpoint in "board_init()" and step through until you find the location of the hard fault.

Carton32, 2018-12-17:
I could debug my app with your help. Thank you and sorry for posting this here. I thought at first this could be a RIOT bug...

The problem was in some gpio_init() of board_init(). I still can't explain why the code was working right after a flash but not after a hard reset (power off/on).

Anyway, thanks for your help again !

closed 2018-12-17 by Carton32