RIOT-OS/RIOT #10620: rn2xx3: rn2xx3_write_cmd_no_wait can't be correct

keestux, 2018-12-16:
In the driver `rn2xx3`, the function `rn2xx3_write_cmd_no_wait` there is this code:

    mutex_lock(&(dev->cmd_lock));
    _uart_write_str(dev, dev->cmd_buf);
    _uart_write_str(dev, closing_seq);
    DEBUG("[rn2xx3] RET: %s\n", dev->resp_buf);
    mutex_unlock(&(dev->cmd_lock));
    return rn2xx3_process_response(dev);

Immediately after writing the command to the UART it prints and processes the response. That can't be correct. First of all, the output is (I hope) interrupt driven. That means the command isn't completely send to the device yet. And even if it was send out, the MCU is way faster than the serial output and input. It is looking at resp_buf without knowing that it contains valid information.

The function `rn2xx3_write_cmd_no_wait` is used for `sys sleep`. That needs a different approach than what we do in `rn2xx3_write_cmd`. If the command succeeds then there is no response. If the command fails then we can process the response. A dilemma. I suggest to a brief delay, say 100ms, and then see if something came in. If there is, then we can process the response and return the error. Else we can assume the command was successful and return a `RN2XX3_OK`.

aabadie, 2018-12-21:
Thanks for the detailed analysis @keestux. There is indeed something wrong and that should be changed. Maybe we could get rid of `rn2xx3_write_cmd_no_wait`, since it's only used in one place.

I'll see if I can find some to time to have a look.

aabadie, 2019-09-21:
This is a bug, it should not be closed automatically.

closed 2019-10-20 by aabadie