RIOT-OS/RIOT #10628: Can't read data from BH1750 senseor using STM32F3Discovery

kubazdz, 2018-12-18:
#### Description
BH1750 light sensor returns 0 lx

#### Steps to reproduce the issue
I'm trying to read values from BH1750 light sensor using STM32F3Discovery. I'm using GY-302 module. Tried on two different ones and the output is the same

Connections:

STM32              BH1750
5V ---------------> VCC   
GND ------------> GND
PA14 ------------> SDA
PA15 ------------> SCL
5V ---------------> ADDR

Code:

```
/*
 * mian.c
 *
 *  Created on: Dec 10, 2018
 *      Author: kuba
 */

#include <stdio.h>
#include "board.h"
#include "periph/gpio.h"
#include "leds.h"
#include "periph_conf.h"
#include "bh1750fvi.h"

#define I2C_INTERFACE I2C_DEV(1)

int main(void)
{
	board_init();
	uint16_t light;
	int res;
	bh1750fvi_t device;
	device.i2c = I2C_INTERFACE;
	device.addr = BH1750FVI_DEFAULT_ADDR;
	bh1750fvi_params_t bh1750;
	bh1750.i2c = I2C_INTERFACE;
	bh1750.addr = BH1750FVI_DEFAULT_ADDR;
	blink_leds();
	res = bh1750fvi_init(&device, &bh1750);
	printf("%d\n", res);
	switch(res)
	{
		case BH1750FVI_ERR_I2C:
			printf("Error: Cannot connect to the I2C device. Exiting program\n");
			return BH1750FVI_ERR_I2C;
			break;
		case BH1750FVI_OK:
			printf("Connected to the I2C device\n");
			break;
		default:
			break;
	}
	while(1)
	{
		light = bh1750fvi_sample(&device);
		printf("Light: %d lx\n", light);
		dumb_delay(1000*ms);
	}
	return 0;
}
```

#### Expected results

Getting light value

#### Actual results

Output:

```
0
Connected to the I2C device
[bh1750fvi] sample: triggering a conversion
[bh1750fvi] sample: reading the results
Light: 0 lx
[bh1750fvi] sample: triggering a conversion
[bh1750fvi] sample: reading the results
Light: 0 lx
[bh1750fvi] sample: triggering a conversion
[bh1750fvi] sample: reading the results
Light: 0 lx
[bh1750fvi] sample: triggering a conversion
[bh1750fvi] sample: reading the results
Light: 0 lx
...
```

#### Versions
STM32F3Discovery
BH1750 (GY-302 module)

kubazdz, 2018-12-18:
Edit: I just found out that I had wrong pins for I2C connected, and switched it so that they matched pins that are initialized by RIOT

kubazdz, 2018-12-18:
No, sorry I was just checking what happens now. Now i get the BH1750_ERR_I2C in return. I'm not sure if im passing right arguments to the bh1750fvi_init and bh1750fvi_sample functions, so now im trying to do it using i2c periph driver to check if it'll work this way

kubazdz, 2018-12-18:
using i2c periph driver output is the same

kubazdz, 2018-12-19:
I didn't fix the problem, but what i found is that bh1750 driver is not likely to work on stm32f3discovery because it uses i2c_write_byte function when stm32f3 i2c driver doesn't support it. In the i2c_1.c file (in stm32_common) there is only i2c_write_**bytes** function

MrKevinWeiss, 2019-01-02:
Also, just to be sure you are defining `#define I2C_INTERFACE I2C_DEV(1)` meaning that you use 
```c
        .scl_pin        = GPIO_PIN(PORT_F, 1),
        .sda_pin        = GPIO_PIN(PORT_F, 0),
```
and you are **NOT** using  `#define I2C_INTERFACE I2C_DEV(0)`  which would require 
```c
        .scl_pin        = GPIO_PIN(PORT_B, 6),
        .sda_pin        = GPIO_PIN(PORT_B, 7),
```
for the stm32f3discovery board

MrKevinWeiss, 2019-01-03:
Hmm according to the [datasheet](https://www.mouser.com/ds/2/348/bh1750fvi-e-186247.pdf) it says max voltage is 3.6 volts not 5, hopefully you didn't blow the chip.

MrKevinWeiss, 2019-01-03:
I think I figured it out.  The i2c address are incorrect.  According to the datasheet the `BH1750FVI_ADDR_PIN_HIGH` should be 0x5C and the `BH1750FVI_ADDR_PIN_LOW` should be 0x23.  Since the `BH1750FVI_DEFAULT_ADDR == BH1750FVI_ADDR_PIN_HIGH` it means that the truly low state should be what is used, which is actually the default... Sorry :disappointed: 

Quick fix, remove the +5V from the ADDR pin and tie it to ground (you may also leave it floating but I didn't check if there was a pulldown resistor on the module, it worked for me).

Proper fix, wait for my PR to get merged in master fixing the naming.

Also you can checkout the **tests/driver_bh1750** for example code on how to initialize and use.

MrKevinWeiss, 2019-01-03:
@kubazdz Please let me know if this fixes the issue.  I was able to get it running on my setup (though I was using a nucleo-f303re).

Also thanks for catching and reporting this bug, we appreciate it! :+1:

closed 2019-01-03 by aabadie

kubazdz, 2019-01-04:
@MrKevinWeiss I'll try it as soon as I'll can and let you know. Thanks a lot

PR #10694 (2019-01-03): drivers/bh1750: Fix mislabeled i2c address
### Contribution description

The i2c address when the pin is high/low are swapped.
This swaps them to match the datasheet and changes the default to low.

### Testing procedure

Connect a board and run the tests/driver_bh1750 with the addr pin low, it should work by default.  Pull it high and set the pin high.

Or just read the [datasheet](https://www.mouser.com/ds/2/348/bh1750fvi-e-186247.pdf) page 7/17, only macro names were swapped no actual values.

### Issues/PRs references
Fixes #10628