RIOT-OS/RIOT #10634: pyterm: weird behaviour with long lines.

jcarrano, 2018-12-18:
#### Description

Pyterm fails to handle long input lines (more than 64 characters,)

#### Steps to reproduce the issue

Go to `tests/shell`. I used a samr21:

```
$ cd tests/shell
$ BOARD=samr21-xpro make flash term
```

In the terminal, type a long line. No output will show up. Then type something else (say, `help`) a gardbage output follows:

```
$ BOARD=samr21-xpro make term
/home/jcarrano/source/vanillaRIOT/dist/tools/pyterm/pyterm -p "/dev/ttyACM0" -b "115200"
/home/jcarrano/source/vanillaRIOT/dist/tools/pyterm/pyterm:58: DeprecationWarning: The 'warn' method is deprecated, use 'warning' instead
  logging.getLogger("").warn("Twisted not available, please install "
Twisted not available, please install it if you want to use pyterm's JSON capabilities
/home/jcarrano/source/vanillaRIOT/dist/tools/pyterm/pyterm:490: DeprecationWarning: The SafeConfigParser class has been renamed to ConfigParser in Python 3.2. This alias will be removed in future versions. Use ConfigParser directly instead.
  self.config = configparser.SafeConfigParser()
2018-12-18 18:38:11,397 - INFO # Connect to serial port /dev/ttyACM0
Welcome to pyterm!
Type '/exit' to exit.
2018-12-18 18:38:20,745 - INFO # main(): This is RIOT! (Version: 2018.07-devel-885-g77b91-luxemburg-optparse)
2018-12-18 18:38:20,746 - INFO # test_shell.
> sdfwdjewoijoiejdioewdjeoiwdjiowedjeoiwjdewoidjewoidjweoijdioewjdoweidjwoeidjweoijdwoeidjweoidjiwoed
help
2018-12-18 18:38:32,852 - INFO #  sdfwdjewoijoiejdioewdjeoiwdjiowedjeoiwjdewoidjewoidjweoijdioewjdM.X��,l��G�Y% �{�z�'���d�Bhelp
2018-12-18 18:38:32,861 - INFO # shell: command not found: sdfwdjewoijoiejdioewdjeoiwdjiowedjeoiwjdewoidjewoidjweoijdioewjd
> 
```

Notice the garbage characters.

#### Expected results

Using another term (I use miniterm.py) gives the expected result:

```
$ miniterm.py --raw /dev/ttyACM0 115200
--- Miniterm on /dev/ttyACM0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
main(): This is RIOT! (Version: 2018.07-devel-885-g77b91-luxemburg-optparse)
test_shell.
> sdfwdjewoijoiejdioewdjeoiwdjiowedjeoiwjdewoidjewoidjweoijdioewjdoweidjwoeidjweoijdwoeidjweoidjiwoed
shell: command not found: sdfwdjewoijoiejdioewdjeoiwdjiowedjeoiwjdewoidjewoidjweoijdioewjdoweidjwoeidjweoijdwoeidjweoidjiwoed
> 
> 
```

#### Versions

```
Operating System Environment
-----------------------------
       Operating System: "Arch Linux" 
                 Kernel: Linux 4.19.4-arch1-1-ARCH x86_64 unknown

Installed compiler toolchains
-----------------------------
             native gcc: gcc (GCC) 8.2.1 20181127
      arm-none-eabi-gcc: arm-none-eabi-gcc (Arch Repository) 8.2.0
                avr-gcc: missing
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
                  clang: clang version 7.0.0 (tags/RELEASE_700/final)

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "3.0.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
               avr-libc: missing (missing)

Installed development tools
---------------------------
                  cmake: cmake version 3.13.1
               cppcheck: Cppcheck 1.85
                doxygen: 1.8.14
                 flake8: 3.6.0 (mccabe: 0.6.1, pycodestyle: 2.4.0, pyflakes: 2.0.0) CPython 3.7.1 on Linux
                    git: git version 2.20.0
                   make: GNU Make 4.2.1
                openocd: Open On-Chip Debugger 0.10.0
                 python: Python 3.7.1
                python2: Python 2.7.15
                python3: Python 3.7.1
             coccinelle: missing
```

gebart, 2018-12-18:
Are you sure this is a bug in pyterm? The UART stdin input buffer is 64 bytes IIRC.

gebart, 2018-12-18:
I replied too soon, I noticed that you tested another terminal. Does it also send the whole line in one go or does it send char by char like minicom?

kaspar030, 2018-12-19:
> The problem still appears. I don't understand why the size of the UARD stdin buffer would affect this. Where is that buffer? Is is part of our code or is it in newlib?

That buffer is defined in ```sys/stdio_uart/stdio_uart.c```. It is necessary because within the UART ISRs, we have to get rid of incoming characters *fast*, thus there's a ringbuffer that can always (unless full) be written to within an ISR. If that buffer is full, e.g. when receiving bytes faster than the shell can consume them, bytes get dropped.

jcarrano, 2018-12-19:
I confirm this is a bug in RIOT and not in pyterm. I have a minimal example. I will close this and open another PR.

> If that buffer is full, e.g. when receiving bytes faster than the shell can consume them, bytes get dropped.

It is not about speed, In my example I lowered the baud rate to 1200 and it is still failing (and even before, it would be very unlikely that the shell could not react fast enough to a mere 11k chars a second).

closed 2018-12-19 by jcarrano

jcarrano, 2018-12-19:
I moved this to a new issue (#10639 )