RIOT-OS/RIOT #10645: stdio_init happens late during boot

gebart, 2018-12-20:
This applies to newlib platforms at least, possibly others.

### Actual behavior

stdio_init is called from _init which is run after board_init. This has the downside that it is not possible to use printf or DEBUG from board_init or from the init function of periph drivers which are initialized by periph_init, e.g. SPI, since those run before _init. 

### Desired behavior

stdio_init should be called as early as possible, for stdio_uart this would likely be immediately after clocking has been set up, (which is a prerequisite for generating the proper baud rate) 

### References

https://github.com/RIOT-OS/RIOT/pull/10615#discussion_r241721679

kaspar030, 2018-12-20:
How about moving it into kernel_init()?

dylad, 2018-12-20:
> How about moving it into kernel_init()?

Moving it to `kernel_init()` will not help since it is called even after `__libc_init_array() ` (for cortexm_common).
Maybe we should moving it to `cpu_init()` or `board_init()` ?

gebart, 2018-12-21:
board init might be the best location since the board developer should be aware of which hardware modules need to be initialized before stdio can work.

dylad, 2018-12-21:
As I see, most (almost all ?) `periph_init()` are called within `cpu_init()` of each platform. Moreover, `cpu_init()` is called from `board_init()`. But we cannot call `periph_init()` before our clocks are setup which happen during `cpu_init()` BUT to solve our issue we must call `stdio_init()` BEFORE the end `cpu_init()`. This also means we cannot call `stdio_init()` from `board_init()`.

It gives us two options here : 
- call `stdio_init` within` cpu_init()` but before `periph_init()`
- call `stdio_init` at the very beginning of `periph_init()`

I guess the later will reduce the number of modified files.

aabadie, 2018-12-21:
> I guess the latter will reduce the number of modified files.

This is what I did when working STM32s I2C unification, especially because of this.

maribu, 2019-01-17:
@aabadie, @dylad: Have you already created a PR for

> - call `stdio_init` at the very beginning of `periph_init()`

?

If not, I might just do that ;-)

Such a PR might result in some serious bug fixing. E.g. the MSB-A2 did not run `periph_init()` at all until recently. Other bugs like that on the older and less commonly used boards might be exposed by that. (Which will look like a bad thing at first, but in retrospect, when all the hard labor of bug-fixing is done, it will likely look like a good thing ;-))

maribu, 2020-01-31:
Fixed in https://github.com/RIOT-OS/RIOT/pull/11367

closed 2020-01-31 by maribu