RIOT-OS/RIOT #10650: [RFC] $(APPDIR) in INCLUDES of make system

gschorcht, 2018-12-22:
#### Description

Applications often need to include header files from their directory, for example, to override a parameter configuration file of a driver. To include such a file instead of the overridden file, $(APPDIR) has to be added as the first entry to the `INCLUDES` variable. This can either be done in application's `Makefile` or in application's `Makefile.local`.

#### The question
Would it make sense to always add the application directory `$(APPDIR)` as the first include path to the `INCLUDES` variable in `$(RIOTBASE)/Makfile.include`?

gschorcht, 2019-08-10:
@cladmi @jia200x @kaspar030 It's a question to the maintainers of the build system.

cladmi, 2019-08-13:
I would not say it is a "build system" question but more from a user point of view.

Is it wanted that by default, if you create a header in the directory to just share between two source files, it could overwrite a system one for another module you do not know.

I am usually more in favor of explicit over magic so do not mind being requested to write `INCLUDES += -I$(CURDIR)"` or even better `INCLUDES += -I$(CURDIR)/include`.

Adding by default a `$(CURDIR)/include` could be safer, but would need to be documented too.

closed 2020-02-14 by gschorcht