RIOT-OS/RIOT #10651: Tagged version of LOG_* macros

gschorcht, 2018-12-24:
#### Description

There are many different output formats used with the `LOG_ *` macros. The formats that have been found are:
```c
LOG_INFO("message\n");
LOG_INFO("INFO [MODULE] message\n");
LOG_INFO("[module] message\n");
LOG_INFO("module: message\n");
LOG_INFO("function: message\n");
```
The same for LOG_ERROR and LOG_WARNING.

#### Feature request

To avoid that each module is using its own output format and to be able to implement a uniform format for all LOG_ * messages, there should be tagged versions of these macros LOG_TAG_INFO (tag, ...) that generates a consistent output:
```
[tag] message
```

gebart, 2018-12-24:
I think this would be a useful improvement. It would allow changing formatting in the future to allow e. g. JSON formatted logs for easier processing for automated testing. If `-DMODULE=$(MODULE)` is passed on the compiler command line it could be implemented quite straightforward as

```c
#define LOG_INFO(...) printf( #MODULE ":" __VA_ARGS__)
```
or something similar, (unsure about the preprocessor stringification syntax, writing from the phone)

gschorcht, 2019-08-10:
I have it on my ToDo list.

closed 2020-09-15 by stale[bot]