RIOT-OS/RIOT #10658: cpu/esp*: tracking issue for refactoring of ESP MCU implementations

gschorcht, 2018-12-27:
#### Description

There are already parts of MCU implementations for ESP8266 and ESP32 SoCs which are almost identical or at least very similar. Examples are the implementations of

- context handling, context switching and low level interrupt handling (`cpu/esp*/vendor/xtensa/*`)
- ESP-NOW netdev driver and interface (`cpu/esp*/esp-now`)

All other parts are completely different at the moment, since ESP32 and ESP8266 use SDKs with libraries which are completely different. However, with the ESP8266 RTOS SDK version under development, it seems that upcoming SDK versions for the ESP8266 will be at least partially compatible with the ESP-IDF for ESP32.

Therefore, it is very likely that both MCUs can use more and more the same code base. This issue is intended to track the modules that can be refactored designed to use the same code base and their progress.

#### Refactored modules

- [ ] `cpu/esp*/vendor/xtensa`
- [ ] `cpu/esp*/esp-now`

#### Refactoring prerequisites / steps

- [ ] adopt makefiles for ESP32 to use `cpu/common/esp/*`
- [ ] adopt makefiles for ESP8266 to use `cpu/common/esp/*`
- [ ] adopt `cpu/esp*/vendor/xtensa/*` to use architecture specific data in thread control block

#### PR dependencies

 `cpu/esp*/esp-now` depends on #10581, #9917

phillaverdiere, 2019-11-25:
I am triaging older issues.  Can this one be closed?

gschorcht, 2019-11-26:
Now that PR # 11108 has been merged, there are a lot of other possible code deduplications that are not listed here yet. I have to update the list and would leave the tracking issue open.