RIOT-OS/RIOT #10664: TCP Sockets can not be used / built

KillerLink, 2018-12-27:
#### Description
Can not build code using TCP sockets by writing `USEMODULE += gnrc_sock_tcp`.
I assume part of the probleme might be missing implementation? As you can see in [1] there only is a directory for UDP (which i successfully used previously) but not TCP. Since the documentation [2] describes the use of TCP by example and explicitly states 

> First you need to include a module that implements this API in your application's Makefile. For example the implementation for GNRC is called gnrc_sock_tcp.

i considered this a bug (as opposed to a feature request).

#### Steps to reproduce the issue
Try to build a project around TCP sockets, for example as described in the documentation here [2]

#### Expected results
Successfull Compilation

#### Actual results
Compiler Error: 
`arm-none-eabi-gcc: error: <path-redacted>/bin/cc2538dk/gnrc_sock_tcp.a: No such file or directory`

#### Versions
```
Operating System Environment
-----------------------------
       Operating System: "Arch Linux" 
                 Kernel: Linux 4.18.16-arch1-1-ARCH x86_64 unknown

Installed compiler toolchains
-----------------------------
             native gcc: gcc (GCC) 8.2.1 20180831
      arm-none-eabi-gcc: arm-none-eabi-gcc (Arch Repository) 8.2.0
                avr-gcc: avr-gcc (GCC) 8.2.0
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
                  clang: clang version 7.0.0 (tags/RELEASE_700/final)

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "3.0.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
               avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                  cmake: cmake version 3.12.4
               cppcheck: missing
                doxygen: 1.8.14
                 flake8: missing
                    git: git version 2.19.1
                   make: GNU Make 4.2.1
                openocd: Open On-Chip Debugger 0.10.0
                 python: Python 3.7.1
                python2: Python 2.7.15
                python3: Python 3.7.1
             coccinelle: missing
```
#### References
[1] https://github.com/RIOT-OS/RIOT/tree/master/sys/net/gnrc/sock
[2] https://riot-os.org/api/group__net__sock__tcp.html#ga0144778bc074e291fcb0d43bd35c0179

pipex, 2019-03-05:
I have the same issue. It seems that gnrc_sock_tcp is not implemented, or the documentation in the [API tcp example](https://riot-os.org/api/group__net__sock__tcp.html#ga0144778bc074e291fcb0d43bd35c0179) is outdated. Could you clarify on which one is it?

PeterKietzmann, 2019-03-07:
AFAIK there is no `gnrc_sock_tcp` implementation right now, although @brummer-simon was working on it. Simon, wasn't there an open issue that tracks ongoing efforts?

If you want a sock like TCP API, have a look at [lwip_sock_tcp](https://github.com/RIOT-OS/RIOT/tree/master/tests/lwip_sock_tcp)

PeterKietzmann, 2019-03-07:
@brummer-simon if this issue is a duplicate, please close

brummer-simon, 2019-03-07:
Integrating gnrc_tcp into gnrc_sock_tcp is the last point on my ever growing gnrc_tcp todo-list.

I am not sure if this is a duplicated Issue but it is still valid.

brummer-simon, 2019-03-07:
The API "gnrc_sock_tcp" isself is valid but nobody is currently implementing it.

miri64, 2019-03-07:
> The API "gnrc_sock_tcp" isself is valid but nobody is currently implementing it.

Not true. There is an implementation [by lwIP](https://github.com/RIOT-OS/RIOT/blob/87d0757689213152fdd9524fea116570345e0367/pkg/lwip/contrib/sock/tcp/lwip_sock_tcp.c) ;-).

brummer-simon, 2019-03-07:
Okay. gnrc_tcp doesn't implement it (but it should) ;)

miri64, 2019-03-07:
> Okay. gnrc_tcp doesn't implement it (but it should) ;)

Yepp ;-)

miri64, 2019-03-07:
Okay I also have to correct my comment. lwIP of course doesn't implement _gnrc__sock_tcp, but the `sock_tcp` API ;-).

pipex, 2019-03-12:
Thank your all for your replies. I indeed managed to use TCP sockets with lwIP and the example in tests/lwip

fengelhardt, 2019-10-01:
I could offer some spare time to work on this issue, if requested. No clue how long it takes me, though. I managed to understand the API of gnrc_tcp, and gnrc_sock_udp seems like a good model for a gnrc_sock_tcp implementation.

brummer-simon, 2019-10-01:
> I could offer some spare time to work on this issue, if requested. No clue how long it takes me, though. I managed to understand the API of gnrc_tcp, and gnrc_sock_udp seems like a good model for a gnrc_sock_tcp implementation.

Thanks for the offering but there are some issues in the underlaying Implementations that should be addressed before integration into gnrc_sock_tcp should occur. If you want I could notify you as soon as I think its ready for integration.

fengelhardt, 2019-10-02:
> Thanks for the offering but there are some issues in the underlaying Implementation that should be addressed before integration into gnrc_sock_tcp should occur. If you want I could notify you as soon as I think its ready for integration.

Ah, thank you for the info. I already thought about starting to work on the tcp sockets on my own, but good that I asked :D
Yes, please notify me if/when I could start working on this.

PR #16459 (2021-07-08): gnrc_tcp: rewrite passive open
### Contribution description
This PR rewrites the passive connection establishment mechanism of GNRC_TCP.
The new version aligns the GNRC_TCP interface with the SOCK_TCP interface with the goal to make a upcoming SOCK integration straight forward and trivial.

Since this is a large change, I've split the PR into three commits (api, implementation and tests) to help reviewing this.
The new tests are written in different style than existing tests, please ignore the difference in style for now, I plan to align the existing  test structure in a dedicated PR on after  [... cut]
PR #16493 (2021-07-13): gnrc_tcp: align with sock tcp
### Contribution description
This PR implements all missing functions and API changes for SOCK_TCP integration to GNRC_TCP. 

### Testing procedure
All new functions and return codes are tested within the GNRC_TCP suite.
Just execute the tests under tests/gnrc_tcp

### Issues/PRs references
Depends on PR #16459
Depends on PR #16461
Fixes #10664 - TCP Sockets can not be used / built
closed 2021-07-08 by fjmolinas