RIOT-OS/RIOT #10672: gnrc: runs full on very heavy load

miri64, 2018-12-29:
#### Description
When using a 6Lo capable node, I am able to cause one node to have a full packet buffer. The packet space is never released, so the node basically becomes unreachable.

This was originally reported in https://github.com/RIOT-OS/RIOT/pull/10581#issuecomment-450367114

#### Steps to reproduce the issue
Compile `gnrc_networking` on native with

```diff
diff --git a/examples/gnrc_networking/Makefile b/examples/gnrc_networking/Makefile
index 4d361f825..693378f50 100644
--- a/examples/gnrc_networking/Makefile
+++ b/examples/gnrc_networking/Makefile
@@ -18,7 +18,7 @@ BOARD_INSUFFICIENT_MEMORY := arduino-duemilanove arduino-mega2560 arduino-uno \
 
 # Include packages that pull up and auto-init the link layer.
 # NOTE: 6LoWPAN will be included if IEEE802.15.4 devices are present
-USEMODULE += gnrc_netdev_default
+USEMODULE += socket_zep
+USEMODULE += gnrc_pktbuf_cmd
 USEMODULE += auto_init_gnrc_netif
 # Specify the mandatory networking modules for IPv6 and UDP
 USEMODULE += gnrc_ipv6_router_default
```

Run two instances of it

`native1`:

```sh
TERMFLAGS="-z [::]:17754,[::1]:17755" make -C examples/gnrc_networking term
```

`native2`:

```sh
TERMFLAGS="-z [::]:17755,[::1]:17754" make -C examples/gnrc_networking term
```

Note: you might need to start both instances rather quickly between each start, otherwise the already running instance might abort with

```
/home/mlenders/Repositories/RIOT-OS/RIOT/examples/gnrc_networking/bin/native/gnrc_networking.elf: zep: read: Connection refused
```

Start pinging with full MTU-sized packages from both nodes (nearly) at the same time:

```
native1> ping6 200 fe80::25a:4550:a00:455a 1232 0
native2> ping6 200 fe80::25a:4550:a00:455b 1232 0
```

#### Expected results
There will be timeouts and maybe full packet buffer messages, but afterwards the packet buffer will be free (check with `pktbuf` command).

#### Actual results
The packet buffer runs full and won't be released (one node will only get ping timeouts rather quickly, the other nodes only reports `error: packet buffer full` multiple times and then return to shell. The `pktbuf` commands reveals that the packet buffer contains packets that are never released

#### Versions
Tested in `2018.10` and current master as of writing this issue (07ceddd9b157692845e2418b64537d1b05626865). I also had a look `2018.04` (where the `gnrc_pktbuf_cmd` pseudo-module does not exist) here both nodes ran full (but that might only be incidental).

```
$ ./dist/tools/ci/print_toolchain_versions.sh 

Operating System Environment
-----------------------------
       Operating System: "Arch Linux" 
                 Kernel: Linux 4.19.12-arch1-1-ARCH x86_64 unknown

Installed compiler toolchains
-----------------------------
             native gcc: gcc (GCC) 8.2.1 20181127
      arm-none-eabi-gcc: arm-none-eabi-gcc (Arch Repository) 8.2.0
                avr-gcc: avr-gcc (GCC) 8.2.0
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
                  clang: clang version 7.0.1 (tags/RELEASE_701/final)

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "3.0.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
               avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                  cmake: cmake version 3.13.1
               cppcheck: missing
                doxygen: 1.8.14
                 flake8: missing
                    git: git version 2.20.1
                   make: GNU Make 4.2.1
                openocd: missing
                 python: Python 3.7.1
                python2: Python 2.7.15
                python3: Python 3.7.1
             coccinelle: missing
```

(maybe @gschorcht also can report his build environment)

gschorcht, 2018-12-29:
Build environment for ESP32:
```
Operating System Environment
-----------------------------
       Operating System: "Ubuntu" "18.04.1 LTS (Bionic Beaver)"
                 Kernel: Linux 4.15.0-42-generic x86_64 x86_64

Installed compiler toolchains
-----------------------------
             native gcc: gcc (Ubuntu 7.3.0-27ubuntu1~18.04) 7.3.0
      arm-none-eabi-gcc: missing
                avr-gcc: missing
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
                  clang: missing

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: missing
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
               avr-libc: missing (missing)

Installed development tools
---------------------------
                  cmake: cmake version 3.10.2
               cppcheck: Cppcheck 1.82
                doxygen: 1.8.13
                 flake8: missing
                    git: git version 2.17.1
                   make: GNU Make 4.1
                openocd: missing
                 python: Python 2.7.15rc1
                python2: Python 2.7.15rc1
                python3: Python 3.6.7
             coccinelle: spatch version 1.0.4 with Python support and with PCRE support
```

miri64, 2018-12-29:
Yep...

miri64, 2018-12-29:
Since I can't reproduce in `native` with `netdev_tap` this points to an issue with 6Lo.

miri64, 2018-12-29:
@TimoRoth This is why I don't believe it is the `gnrc_icmpv6_echo` implementation.

gschorcht, 2018-12-29:
Hm, I realized now that I lost a prepared big comment in #10581 last night with the results of 3 hours of investigation with `airdump` :worried: This could have helped you a lot to investigate. When I was investigating the problem with `airdump` I could observe the following:

- 6Lo is working as expected. There are fragments with 1 byte `esp_now_header_t`, 5 bytes 6Lo header and 240 data.
- Data of ICMPv6 packets are completely filled with the ID byte so there should no other data bytes as payload in the packet buffer.
- Fragments can be interleaved, lost or duplicated.
- Number of Ping timeouts increases when the number of lost fragments increases.

I could reproduce that 2 or 3 ping timeouts occurred before the reassembly buffer in 6LoWPAN becomes full: https://github.com/RIOT-OS/RIOT/blob/a87c2c9563320776d3b9bf8247c92787c98d6f83/sys/net/gnrc/network_layer/sixlowpan/frag/rbuf.c#L84 After that one further ping timeout happens before the `error: packet buffer full` message occurs.

My guess: When the number of lost fragments increases, the reassembly buffer is filled with a large number of 6Lo fragments of different packages. The buffer becomes full before the timeout drops the fragments.

gschorcht, 2018-12-29:
But, increasing the reassembly buffer size alone didn't solve the problem.

miri64, 2018-12-29:
Also doesn't explain, why the packet is never released again. The datagrams with lost fragments should be garbage collected after 3 sec

https://github.com/RIOT-OS/RIOT/blob/a87c2c9563320776d3b9bf8247c92787c98d6f83/sys/net/gnrc/network_layer/sixlowpan/frag/rbuf.h#L34

https://github.com/RIOT-OS/RIOT/blob/a87c2c9563320776d3b9bf8247c92787c98d6f83/sys/net/gnrc/network_layer/sixlowpan/frag/rbuf.c#L219-L247

https://github.com/RIOT-OS/RIOT/blob/a87c2c9563320776d3b9bf8247c92787c98d6f83/sys/net/gnrc/network_layer/sixlowpan/frag/gnrc_sixlowpan_frag.c#L333-L336

https://github.com/RIOT-OS/RIOT/blob/a87c2c9563320776d3b9bf8247c92787c98d6f83/sys/net/gnrc/network_layer/sixlowpan/gnrc_sixlowpan.c#L347-L350

miri64, 2018-12-29:
I was at least able to confirm, using [this branch] that it neither is our ping implementation nor some lingering datagrams in the reassembly buffer that cause this problem.

miri64, 2018-12-29:
(I also added temporarily a `memset()` when freeing segments in the packet buffer. There is still readable data in the packet buffer, so this are definitely unfreed packets.

miri64, 2018-12-29:
I was able to run reproduce in Valgrind (using `gnrc_pktbuf_malloc` instead of `gnrc_pktbuf_static`). To be able to run with `socket_zep`

I needed to patch the `cpu/native/Makefile.include` (maybe something for @cladmi or @jcarrano to investigate?):

```diff
diff --git a/boards/native/Makefile.include b/boards/native/Makefile.include
index 626011712..22a8d9b61 100644
--- a/boards/native/Makefile.include
+++ b/boards/native/Makefile.include
@@ -167,7 +167,7 @@ all-valgrind: all
 all-cachegrind: all
 
 term-valgrind:
-       $(VALGRIND) $(VALGRIND_FLAGS) $(ELFFILE) $(PORT)
+       $(VALGRIND) $(VALGRIND_FLAGS) $(ELFFILE) $(TERMFLAGS)
 
 debug-valgrind-server:
        $(VALGRIND) $(VALGRIND_FLAGS) $(ELFFILE) $(PORT)
```

With `VALGRIND_FLAGS="--leak-check=full --show-leak-kinds=all"` I was able to reliably get the same heap summary at least twice:

```
HEAP SUMMARY:
    in use at exit: 1,319 bytes in 6 blocks
  total heap usage: 122,893 allocs, 122,887 frees, 9,722,785 bytes allocated

3 bytes in 1 blocks are still reachable in loss record 1 of 6
   at 0x4837C37: realloc (vg_replace_malloc.c:826)
   by 0x1123FC: _realloc_data (gnrc_pktbuf_malloc.c:175)
   by 0x112497: gnrc_pktbuf_realloc_data (gnrc_pktbuf_malloc.c:191)
   by 0x138307: gnrc_sixlowpan_iphc_send (gnrc_sixlowpan_iphc.c:1023)
   by 0x13348F: _send (gnrc_sixlowpan.c:292)
   by 0x133593: _event_loop (gnrc_sixlowpan.c:333)
   by 0x498827A: makecontext (in /usr/lib32/libc-2.28.so)

16 bytes in 1 blocks are still reachable in loss record 2 of 6
   at 0x483562B: malloc (vg_replace_malloc.c:299)
   by 0x1127F9: _create_snip (gnrc_pktbuf_malloc.c:284)
   by 0x1120A5: gnrc_pktbuf_add (gnrc_pktbuf_malloc.c:97)
   by 0x111A5A: gnrc_netif_hdr_build (gnrc_netif_hdr.c:20)
   by 0x120AED: _create_netif_hdr (gnrc_ipv6.c:266)
   by 0x120EB4: _send_unicast (gnrc_ipv6.c:405)
   by 0x1212E2: _send (gnrc_ipv6.c:593)
   by 0x12093D: _event_loop (gnrc_ipv6.c:193)
   by 0x498827A: makecontext (in /usr/lib32/libc-2.28.so)

20 bytes in 1 blocks are still reachable in loss record 3 of 6
   at 0x483562B: malloc (vg_replace_malloc.c:299)
   by 0x1127CE: _create_snip (gnrc_pktbuf_malloc.c:276)
   by 0x1120A5: gnrc_pktbuf_add (gnrc_pktbuf_malloc.c:97)
   by 0x13783B: gnrc_sixlowpan_iphc_send (gnrc_sixlowpan_iphc.c:709)
   by 0x13348F: _send (gnrc_sixlowpan.c:292)
   by 0x133593: _event_loop (gnrc_sixlowpan.c:333)
   by 0x498827A: makecontext (in /usr/lib32/libc-2.28.so)

20 bytes in 1 blocks are still reachable in loss record 4 of 6
   at 0x483562B: malloc (vg_replace_malloc.c:299)
   by 0x1127CE: _create_snip (gnrc_pktbuf_malloc.c:276)
   by 0x1120A5: gnrc_pktbuf_add (gnrc_pktbuf_malloc.c:97)
   by 0x11EE00: gnrc_icmpv6_build (gnrc_icmpv6.c:126)
   by 0x11F129: gnrc_icmpv6_echo_build (gnrc_icmpv6_echo.c:37)
   by 0x11F25A: gnrc_icmpv6_echo_req_handle (gnrc_icmpv6_echo.c:71)
   by 0x11ED4A: gnrc_icmpv6_demux (gnrc_icmpv6.c:90)
   by 0x120727: _demux (gnrc_ipv6.c:112)
   by 0x1217DA: _receive (gnrc_ipv6.c:799)
   by 0x12092A: _event_loop (gnrc_ipv6.c:188)
   by 0x498827A: makecontext (in /usr/lib32/libc-2.28.so)

20 bytes in 1 blocks are still reachable in loss record 5 of 6
   at 0x483562B: malloc (vg_replace_malloc.c:299)
   by 0x1127CE: _create_snip (gnrc_pktbuf_malloc.c:276)
   by 0x1120A5: gnrc_pktbuf_add (gnrc_pktbuf_malloc.c:97)
   by 0x111A5A: gnrc_netif_hdr_build (gnrc_netif_hdr.c:20)
   by 0x120AED: _create_netif_hdr (gnrc_ipv6.c:266)
   by 0x120EB4: _send_unicast (gnrc_ipv6.c:405)
   by 0x1212E2: _send (gnrc_ipv6.c:593)
   by 0x12093D: _event_loop (gnrc_ipv6.c:193)
   by 0x498827A: makecontext (in /usr/lib32/libc-2.28.so)

1,240 bytes in 1 blocks are still reachable in loss record 6 of 6
   at 0x483562B: malloc (vg_replace_malloc.c:299)
   by 0x1127F9: _create_snip (gnrc_pktbuf_malloc.c:284)
   by 0x1120A5: gnrc_pktbuf_add (gnrc_pktbuf_malloc.c:97)
   by 0x11EE00: gnrc_icmpv6_build (gnrc_icmpv6.c:126)
   by 0x11F129: gnrc_icmpv6_echo_build (gnrc_icmpv6_echo.c:37)
   by 0x11F25A: gnrc_icmpv6_echo_req_handle (gnrc_icmpv6_echo.c:71)
   by 0x11ED4A: gnrc_icmpv6_demux (gnrc_icmpv6.c:90)
   by 0x120727: _demux (gnrc_ipv6.c:112)
   by 0x1217DA: _receive (gnrc_ipv6.c:799)
   by 0x12092A: _event_loop (gnrc_ipv6.c:188)
   by 0x498827A: makecontext (in /usr/lib32/libc-2.28.so)

LEAK SUMMARY:
   definitely lost: 0 bytes in 0 blocks
   indirectly lost: 0 bytes in 0 blocks
     possibly lost: 0 bytes in 0 blocks
   still reachable: 1,319 bytes in 6 blocks
        suppressed: 0 bytes in 0 blocks

For counts of detected and suppressed errors, rerun with: -v
ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
```

(with differing frees and allocs counts of course)

So it very much looks like, that it is actually the sending of fragmented packets that causes leaks.

miri64, 2018-12-29:
I found some bug, but it doesn't resolve it completely (I still have a full packet buffer later): https://github.com/RIOT-OS/RIOT/pull/10679

miri64, 2018-12-29:
(kind of a sad news good news situation: I can't reproduce the bug with Valgrind anymore with #10679 anymore, but still with normal `gnrc_pktbuf_static`).

miri64, 2018-12-30:
Both fixing PRs are merged and I can't produce the steps to reproduce anymore.

closed 2018-12-30 by miri64

gschorcht, 2019-01-03:
@miri64 Unfortunatly, the problem with stucked 6Lo packets in GNRC packet buffer still remains :worried:

When I was testing my fixes and optimizations of `esp_now_netdev` in PR #10700, I could observe it several times.

I could still observe the same problem with 3 `iotlab-m3` nodes with the current master branch. Should I reopen this issue or we create a new one?

gschorcht, 2019-01-03:
@TimoRoth Could you take a look to the changes?

TimoRoth, 2019-01-03:
The only issue I found in #10700 so far is fixed in a later commit of it already. I'll give it a try this weekend and report how it performs.
As to this issue, I have also observed leftover packets things in the gnrc packet buffer, and I can't possibly see how esp-now or esp-eth would be causing it, so I very much think there is still a lingering issue somewhere.

gschorcht, 2019-01-03:
Sorry, wrong issue.