RIOT-OS/RIOT #10682: cpu/esp_common: esp_now netdev problems

gschorcht, 2018-12-30:
#### Description

Although the problems described in issue #10672 were solved, there seem to be further problems that occur on heavy load, at least for ESP-NOW:

- sometimes 6Lo fragments are kept in the packet buffer
- a single ping with no data is not possible after heavy load with ping timeouts

#### Steps to reproduce the issue

Precondition:
- Test with two ESP32 nodes, one pinging and one answering.
- Test with heavy traffic executed: 
    ```
    ping6 20 fe80::32ae:a4ff:fe18:7a3d 1232 0`
    ```
- Packet buffers on both sides are empty after the test!

Send one single small ping:
```
ping6 1 fe80::32ae:a4ff:fe18:7a3d
```
On air, we can dump the following WiFi frame
```
0040   ** ** ** ** dd 3b 18 fe 34 04 01 01 41 60 00 00
0050   00 00 0c 3a 40 fe 80 00 00 00 00 00 00 32 ae a4
0060   ff fe 41 60 f9 fe 80 00 00 00 00 00 00 32 ae a4
0070   ff fe 18 7a 3d 80 00 52 cc 00 54 00 01 54 54 54
0080   54
```
where `dd 3b 18 fe 34 04 01 01` is the ESP-NOW header including the one byte `esp_now_header_t`.

On answering node this leads to
```
6lo rfrag: entry 0x3ffbc53c (30:AE:A4:41:60:F9, 30:AE:A4:18:7A:3D, 1280, 18) created
6lo rfrag: add interval (960, 1199) to entry (30:AE:A4:41:60:F9, 30:AE:A4:18:7A:3D, 1280, 18)
```
There seems to be something inconsistent in the `rbuf`.

gschorcht, 2018-12-30:
@TimoRoth Unfortunatly, it seems to be a problem of `esp_now_netdev` ringbuffer since it `_recv` returns:
```
00000000  01  E5  00  00  31  96  57  57  57  57  57  57  57  57  57  57
00000010  57  57  57  57  57  57  57  57  57  57  57  57  57  57  57  57
00000020  57  57  57  57  57  57  57  57  57  57  57  57  57  57  57  57
00000030  57  57  57  57  57  57  57  57  57  57  57  57  57  57  57  57
00000040  57  57  57  57  57  57  57  57  57  57  57  57  57  57  57  57
00000050  57  57  57  57  57  57
```

gschorcht, 2018-12-30:
Reason is not really clear yet. When the test has been finished, available bytes in `esp_now_netdev` ringbuffer should be 0. But in my test, for example:
- on both nodes, 729 bytes available in the buffer
- on answering node, the read position is 343

When the single small ping frame arrives on answering node
- it is written to ringbuffer position 1072
- but 6Lo reads from 729 wrong data, an old 6Lo fragment

gschorcht, 2018-12-30:
@TimoRoth I have counted the generated `NETDEV_EVENT_ISR` and `NETDEV_EVENT_RX_COMPLETE` events. It seems that some `NETDEV_EVENT_ISR` generated by `esp_now_netdev` gets lost. This happens first exactly at the same time when first `ping timeout` occurs. The result of lost `NETDEV_EVENT_ISR` events is that subsequent `NETDEV_EVENT_ISR` and following `NETDEV_EVENT_RX_COMPLETE` events trigger the read of the wrong frame.

gschorcht, 2018-12-30:
The question is now why `NETDEV_EVENT_ISR` event gets lost an whether this is caused by `esp_now_netdev`.

TimoRoth, 2018-12-30:
My first idea, without looking at the code, would be a failing msg_try_send somewhere, because the queue on the other end is full. If that's the case, I'm not really sure how to properly handle the situation though.

gschorcht, 2018-12-30:
This is exactly what I though, but in that case we should get `gnrc_netif: possibly lost interrupt` message
 https://github.com/RIOT-OS/RIOT/blob/d3f78a5a4008a640ce4ad24743c77bfdc97c7730/sys/net/gnrc/netif/gnrc_netif.c#L1362-L1369 from here.

TimoRoth, 2018-12-30:
Hm, as we are not in an ISR but in the wifi thread when calling the event cb, this msg_send will actually block and not return until the message was delivered.

gschorcht, 2018-12-30:
`gnrc_netif` has a message queue of size 8. Should not be blocking until the message queue is full.

gschorcht, 2018-12-30:
Unfortunately you are right. Since we are not in an ISR the `wifi` thread would indeed block when queue is full and we don't get the `gnrc_netif: possibly lost interrupt` message at all. Bad for two reasons:
- We don't have any option to use `msg_try_send` instead of `msg_send`.
- The `wifi` thread can be blocked which must not happen.

On the other hand, if the message queue would become full and the `wifi` thread would be blocked, it should not be possible that the event gets lost.

miri64, 2018-12-30:
> * We don't have any option to use `msg_try_send` instead of `msg_send`.

Might be a good idea to change that to a `msg_try_send()`. But this doesn't fix the problem of the filling ringbuffer of course :-/.

gschorcht, 2018-12-30:
I have already an idea but I'm not at my desk. I will try it later tonight.

gschorcht, 2018-12-31:
I tried the following two changes:

##### 1. Approach
I tried to simulate that we are in an ISR when we call
```
_esp_now_dev.netdev.event_callback((netdev_t*)&_esp_now_dev, NETDEV_EVENT_ISR);
```
to get the error message when the event is lost. With `irq_interrupt_nesting` counter greater than 0, `irq_is_in()` returns `true`. Since interrupts are disabled and context switches are not possible if `irq_is_in()` returns `true`, this change shouldn't cause any problem.
```diff
--- a/cpu/esp_common/esp-now/esp_now_netdev.c
+++ b/cpu/esp_common/esp-now/esp_now_netdev.c
@@ -228,8 +228,10 @@ static IRAM_ATTR void esp_now_recv_cb(const uint8_t *mac, const uint8_t *data, i
     ringbuffer_add(&_esp_now_dev.rx_buf, (char*)data, len);
 
     if (_esp_now_dev.netdev.event_callback) {
+        irq_interrupt_nesting++;
         _esp_now_dev.recv_event = true;
         _esp_now_dev.netdev.event_callback((netdev_t*)&_esp_now_dev, NETDEV_EVENT_ISR);
+        irq_interrupt_nesting--;
     }
```
It was surprised that I don't get the error message `gnrc_netif: possibly lost interrupt.`.

##### 2. Approach
If we are not in an ISR, why not to call event callback directly with `NETDEV_EVENT_RX_COMPLETE`. Yes, `esp_now_recv_cb` must not be time consuming to avoid blocking the `wifi` thread for too long. I just tried the following change:
```diff
--- a/cpu/esp_common/esp-now/esp_now_netdev.c
+++ b/cpu/esp_common/esp-now/esp_now_netdev.c
@@ -227,13 +227,13 @@ static IRAM_ATTR void esp_now_recv_cb(const uint8_t *mac, const uint8_t *data, i
     ringbuffer_add(&_esp_now_dev.rx_buf, (char*)mac, ESP_NOW_ADDR_LEN);
     ringbuffer_add(&_esp_now_dev.rx_buf, (char*)data, len);
 
+    critical_exit();
+    mutex_unlock(&_esp_now_dev.rx_lock);
+
     if (_esp_now_dev.netdev.event_callback) {
         _esp_now_dev.recv_event = true;
-        _esp_now_dev.netdev.event_callback((netdev_t*)&_esp_now_dev, NETDEV_EVENT_ISR);
+        _esp_now_dev.netdev.event_callback((netdev_t*)&_esp_now_dev, NETDEV_EVENT_RX_COMPLETE);
     }
-
-    critical_exit();
-    mutex_unlock(&_esp_now_dev.rx_lock);
 }
```
With these changes I can execute
```
node1_60f9> ping6 20000 fe80::32ae:a4ff:fe18:7a3d 1232 0
```
without any `ping timeout` or buffer full error. Besides, I can ping in both directions at the same time.
```
node1_60f9> ping6 20000 fe80::32ae:a4ff:fe18:7a3d 1232 0
node2_7a3d> ping6 20000 fe80::32ae:a4ff:fe41:60f9 1232 0
```
Then, I get a number of `ping timeouts`, `pktbuf full` and `rbuf full` errors, but none of the nodes locks. The packet loss is only 2 ... 4%!

It is also possible to ping a node from two different nodes without any problems.
```
node1_60f9> ping6 20000 fe80::32ae:a4ff:fe18:7a3d 1232 0
node3_60f9> ping6 20000 fe80::32ae:a4ff:fe18:7a3d 1232 0
```
What about the timing problem and the `wifi` thread?

Sending `NETDEV_EVENT_RX_COMPLETE` directly leads to following procedure:
- `esp_now_recv_cb` calls  `gnrc_netif::_event_cb` with `NETDEV_EVENT_RX_COMPLETE`.
- `gnrc_netif::_event_cb` calls  `esp_now_gnrc::_recv`.
- `esp_now_gnrc::_recv` calls `esp_now_netdev::_recv` to fetch the packet and returns.
- `gnrc_netif::_event_cb` passes the packet to subscribers (in case of esp_now the 6Lo) with `gnrc_netif::_pass_on_packet` using `msq_try_send`.
- `gnrc_netif::_event_cb` returns to `esp_now_recv_cb` which also simply returns.

Since the `wifi` thread has the highest priority, sending a packet to the subscribers in `gnrc_netif::_pass_on_packet` will not lead to a context switch. So the most time consuming part is therefore the execution of `esp_now_netdev::_recv`. Since the tests seem to work also under heavy load, I would guess that the execution time of  `esp_now_netdev::_recv` isn't a problem. The average time consumed by `esp_now_recv_cb` is only 143 us.

@miri64 @TimoRoth If you are comfortable with this approach, I could provide a PR.

gschorcht, 2018-12-31:
In principle, `esp_now_recv_cb` calls `esp_now_netdev::_recv`directly. If this is faster than the next fragment can be received, we should neither need the ringbuffer nor the mutex `dev->rx_buf`. A single `_buffer` for one esp-now packet and a flag `_buffer_used` which indicates whether the buffer is already used should be enough.
```c
static IRAM_ATTR void esp_now_recv_cb(const uint8_t *mac, const uint8_t *data, int len)
{
    if (_buffer_used) {
        DEBUG("%s: buffer full, dropping incoming packet of %d bytes\n", __func__, len);
        return;
    }

    critical_enter();
    _buffer_used = true;
    _buffer[0] = len;
    memcpy(_buffer + 1, mac, ESP_NOW_ADDR_LEN);
    memcpy(_buffer + 1 + ESP_NOW_ADDR_LEN, data, len);
    critical_exit();

    if (_esp_now_dev.netdev.event_callback) {
        _esp_now_dev.recv_event = true;
        _esp_now_dev.netdev.event_callback((netdev_t*)&_esp_now_dev, NETDEV_EVENT_RX_COMPLETE);
    }
}

static int _recv(netdev_t *netdev, void *buf, size_t len, void *info)
{
    ...
    if (buf && len) {
        ...
        memcpy(buf, _buffer + 1, size);
        _buffer_used = false;
    }    
}

In worst case the next fragment arrives before the fragment has been read from the buffer. I have to test a bit more, but in first tests it worked as stable as with mutex and ringbuffer. But without ringbuffer an mutex, `esp_now_recv_cb` takes only 35 us.

miri64, 2018-12-31:
Is the `wifi` thread distinct from the thread, `gnrc_netif` is running in?

miri64, 2018-12-31:
@kaibeckmann This discussion might also be of interest for your `rail` port.

gschorcht, 2018-12-31:
@miri64 
> Is the wifi thread distinct from the thread, gnrc_netif is running in?

Yes, the `wifi` thread is a separate thread with prio 1 handling the `WiFi` hardware. Since `gnrc_netif::_event_cb` is called from `esp_now_recv_cb` which is called in `wifi` thread context, `gnrc_netif::_event_cb` is executed in the `wifi` thread context.

TimoRoth, 2018-12-31:
It definitely seems like an improvement to me, and the ringbuffer is indeed not needed anymore then, since the buffer content will be read in the same thread, at least from how I understand the flow. I think a separate field for the rx_len would be nicer than using the first byte of the buffer, but otherwise it seems ok to me.

gschorcht, 2018-12-31:
I would suggest to change it in two steps. In first step, we improve the stability using approach 2 in https://github.com/RIOT-OS/RIOT/issues/10682#issuecomment-450600883. In second step, we can optimize it. I will also have to check whether these changes work with ESP-NOW on esp8266.

I can provide a PR for step 1.

TimoRoth, 2018-12-31:
There is also still the issue remaining with the stuck gnrc pktbuffers, which I doubt are caused by that. But that needs addressed separately.

gschorcht, 2018-12-31:
After my changes I couldn't observe any packet buffer problems anymore.

gschorcht, 2019-01-02:
> > * We don't have any option to use `msg_try_send` instead of `msg_send`.
> 
> Might be a good idea to change that to a `msg_try_send()`. But this doesn't fix the problem of the filling ringbuffer of course :-/.

@miri64 It would help to use `msg_try_send()` if https://github.com/RIOT-OS/RIOT/blob/d3f78a5a4008a640ce4ad24743c77bfdc97c7730/sys/net/gnrc/netif/gnrc_netif.c#L1358 would return an error if `msg_try_send` failed. In that case, the caller would be able to decide what to do independent on whether it is done in interrupt or thread context. In our case, for example, it would probably be better to drop the new frame than to block the thread which is handling WiFi hardware events.

I'm wondering if it's not also a problem for other `netdev` drivers and what they do if the ISR event gets lost because they don't receive any feedback.

gschorcht, 2019-01-02:
Finally, I could figure out the problem why some ISR events gets lost. I will provide a PR in next hours.

closed 2019-01-11 by MrKevinWeiss