RIOT-OS/RIOT #10701: gnrc: packets in gnrc packet buffer are not freed on very heavy load

gschorcht, 2019-01-03:
#### Description

The behavior was originally reported in https://github.com/RIOT-OS/RIOT/pull/10581#issuecomment-450367114 and later reproduced in issue #10672. 

I could reproduce a similar or the same behavior for `iotlab-m3` nodes. Fixes provides with PR #10679 and PR #10680 were already applied.

If you are using 6Lo-capable nodes, it is possible that packet space for some packets will never be released in `gnrc/pktbuf` on heavy load.

#### Steps to reproduce the issue

Flash two `iotlab-m3` nodes with `examples/gnrc_networking` with command:
```
USEMODULE=gnrc_pktbuf_cmd make BOARD=iotlab-m3 -C examples/gnrc_networking flash
```
Produce heavy network traffic with `ping` command with large data size and a period of 0 ms.
```
node1> ping6 100 <node2_addr> 1232 0
```
```
node2> ping6 100 <node1_addr> 1232 0
```
This will result in a lot of `ping timeout`s and probably `error: packet buffer full` messages. Once, the `ping` commands have been finished, use the `pktbuf` command to print the packet buffer spaces on both nodes.

#### Expected results

Packet buffer should be empty on both nodes.

#### Actual results

There are still some space allocated in packet buffer, e.g.:
```
node1> pktbuf
packet buffer: first byte: 0x20001b00, last byte: 0x20003300 (size: 6144)
  position of last byte used: 5648
=========== chunk   0 (0x20001b00 size:   96) ===========
00000000  18  1B  00  20  20  1C  00  20  28  00  00  00  01  00  00  00
00000010  02  00  00  00  00  00  00  00  00  00  00  00  30  1B  00  20
00000020  18  00  00  00  01  00  00  00  FF  10  6B  11  15  C5  00  00
00000030  08  08  07  00  00  FF  C9  FF  15  11  6B  10  65  F6  5D  02
00000040  15  11  6B  10  65  FC  5C  22  00  1B  00  20  48  1C  00  20
00000050  D8  04  00  00  01  00  00  00  03  55  55  55  55  55  55  55
~ unused: 0x20001b60 (next: 0x20002120, size:  192) ~
=========== chunk   1 (0x20001c20 size: 1280) ===========
00000000  60  00  00  00  04  D8  3A  40  FE  80  00  00  00  00  00  00
00000010  17  11  6B  10  65  F6  5D  02  FE  80  00  00  00  00  00  00
00000020  17  11  6B  10  65  FC  5C  22  81  00  9D  DD  00  55  00  09
00000030  55  55  55  55  55  55  55  55  55  55  55  55  55  55  55  55
00000040  55  55  55  55  55  55  55  55  55  55  55  55  55  55  55  55
...
000004B0  55  55  55  55  55  55  55  55  55  55  55  55  55  12  00  00
000004C0  00  00  00  00  20  12  00  00  54  54  54  54  54  54  54  54
000004D0  54  54  54  54  54  54  54  54  54  54  54  54  54  54  54  54
000004E0  54  54  54  54  54  54  54  54  54  54  54  54  54  54  54  54
000004F0  54  54  54  54  54  54  54  54  54  54  54  54  54  54  54  54
~ unused: 0x20002120 (next: 0, size: 4576) ~```
```
```
node2> pktbuf
pktbuf
packet buffer: first byte: 0x20001b00, last byte: 0x20003300 (size: 6144)
  position of last byte used: 5648
=========== chunk   0 (0x20001b00 size:   24) ===========
00000000  48  1B  00  20  98  26  00  20  28  00  00  00  01  00  00  00
00000010  02  00  00  00  00  00  00  00
~ unused: 0x20001b18 (next: 0x20001b90, size:   48) ~
=========== chunk   1 (0x20001b48 size:   72) ===========
00000000  00  00  00  00  60  1B  00  20  18  00  00  00  01  00  00  00
00000010  FF  10  6B  11  15  C5  00  00  08  08  07  00  00  FF  DD  FF
00000020  15  11  6B  10  65  F9  5D  3A  15  11  6B  10  65  F6  5D  02
00000030  00  1B  00  20  C0  26  00  20  D8  04  00  00  01  00  00  00
00000040  03  53  53  53  53  53  53  53
~ unused: 0x20001b90 (next: 0x20002b98, size: 2824) ~
=========== chunk   2 (0x20002698 size: 1280) ===========
00000000  60  00  00  00  04  D8  3A  40  FE  80  00  00  00  00  00  00
00000010  17  11  6B  10  65  F9  5D  3A  FE  80  00  00  00  00  00  00
00000020  17  11  6B  10  65  F6  5D  02  80  00  72  90  00  53  00  18
00000030  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
...
000004F0  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
```
Node  | HWaddr | Long HWaddr | inet6 addr
--------|-----------|-----------------|-------------
node1 | 5C:22 | 15:11:6B:10:65:FC:5C:22 | fe80::1711:6b10:65f6:5d02
node2 | 5D:02 | 15:11:6B:10:65:F6:5D:02 | fe80::1711:6b10:65f6:5d02

### versions

Tested with the current master.
```
Operating System Environment
-----------------------------
       Operating System: "Ubuntu" "16.04.5 LTS (Xenial Xerus)"
                 Kernel: Linux 4.15.0-43-generic x86_64 x86_64

Installed compiler toolchains
-----------------------------
             native gcc: gcc (Ubuntu 5.4.0-6ubuntu1~16.04.10) 5.4.0 20160609
      arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
                avr-gcc: avr-gcc (GCC) 4.9.2
       mips-mti-elf-gcc: mips-mti-elf-gcc (Codescape GNU Tools 2016.05-03 for MIPS MTI Bare Metal) 4.9.2
             msp430-gcc: msp430-gcc (GCC) 4.6.3 20120301 (mspgcc LTS 20120406 unpatched)
   riscv-none-embed-gcc: riscv-none-embed-gcc (GNU MCU Eclipse RISC-V Embedded GCC, 64-bits) 7.2.0
   xtensa-esp32-elf-gcc: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
   xtensa-lx106-elf-gcc: xtensa-lx106-elf-gcc (crosstool-NG crosstool-ng-1.22.0-60-g37b07f6) 4.8.5
                  clang: clang version 3.8.0-2ubuntu4 (tags/RELEASE_380/final)

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "3.0.0"
    mips-mti-elf-newlib: "2.1.0"
riscv-none-embed-newlib: "2.5.0"
xtensa-esp32-elf-newlib: "2.2.0"
xtensa-lx106-elf-newlib: "2.0.0"
               avr-libc: "1.8.0svn" ("20111229")

Installed development tools
---------------------------
                  cmake: cmake version 3.10.0
               cppcheck: Cppcheck 1.72
                doxygen: 1.8.11
                 flake8: missing
                    git: git version 2.7.4
                   make: GNU Make 4.1
                openocd: missing
                 python: Python 2.7.12
                python2: Python 2.7.12
                python3: Python 3.5.2
             coccinelle: spatch version 1.0.4 with Python support and with PCRE support
```

gschorcht, 2019-01-03:
@miri64 I could reproduce the problem with only 5 simultaneous pings in each direction with MRF24J40.

gschorcht, 2019-01-04:
@miri64 Might this https://github.com/RIOT-OS/RIOT/blob/b77c802f9b68d9272da21386a4d0ccee4677ac59/sys/shell/commands/sc_icmpv6_echo.c#L248-L253 and https://github.com/RIOT-OS/RIOT/blob/b77c802f9b68d9272da21386a4d0ccee4677ac59/sys/shell/commands/sc_icmpv6_echo.c#L256-L262 be potential leaks?

I went through `pktbuf` code and if I didn't miss something the `gnrc_pktbuf_release(pkt)` releases `pkt` including all `pkt->next` snips, but `gnrc_pktbuf_add` doesn't release `next` snip(s) if the new buffer space couldn't be allocated. So I would guess that the packet allocated before and used as parameter `pkt` is not freed in the code above, if the new packet snip could not be allocated.

If I'm correct, the code probably looks like the following:
```diff
+       gnrc_pktsnip_t *next = pkt;
-       pkt = gnrc_ipv6_hdr_build(pkt, NULL, &addr);
+       pkt = gnrc_ipv6_hdr_build(next, NULL, &addr);

        if (pkt == NULL) {
+           gnrc_pktbuf_release(next);
            puts("error: packet buffer full");
            continue;
        }
...
+           next = pkt;
-           pkt = gnrc_pktbuf_add(pkt, NULL, sizeof(gnrc_netif_hdr_t), GNRC_NETTYPE_NETIF);
+           pkt = gnrc_pktbuf_add(next, NULL, sizeof(gnrc_netif_hdr_t), GNRC_NETTYPE_NETIF);

            if (pkt == NULL) {
+               gnrc_pktbuf_release(next);
                puts("error: packet buffer full");
                continue;
            }
```

miri64, 2019-01-04:
I wasn't really able to reproduce the `iotlab-m3` scenario in current master :confused:. @gschorcht have you tried your fix? It looks valid (I would choose the name `payload` though, to make the code less confusing). Also, please try with #9523 (we should still fix first before adopting #9523).

gschorcht, 2019-01-04:
> I wasn't really able to reproduce the `iotlab-m3` scenario in current master

Strange, I could reproduce it again and again. Will try it again too.

> @gschorcht have you tried your fix? 

Yes, but didn't help since in all case already the first allocation failed. https://github.com/RIOT-OS/RIOT/blob/b77c802f9b68d9272da21386a4d0ccee4677ac59/sys/shell/commands/sc_icmpv6_echo.c#L238

gschorcht, 2019-01-04:
> Are you using IoT-LAB for it? Maybe you can tell me which nodes you used then.

Yes, I was using: `m3-100.grenoble.iot-lab.info`, `m3-101.grenoble.iot-lab.info`, `m3-102.grenoble.iot-lab.info`

miri64, 2019-01-15:
Can recreate with `samr21-xpro` as well. `iotlab-m3` only works with reverting the regression introduced https://github.com/RIOT-OS/RIOT/pull/9171

miri64, 2019-01-15:
For now I applied the fix you proposed in https://github.com/RIOT-OS/RIOT/issues/10701#issuecomment-451326171 (#10772). I'm still hunting for the reception bug.

miri64, 2019-01-15:
This is really weird... I clearly see received ICMPv6 echo requests on the pinged node remain in the packet buffer, but they have some "holes" at the end which are always 96 bytes (and sometimes in the middle). Sometimes there are also 96 unset bytes at the end (i.e. remain 0x00 and not the ID of the request how it should be). I verified by using the following patch

```diff
diff --git a/sys/net/gnrc/pktbuf_static/gnrc_pktbuf_static.c b/sys/net/gnrc/pktbuf_static/gnrc_pktbuf_static.c
index eba560a..9d2c27e 100644
--- a/sys/net/gnrc/pktbuf_static/gnrc_pktbuf_static.c
+++ b/sys/net/gnrc/pktbuf_static/gnrc_pktbuf_static.c
@@ -461,6 +461,7 @@ static void _pktbuf_free(void *data, size_t size)
         prev = ptr;
         ptr = ptr->next;
     }
+    memset(data, 0x88, size);
     new->next = ptr;
     new->size = _align(size);
     /* calculate number of bytes between new _unused_t chunk and end of packet
```

to mark freed regions. This might hint to some race-condition within the packet buffer... :-/

gschorcht, 2019-01-15:
I also tried it again and again to figure a scheme. What I could observer when there are hanging packets is:
- There is always one chunk with size of 1280 bytes which seems to be contain a complete ICMPv6 packet.
- This ICMPv6 can either be an Echo request or an Echo reply.
- According to the IPv6 addresses, it seems always to be a received packet.
- The "holes" can happen at different positions in the chunk and with different sizes, see the hexdump below.
- The holes are either completely 0 or contain some garbage which always contain something like L2 addresses several times.
```
00000000  60  00  00  00  04  D8  3A  40  FE  80  00  00  00  00  00  00
00000010  21  23  23  23  23  A1  59  3E  FE  80  00  00  00  00  00  00
00000020  21  23  23  23  23  A3  1C  2A  81  00  B6  C5  00  53  00  19
00000030  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
...
000003C0  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
000003D0  00  00  00  00  10  13  00  00  10  00  00  00  00  00  00  00
000003E0  FF  FF  FF  FF  53  53  53  53  54  1B  FF  3F  10  00  00  00
000003F0  23  23  23  23  23  A1  59  3E  EC  1A  FF  3F  18  00  00  00
00000400  65  00  00  00  00  00  00  00  01  00  00  00  23  A1  59  3E
00000410  BC  1A  FF  3F  18  00  00  00  10  00  00  00  00  00  00  00
00000420  FF  FF  FF  FF  53  53  53  53  EC  1A  FF  3F  10  00  00  00
00000430  23  23  23  23  23  A1  59  3E  E4  1A  FF  3F  18  00  00  00
00000440  03  00  00  00  00  00  00  00  01  00  00  00  53  53  53  53
00000450  00  00  00  00  90  12  00  00  00  00  00  00  88  12  00  00
00000460  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
...
000004A0  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
000004B0  53  53  53  53  53  53  53  53  53  53  53  53  53  12  00  00
000004C0  00  00  00  00  20  12  00  00  53  53  53  53  53  53  53  53
000004D0  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53
000004E0  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53  53

```

miri64, 2019-01-15:
> * This ICMPv6 can either be an Echo request or an Echo reply.

Ok, I don't see echo replies (but for me the pinging nodes are always clean, but that might be just a timing issue). 

With the following diff (and the one from before)

```diff
diff --git a/sys/net/gnrc/network_layer/sixlowpan/frag/gnrc_sixlowpan_frag.c b/sys/net/gnrc/network_layer/sixlowpan/frag/gnrc_sixlowpan_frag.c
index 835b607..96ec9d7 100644
--- a/sys/net/gnrc/network_layer/sixlowpan/frag/gnrc_sixlowpan_frag.c
+++ b/sys/net/gnrc/network_layer/sixlowpan/frag/gnrc_sixlowpan_frag.c
@@ -363,6 +363,17 @@ void gnrc_sixlowpan_frag_rbuf_dispatch_when_complete(gnrc_sixlowpan_rbuf_t *rbuf
         new_netif_hdr->lqi = netif_hdr->lqi;
         new_netif_hdr->rssi = netif_hdr->rssi;
         LL_APPEND(rbuf->pkt, netif);
+        if (rbuf->pkt->size > 48) {
+            uint8_t *data = (uint8_t *)rbuf->pkt->data;
+            if (data[40] == 0x80) {
+                printf("id: %02x%02x, seq: %02x%02x complete\n",
+                       data[44], data[45],
+                       data[46], data[47]);
+                if (rbuf->pkt->size == 1280) {
+                    printf("data[1279] = %02x\n", data[1279]);
+                }
+            }
+        }
         gnrc_sixlowpan_dispatch_recv(rbuf->pkt, NULL, 0);
         gnrc_sixlowpan_frag_rbuf_remove(rbuf);
     }
```

I was able to confirm that the sticky package is indeed handed over broken by the reassembly buffer so somehow it registers as complete, while not actually being (or for some reason it wasn't copied properly...):

```
2019-01-15 18:57:52,904 - INFO #  id: 005a, seq: 0001 complete
2019-01-15 18:57:52,906 - INFO # data[1279] = 5a
2019-01-15 18:57:53,088 - INFO # id: 005a, seq: 0002 complete
2019-01-15 18:57:53,090 - INFO # data[1279] = 5a
2019-01-15 18:57:54,164 - INFO # id: 005a, seq: 0003 complete
2019-01-15 18:57:54,166 - INFO # data[1279] = 5a
2019-01-15 18:57:54,412 - INFO # id: 0059, seq: 0002 complete
2019-01-15 18:57:54,414 - INFO # data[1279] = 59
2019-01-15 18:57:55,246 - INFO # id: 005a, seq: 0004 complete
2019-01-15 18:57:55,247 - INFO # data[1279] = 5a
2019-01-15 18:57:55,501 - INFO # id: 0059, seq: 0003 complete
2019-01-15 18:57:55,503 - INFO # data[1279] = 59
2019-01-15 18:57:56,331 - INFO # id: 005a, seq: 0005 complete
2019-01-15 18:57:56,333 - INFO # data[1279] = 5a
2019-01-15 18:57:56,542 - INFO # id: 005a, seq: 0006 complete
2019-01-15 18:57:56,544 - INFO # data[1279] = 5a
2019-01-15 18:57:59,774 - INFO # id: 005a, seq: 0009 complete
2019-01-15 18:57:59,775 - INFO # data[1279] = 5a
2019-01-15 18:57:59,988 - INFO # id: 005a, seq: 000a complete
2019-01-15 18:57:59,990 - INFO # data[1279] = 88
2019-01-15 18:58:00,058 - INFO # id: 0059, seq: 0007 complete
2019-01-15 18:58:00,059 - INFO # data[1279] = 59
2019-01-15 18:58:01,140 - INFO # id: 0059, seq: 0008 complete
2019-01-15 18:58:01,141 - INFO # data[1279] = 59
2019-01-15 18:58:02,227 - INFO # id: 0059, seq: 0009 complete
2019-01-15 18:58:02,228 - INFO # data[1279] = 59
2019-01-15 18:58:03,306 - INFO # id: 0059, seq: 000a complete
2019-01-15 18:58:03,308 - INFO # data[1279] = 59
```

Note the 88 at `2019-01-15 18:57:59,990`

(I intentionally put the pinging nodes out of sync with the ID to be able to distinguish their messages).

The state of the packet buffer was:

[... 11 lines cut ...]
2019-01-15 18:58:17,627 - INFO # 00000010  FF  88  88  88  88  88  88  88  08  08  07  00  00  FF  D1  FF
2019-01-15 18:58:17,634 - INFO # 00000020  79  65  08  22  86  93  9D  5A  79  65  3B  4C  F3  BB  EE  A2
2019-01-15 18:58:17,639 - INFO # ~ unused: 0x20001b74 (next: 0x20002164, size:  240) ~
2019-01-15 18:58:17,644 - INFO # =========== chunk   2 (0x20001c64 size: 1280) ===========
2019-01-15 18:58:17,650 - INFO # 00000000  60  00  00  00  04  D8  3A  40  FE  80  00  00  00  00  00  00
2019-01-15 18:58:17,657 - INFO # 00000010  7B  65  08  22  86  93  9D  5A  FE  80  00  00  00  00  00  00
2019-01-15 18:58:17,663 - INFO # 00000020  7B  65  3B  4C  F3  BB  EE  A2  80  00  D3  97  00  5A  00  0A
2019-01-15 18:58:17,670 - INFO # 00000030  5A  5A  5A  5A  5A  5A  5A  5A  5A  5A  5A  5A  5A  5A  5A  5A
...
2019-01-15 18:58:18,089 - INFO # 00000430  5A  5A  5A  5A  5A  5A  5A  5A  5A  5A  5A  5A  5A  5A  5A  5A
2019-01-15 18:58:18,096 - INFO # 00000440  88  88  88  88  88  88  88  88  88  88  88  88  88  88  88  88
2019-01-15 18:58:18,102 - INFO # 00000450  88  88  88  88  88  88  88  88  88  88  88  88  88  88  88  88
2019-01-15 18:58:18,109 - INFO # 00000460  88  88  88  88  88  88  88  88  88  88  88  88  88  88  88  88
2019-01-15 18:58:18,115 - INFO # 00000470  88  88  88  88  88  88  88  88  88  88  88  88  88  88  88  88
2019-01-15 18:58:18,122 - INFO # 00000480  88  88  88  88  88  88  88  88  88  88  88  88  88  88  88  88
2019-01-15 18:58:18,128 - INFO # 00000490  88  88  88  88  88  88  88  88  6C  26  00  20  70  00  00  00
2019-01-15 18:58:18,135 - INFO # 000004A0  88  88  88  88  88  88  88  88  CC  26  00  20  60  00  00  00
2019-01-15 18:58:18,142 - INFO # 000004B0  88  88  88  88  88  88  88  88  88  88  88  88  88  88  88  88
2019-01-15 18:58:18,148 - INFO # 000004C0  88  88  88  88  88  88  88  88  88  88  88  88  88  88  88  88
2019-01-15 18:58:18,155 - INFO # 000004D0  88  88  88  88  88  88  88  88  88  88  88  88  88  88  88  88
2019-01-15 18:58:18,161 - INFO # 000004E0  88  88  88  88  88  88  88  88  88  88  88  88  88  88  88  88
2019-01-15 18:58:18,168 - INFO # 000004F0  88  88  88  88  88  88  88  88  88  88  88  88  88  88  88  88
```

Note that the id (position 0x02b in chunk 2) and seq (0x02e) are identical with the provided packet that was broken.

miri64, 2019-01-15:
:woman_facepalming: duplicate fragments are counted double... I'll prepare a PR.

miri64, 2019-01-15:
See #10776. If that fixes this issue I propose to add the testing procedures to our release specs so we notice regressions.

cgundogan, 2019-01-15:
@gschorcht this issue got closed automatically by #10776 getting merged. Could you nevertheless try to reproduce the former issue with current master (and hence the fix applied)?

closed 2019-01-15 by cgundogan

gschorcht, 2019-01-16:
> Why :worried: ried: ?

Used wrong smiley by mistake.

gschorcht, 2019-01-16:
Great job, I'm reall happy.

BTW, although the tests with MRF24J40 throw a lot of ping timeouts due to the generated network load, the packet buffer full error doesn't occur anymore.

miri64, 2019-01-16:
> @miri64 Thanks for figuring out the problem.

Always a very weird mixture of fun and disappointment to find this kind of well hidden, but simple to fix bugs :rofl:

miri64, 2019-01-16:
> See #10776. If that fixes this issue I propose to add the testing procedures to our release specs so we notice regressions.

See https://github.com/RIOT-OS/Release-Specs/pull/90

PR #10776 (2019-01-15): gnrc_sixlowpan_frag: drop received duplicates

### Contribution description
Otherwise duplicate fragments are added to the number of already
received datagram bytes in L153 leading to possible race-conditions.

### Testing procedure
Ping should still work for large packets. Also the expected results from #10701 should be the outcome.

### Issues/PRs references
Fixes #10701 (hopefully).