RIOT-OS/RIOT #10733: unsatisfied feature requirements: periph_gpio_irq periph_spi EXPECT ERRORS

Sunnyone1234, 2019-01-08:
#### Description

#### Steps to reproduce the issue

#### Expected results

#### Actual results

#### Versions

aabadie, 2019-01-08:
Note: I updated your initial comment it more readable. In Markdown, the syntax to highlight outputs is to put triple back quotes around them not `---------` (this is for titles).

Concerning your problem, it works for me with latest master.

What version of RIOT you using ? (you can paste the output of `git log -1`)

Sunnyone1234, 2019-01-08:
> Note: I updated your initial comment to make it more readable. In Markdown, the syntax to highlight outputs is to use put triple back quotes around them not `---------` (this is for titles).
> 
> Concerning your problem, it works for me with latest master.
> 
> What version of RIOT you using ? (you can paste the output of `git log -1`)

Thanks i have used 
-------
    commit 1d95da01fa9b37837b99a6e450c093a84a59f0e9 (HEAD -> master)
    Merge: ce5cd0591 672368954
    Author: Kevin "Bear Puncher" Weiss <weiss.kevin604@gmail.com>
    Date:   Fri Nov 2 15:31:26 2018 +0100

    Merge pull request #9211 from Josar/pr/xtimer_target_overflow
    
    xtimer: timer & target overflow, hang resolved.

Sunnyone1234, 2019-01-08:
> Works for me with `gcc version 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]` as well as in Docker. When building for native, `netdev_ieee802154` should not be a dependency and thus, `periph_spi` and `periph_gpio*` shouldn't. However, your build seems to incorporate these modules.
> 
>     * You sure there are no unstaged changes in your local repository?
> 
>     * Could you try in a new terminal to avoid any accidental environmental parameters?

Thanks Peter, i konw why. I habe added USEMODULE += at86rf233 in Makefile under gnrc_networking. So i can not make it without my board. so stupid. Thanks a lot~

Sunnyone1234, 2019-01-08:
> Thanks Peter, i konw why. I habe added USEMODULE += at86rf233 in Makefile under gnrc_networking. So i can not make it without my board. so stupid. Thanks a lot~

So should i close this issue or it will be closed by others. Actually i'm very fresh here by git...

PeterKietzmann, 2019-01-08:
@Sunnyone1234 no worries :-). And I'm closing the issue for you know.

closed 2019-01-08 by PeterKietzmann