RIOT-OS/RIOT #10749: Document PGP key for security@riot-os.org

nmeum, 2019-01-10:
If someone were to report a security critical issue in RIOT, e.g. #10739, using responsible disclosure via mail to `security@riot-os.org` as described in https://github.com/RIOT-OS/RIOT/blob/master/CONTRIBUTING.md#bug-reports They would probably like to make sure that no one, except the intended recipient, is able to read such a bug report.  For that purpose, it would be nice if you could document a PGP key for this email address to allow sending encrypted emails to it.

miri64, 2019-01-11:
Since security@riot-os.org is a mailing list that PGP encryption is not easy to achieve. There would be some shared private key required for which some secure sharing mechanism needs to be in place which we don't have (and I don't believe exist; except for running around with USB sticks, which because of the distributed nature of the RIOT community is impractical). Furthermore, if a maintainer does - for whatever personal reasons they have for it - chooses to not use PGP, they might be unable to read a security that affects their area of expertise.

As far as I'm aware of not even security@kernel.org (thanks @kaspar030 for pointing this out) is allowing encrypted access. Do you have an example for a security mailing list of a free software project that offers that?

Yes these arguments are less from a "security must be unbreakable"-perspective weak and more from perspective of practicality working in a loosely organized community, but I guess this loosely goes into the direction of "security vs. freedom" ;-). All in all, yes, having an unencrypted security mailing list still allows a potential attacker to read about security bugs in RIOT (or Linux ;-)), but it's still less disclosed then shouting it in a public forum like the issue tracker.

miri64, 2019-01-11:
(this was a mixture of the current state of discussion on security@riot-os.org and my own thoughts on the matter).

Teufelchen1, 2023-06-14:
Wasn't that fixed in  #17189?

miri64, 2023-06-14:
Aye

closed 2023-06-14 by miri64