RIOT-OS/RIOT #10761: iotlab-m3: poor ping6 performance and a possibly broken 6lo fragmentation?

cgundogan, 2019-01-13:
#### Description
I was pinging between two iotlab-m3 nodes on the iotlab testbed using the grenoble site, when I noticed a poor ping performance and a possible state corruption of the 6lo fragmentation. I am not sure whether this is due to other experiments/interferences on the testbed. I certainly tried other channels with the same result. Tomorrow I can try to reproduce locally with two iotlab-m3 boards and use the debugger to find out more.

#### Steps to reproduce the issue
2 iotlab-m3 nodes (A and B) with `gnrc_networking`.
A pings B with an increasing payload size.

#### Expected results
ping should work with nearly 100% success rate

#### Actual results
The run below contains the following steps:
1. Node-A pings Node-B with payload 100 (works)
2. Node-A pings Node-B with payload 200 (100% timeout)
3. Node-A pings Node-B with payload 100 (100% timeout)
4. Reboot Node-A
5. Node-A pings Node-B with payload 100 (100% timeout)
6. Reboot Node-B
7. Node-A pings Node-B with payload 100 (works again)

There are no hardfaults during the run.

Side note: The Version string shows `Version: 2018.10-RC1-963-g35868`, but I am definitely on master.

```
main(): This is RIOT! (Version: 2018.10-RC1-963-g35868)
RIOT network stack example application
All up, running the shell now

> ping6 5 fe80::1711:6b10:65fc:5c22 100 100
ping6 5 fe80::1711:6b10:65fc:5c22 100 100
ping timeout
108 bytes from fe80::1711:6b10:65fc:5c22: id=83 seq=2 hop limit=64 time = 21.438 ms
108 bytes from fe80::1711:6b10:65fc:5c22: id=83 seq=3 hop limit=64 time = 27.580 ms
108 bytes from fe80::1711:6b10:65fc:5c22: id=83 seq=4 hop limit=64 time = 25.025 ms
108 bytes from fe80::1711:6b10:65fc:5c22: id=83 seq=5 hop limit=64 time = 22.078 ms
--- fe80::1711:6b10:65fc:5c22 ping statistics ---
5 packets transmitted, 4 received, 20% packet loss, time 1.06505139 s
rtt min/avg/max = 21.438/24.030/27.580 ms

> ping6 5 fe80::1711:6b10:65fc:5c22 200 100
ping6 5 fe80::1711:6b10:65fc:5c22 200 100
ping timeout
ping timeout
ping timeout
ping timeout
ping timeout
--- fe80::1711:6b10:65fc:5c22 ping statistics ---
5 packets transmitted, 0 received, 100% packet loss

> ping6 5 fe80::1711:6b10:65fc:5c22 100 100
ping6 5 fe80::1711:6b10:65fc:5c22 100 100
ping timeout
ping timeout
ping timeout
ping timeout
ping timeout
--- fe80::1711:6b10:65fc:5c22 ping statistics ---
5 packets transmitted, 0 received, 100% packet loss

> reboot
reboot
main(): This is RIOT! (Version: 2018.10-RC1-963-g35868)
RIOT network stack example application
All up, running the shell now

> ping6 5 fe80::1711:6b10:65fc:5c22 100 100
ping6 5 fe80::1711:6b10:65fc:5c22 100 100
ping timeout
ping timeout
ping timeout
ping timeout
ping timeout
--- fe80::1711:6b10:65fc:5c22 ping statistics ---
5 packets transmitted, 0 received, 100% packet loss

HERE NODE-B REBOOTS

> ping6 5 fe80::1711:6b10:65fc:5c22 100 100
ping6 5 fe80::1711:6b10:65fc:5c22 100 100
108 bytes from fe80::1711:6b10:65fc:5c22: id=84 seq=1 hop limit=64 time = 23.674 ms
108 bytes from fe80::1711:6b10:65fc:5c22: id=84 seq=2 hop limit=64 time = 22.715 ms
108 bytes from fe80::1711:6b10:65fc:5c22: id=84 seq=3 hop limit=64 time = 27.893 ms
108 bytes from fe80::1711:6b10:65fc:5c22: id=84 seq=4 hop limit=64 time = 24.510 ms
108 bytes from fe80::1711:6b10:65fc:5c22: id=84 seq=5 hop limit=64 time = 22.716 ms
--- fe80::1711:6b10:65fc:5c22 ping statistics ---
5 packets transmitted, 5 received, 0% packet loss, time 0.06531306 s
rtt min/avg/max = 22.715/24.301/27.893 ms
```

#### Versions
RIOT master

miri64, 2019-01-13:
Mhhhh maybe this is related to https://github.com/RIOT-OS/RIOT/issues/10701? I was planning to have a look into that next week.

miri64, 2019-01-13:
Also please try with #9523

miri64, 2019-01-15:
Can reproduce on my desk.

miri64, 2019-01-15:
Interestingly only on `iotlab-m3`. With `samr21-xpro` it works.

miri64, 2019-01-15:
Or @vincent-d

miri64, 2019-01-15:
The corresponding PR to the commit is https://github.com/RIOT-OS/RIOT/pull/9171

aabadie, 2019-01-15:
This is serious and will require more investigation for the DMA use with SPI on stm32f1. It is possible to disable this feature just for SPI for the moment. It's not a real fix but I think it will help.
@vincent-d has a better knownledge than me on STM32.

aabadie, 2019-01-15:
@miri64, can you try #10775 and tell if it improves the situation ?

vincent-d, 2019-01-15:
Unfortunately, I don't have any stm32f1 at hand to be able to test this...so I guess the best is to disable the feature on stm32f1 for now.

closed 2019-01-15 by miri64

cgundogan, 2019-01-15:
FYI, I can also confirm that this issue is solved with #10775.

PR #10775 (2019-01-15): boards/iotlab-m3: disable DMA with SPI

### Contribution description

This PR is an attempt to improve the situation with #10761. The idea is just to disable the use of DMA on the SPI interface used by the radio on iotlab-m3.

Even if it needs further investigation with the DMA implementation with stm32f1, for the moment this should make networking usable again.

Not tested.

### Testing procedure

See #10761 

### Issues/PRs references

fixes #10761, fixes regression introduced in #9171