RIOT-OS/RIOT #10787: cpu/esp32: modules crypto and hashes cannot be used together with module esp_wifi

gschorcht, 2019-01-16:
#### Description

[`wpa_supplicant`](https://w1.fi/wpa_supplicant) required by module `esp_wifi` includes its own `crypto` library. Therefore, using RIOT modules `sys/crypto` or `sys/hashes` together with module `esp_wifi` leads to multiple definitions of some functions. On the other hand, RIOT modules `sys/crypto` and `sys/hashes` cannot be used to replace the `crypto` libary of `wpa_supplicant` for the following reasons:

- a number of functions used by `wpa_supplicant` are missing
- function signatures expected by `wpa_supplicant` are slightly different

A first incomplete list of missing functions are
```
aes_wrap
aes_encrypt_init
aes_encrypt_deinit
aes_decrypt_init
aes_decrypt_deinit
hmac_md5_vector
hmac_sha256_vector
md5_vector
sha1_prf = sha1_prf,
sha1_vector = sha1_vector,
sha256_prf
```

miri64, 2019-08-10:
Sorry for the late reply (and first issue I did not know about that I found due to stale bot being active), but I faced a "similar" issue when porting `emb6`, since it has many of the "core utils" named the same (I guess due to our common history with Contiki). In my case I just renamed those versions with [a patch](https://github.com/RIOT-OS/RIOT/blob/b48afd6375df35fe19296a60cc9e6bda34da8923/pkg/emb6/patches/0002-Rename-colliding-files-and-functions.patch). Would that be a possible route for you or do the SDKs prevent that?

gschorcht, 2019-08-10:
It could be possible. Binary libraries of the SDK don't link directly against the symbols of the `crypto` library. Rather they define a data structure of pointers to required `crypto` functions:
https://github.com/RIOT-OS/RIOT/blob/b48afd6375df35fe19296a60cc9e6bda34da8923/cpu/esp32/vendor/esp-idf/esp32/fast_crypto_ops.c#L136-L145

Probably, it is enough to rename all `crypto` functions of `wpa_supplicant` and to change the names when they are referenced in this data structure. I will try.

PR #11994 (2019-11-22): cpu/esp32: esp_wifi and crypto module fix
### Contribution description

This PR fixes the problem that the `esp_wifi` module cannot be used together with RIOT's `crypto` and `hashes` module. This problem was described in #10787.

**Background**

The `esp_wifi` module requires a number of crypto functions to realize WPA/WPA2. For that purpose it uses the crypt functions from the `wpa-supplicant` which is part of the ESP32 SDK. Unfortunately, these crypto functions can't be replaced by the crypto functions of modules `crypto` and `hashes` since

- a number of crypto functions are missing and
- t [... cut]
closed 2019-11-22 by aabadie