RIOT-OS/RIOT #10809: openthread: does not build on current Arch

miri64, 2019-01-17:
#### Description
When trying to build `examples/openthread` for #10599 I wasn't able to do so on my current Arch-based laptop

#### Steps to reproduce the issue
Have a setup Arch distribution and try to compile `examples/openthread`

#### Expected results
The application will build successfully.

#### Actual results
The build errors

```
meshcop/dataset.cpp: In member function 'otError ot::MeshCoP::Dataset::AppendMleDatasetTlv(ot::Message&) const':
meshcop/dataset.cpp:487:56: error: 'void* memcpy(void*, const void*, size_t)' copying an object of non-trivial type 'class ot::MeshCoP::DelayTimerTlv' from an array of 'const class ot::MeshCoP::Tlv' [-Werror=class-memaccess]
             memcpy(&delayTimer, cur, sizeof(delayTimer));
                                                        ^
In file included from meshcop/dataset.hpp:40,
                 from meshcop/dataset.cpp:37:
./meshcop/meshcop_tlvs.hpp:1141:7: note: 'class ot::MeshCoP::DelayTimerTlv' declared here
 class DelayTimerTlv: public Tlv
       ^~~~~~~~~~~~~
  CXX      meshcop/libopenthread_mtd_a-dtls.o
  CXX      meshcop/libopenthread_mtd_a-energy_scan_client.o
cc1plus: all warnings being treated as errors
make[5]: *** [Makefile:3909: meshcop/libopenthread_mtd_a-dataset.o] Fehler 1
make[5]: *** Es wird auf noch nicht beendete Prozesse gewartet....
meshcop/dataset_manager.cpp: In member function 'void ot::MeshCoP::DatasetManager::SendGetResponse(const ot::Coap::Header&, const ot::Ip6::MessageInfo&, uint8_t*, uint8_t) const':
meshcop/dataset_manager.cpp:907:78: error: type qualifiers ignored on cast result type [-Werror=ignored-qualifiers]
             if ((tlv = mNetwork.Get(static_cast<const Tlv::Type>(aTlvs[index]))) != NULL)
                                                                              ^
cc1plus: all warnings being treated as errors
```

#### Versions
Quite recent Arch while writing this issue (I might not have updated for 1 or 2 days or so ;-))

```

Operating System Environment
-----------------------------
       Operating System: "Arch Linux" 
                 Kernel: Linux 4.20.1-arch1-1-ARCH x86_64 unknown

Installed compiler toolchains
-----------------------------
             native gcc: gcc (GCC) 8.2.1 20181127
      arm-none-eabi-gcc: arm-none-eabi-gcc (Arch Repository) 8.2.0
                avr-gcc: avr-gcc (GCC) 8.2.0
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
   xtensa-esp32-elf-gcc: missing
   xtensa-lx106-elf-gcc: missing
                  clang: clang version 7.0.1 (tags/RELEASE_701/final)

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "3.0.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
xtensa-esp32-elf-newlib: missing
xtensa-lx106-elf-newlib: missing
               avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                  cmake: cmake version 3.13.2
               cppcheck: missing
                doxygen: 1.8.15
                 flake8: 3.6.0 (mccabe: 0.6.1, pycodestyle: 2.4.0, pyflakes: 2.0.0) CPython 3.7.2 on Linux
                    git: git version 2.20.1
                   make: GNU Make 4.2.1
                openocd: missing
                 python: Python 3.7.2
                python2: Python 2.7.15
                python3: Python 3.7.2
             coccinelle: missing
```

jcarrano, 2019-02-08:
This is caused by new warnings in GCC 8.1. This upstream PR may have fixed it: https://github.com/openthread/openthread/pull/2944

aabadie, 2020-07-02:
Is this still the case ? Using arm gcc 9.2.1 it builds just fine.

miri64, 2020-07-02:
I did not get that error, but now I get

```
In file included from ./coap/coap.hpp:37,
                 from ./common/instance.hpp:50,
                 from api/channel_monitor_api.cpp:37:
./common/debug.hpp:83: error: "assert" redefined [-Werror]
   83 | #define assert(cond)  \
      | 
In file included from /usr/arm-none-eabi/include/sys/reent.h:503,
                 from /usr/arm-none-eabi/include/stdlib.h:18,
                 from /usr/arm-none-eabi/include/c++/10.1.0/cstdlib:75,
                 from /usr/arm-none-eabi/include/c++/10.1.0/stdlib.h:36,
                 from ../../include/openthread/instance.h:38,
                 from ../../include/openthread/channel_monitor.h:38,
                 from api/channel_monitor_api.cpp:35:
/home/mlenders/Repositories/RIOT-OS/RIOT/core/include/assert.h:107: note: this is the location of the previous definition
  107 | #define assert(cond) ((cond) ? (void)0 : core_panic(PANIC_ASSERT_FAIL, \
      | 
```

Current setup:
```
Operating System Environment
----------------------------
         Operating System: "Arch Linux" 
                   Kernel: Linux 5.7.5-arch1-1 x86_64 unknown
             System shell: GNU bash, version 5.0.17(1)-release (x86_64-pc-linux-gnu)
             make's shell: GNU bash, version 5.0.17(1)-release (x86_64-pc-linux-gnu)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (GCC) 10.1.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (Arch Repository) 10.1.0
                  avr-gcc: missing
         mips-mti-elf-gcc: missing
               msp430-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
   xtensa-esp8266-elf-gcc: missing
                    clang: missing

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.3.0"
      mips-mti-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: "2.2.0"
xtensa-esp8266-elf-newlib: missing
                 avr-libc: missing (missing)

Installed development tools
---------------------------
                   ccache: missing
                    cmake: cmake version 3.17.3
                 cppcheck: missing
                  doxygen: 1.8.18
                      git: git version 2.27.0
                     make: GNU Make 4.3
                  openocd: Open On-Chip Debugger 0.10.0+dev-01157-gd6541a811-dirty (2020-04-04-00:05)
                   python: Python 3.8.3
                  python2: Python 2.7.18
                  python3: Python 3.8.3
                   flake8: error: /usr/bin/python3: No module named flake8
               coccinelle: missing
```

The upstream bug might have been fixed with 4f83dfd975ed893929a204e570c6196642d84190

goldroom, 2020-11-19:
I got the same building error as mentioned in the latest comment by @miri64 .

I tried compiling examples/openthread on:
- Up2date Debian 10 (Buster)
- Up2date Manjaro (which is based on Arch Linux)
- Up2date Ubuntu 18.04.5 LTS (bionic)

jia200x, 2020-11-19:
I'm just updating my arch. I will see how it goes

jia200x, 2020-11-19:
I cannot reproduce the issue.
@goldroom which RIOT commit are you using?

goldroom, 2020-11-19:
@jia200x sorry, I updated my comment shortly after posting:

- It doesn't compile when using gcc-arm-none-eabi-9-2020-q2-update => gcc version 9.3.1 20200408 (release) (GNU Arm Embedded Toolchain 9-2020-q2-update)
- It **does** compile when using gcc-arm-none-eabi-7-2018-q2-update => gcc version 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907] (GNU Tools for Arm Embedded Processors 7-2018-q2-update)

I'm using the latest master commit, i.e. 6b573b9000a14ce51f43daf46592bacb77adeb20.

jia200x, 2020-11-19:
I suspect this is caused by mixing OpenThread and RIOT includes when building the OpenThread pkg. I will see if I can find a workaround

jia200x, 2020-11-19:
@goldroom does it stop building during the compile phase of the pkg? (`./configure` + `make`)
Does it get to build RIOT files?

goldroom, 2020-11-19:
@jia200x 
- "does it stop building during the compile phase of the pkg?" => yes
- "Does it get to build RIOT files?" => no

```
[...]
  AR       libmbedcrypto.a
ar: `u' modifier ignored since `D' is the default (see `U')
 /usr/bin/mkdir -p '.../myRIOT/examples/openthread/bin/samr21-xpro/pkg/openthread/output//lib'
 /usr/bin/install -c -m 644  libmbedcrypto.a '.../myRIOT/examples/openthread/bin/samr21-xpro/pkg/openthread/output//lib'
 ( cd '.../myRIOT/examples/openthread/bin/samr21-xpro/pkg/openthread/output//lib' && arm-none-eabi-ranlib libmbedcrypto.a )
make[6]: Nothing to be done for 'install-data-am'.
make[6]: Nothing to be done for 'install-exec-am'.
make[6]: Nothing to be done for 'install-data-am'.
Making install in src
Making install in core
  CXX      api/libopenthread_ftd_a-border_router_api.o
  CXX      api/libopenthread_ftd_a-channel_manager_api.o
In file included from ./coap/coap.hpp:37,
                 from ./common/instance.hpp:50,
                 from api/channel_manager_api.cpp:37:
./common/debug.hpp:83: error: "assert" redefined [-Werror]
   83 | #define assert(cond)  \
      | 
In file included from .../gcc-arm-none-eabi-9-2020-q2-update/arm-none-eabi/include/sys/reent.h:503,
                 from .../gcc-arm-none-eabi-9-2020-q2-update/arm-none-eabi/include/stdlib.h:18,
                 from .../gcc-arm-none-eabi-9-2020-q2-update/arm-none-eabi/include/c++/9.3.1/cstdlib:75,
                 from .../gcc-arm-none-eabi-9-2020-q2-update/arm-none-eabi/include/c++/9.3.1/stdlib.h:36,
                 from ../../include/openthread/instance.h:38,
                 from ../../include/openthread/channel_manager.h:38,
                 from api/channel_manager_api.cpp:35:
.../myRIOT/core/include/assert.h:107: note: this is the location of the previous definition
  107 | #define assert(cond) ((cond) ? (void)0 : core_panic(PANIC_ASSERT_FAIL, \
      | 
cc1plus: all warnings being treated as errors
make[5]: *** [Makefile:2391: api/libopenthread_ftd_a-channel_manager_api.o] Error 1
make[4]: *** [Makefile:486: install-recursive] Error 1
make[3]: *** [Makefile:649: install-recursive] Error 1
make[2]: *** [Makefile:952: install] Error 2
make[1]: *** [Makefile:51: .../myRIOT/examples/openthread/bin/samr21-xpro/pkg/openthread/output/lib/libopenthread-ftd.a] Error 2
make: *** [.../myRIOT/examples/openthread/../../Makefile.include:660: pkg-build-openthread] Error 2
```

jia200x, 2020-11-19:
@goldroom could you try the following patch in your RIOT repo? https://gist.github.com/jia200x/485b0f7264aa13b328b7a504b9f452de

goldroom, 2020-11-19:
@jia200x this patch fixed it. Now it's also building successfully with `gcc version 9.3.1 20200408 (release) (GNU Arm Embedded Toolchain 9-2020-q2-update)`. Thanks!

jia200x, 2020-11-19:
great!
That patch it's not the final solution, but would mitigate the effect until we bump to a newer OpenThread version. I will open a PR with that.

maribu, 2022-09-19:
Builds fine on Alpine. Arch typically is on par with Alpine on the toolchain versions, so I think this should also work on Arch. Please re-open, if this is not correct.

closed 2022-09-19 by maribu