RIOT-OS/RIOT #10814: UART driver doesn't support more advanced configuration?

bluephoton, 2019-01-18:
#### Description
I'm implementing a protocol that requires setting word length, parity, stop bits and pin inversion. uart.h/.c doesn't seem to allow this.

### Useful links
According the STM32 reference manual, configuring these bits must be done before enabling the device. Unfortunately uart_init() enable the device before it exits.

One way is to provide uart_init_ex(..., bool enable) which will not enable by default in addition to a separate enable function. Then add a uart_configure() that allow more configurations. This will not be breaking change this way.

dylad, 2019-01-18:
@bluephoton The UART interface is pretty limited in the current state.
Maybe #10743 will addressed your concerns.

bluephoton, 2019-01-18:
@dylad this is exactly what I'm looking for. Still I can't see support for pin inversion there, and also 9 bits is required. I don't see how the uart_mode() will work while device is enabled - but I didn't inspect the code carefully. I'd be happy to test this when it is released. But will not work for my scenario without pin inversion and 9 bit.

yegorich, 2019-01-18:
@bluephoton could you elaborate more on how you are going to use 9 bit mode? Protocol description. What physical serial driver (RS232, RS422 etc.) are you using?

PeterKietzmann, 2019-01-18:
@MrKevinWeiss, @yegorich just that you're aware of this.

bluephoton, 2019-01-18:
When I enable 9 bit mode and enable parity control, one bit is used for parity and I get 8 bit data.

### Protocol
I'm implementing SBUS protocol which uses inverted serial with the following:
frame: 1 (Frame Start) + 22 (data) + 1 (flag) + 1 (Frame End) = 25 bytes
1 byte: 1 (start bit) + 8 (data bits) + 1 (parity) + 2 (stop bits) = 12 bits

### Software
My code will receive stream of bytes and build a full frame, then send it to another thread for processing. This is working as expected after using configurations below.

### Hardware
Radio transmitter ~> Radio Receiver -> STM32F7 with RIOT
Radio Receiver is connected to UART2 RX pin and configured using the following code

``` c
MODIFY_REG(__UART_DEV(uart)->CR1, USART_CR1_M_Msk, USART_CR1_M0);           // 9-bits word
MODIFY_REG(__UART_DEV(uart)->CR1, USART_CR1_PCE_Msk, USART_CR1_PCE);        // Enable parity control
MODIFY_REG(__UART_DEV(uart)->CR1, USART_CR1_PS_Msk, 0);                     // Even Parity
MODIFY_REG(__UART_DEV(uart)->CR2, USART_CR2_STOP_Msk, USART_CR2_STOP_1);    // 2 stop bits
MODIFY_REG(__UART_DEV(uart)->CR2, USART_CR2_RXINV_Msk, USART_CR2_RXINV);    // invert RX
```

yegorich, 2019-01-18:
@bluephoton 8E2 should do exactly what you want:

- 1 start bit
- 8 data bits
- 1 parity bit
- 2 stop bits

When you configure 8 bit and parity `uart_mode` switches STM32 UART into 9 bit.

The only inversion must be handled in a custom way.

bluephoton, 2019-01-19:
@yegorich Fantastic! one more parameter (bit flags) can serve to set the inversion mode. Or you can also pass a single complex type param with all the info so you can update later without changing function signature.

How does the enable bit fit in here, what will be the expected call sequence?

yegorich, 2019-01-19:
Pin inversion can be implemented the same way as hardware flow control is made for STM32. Take a look at the `uart_conf_t` structure. RTS/CTS pinmux and initialization are controlled via `MODULE_STM32_PERIPH_UART_HW_FC` macro. So another macro like `MODULE_STM32_PERIPH_UART_PIN_INVERSION ` as also `uint8_t  rx_inv, tx_inv` can be introduced to enable this functionality.

Also, take a look at @haukepetersen [comment ](https://github.com/RIOT-OS/RIOT/pull/5899#issuecomment-451864097) about enabling hw flow control in the `uart_mode()` routine.

```
/**
 * @brief   Structure for UART configuration data
 */
typedef struct {
    USART_TypeDef *dev;     /**< UART device base register address */
    uint32_t rcc_mask;      /**< bit in clock enable register */
    gpio_t rx_pin;          /**< RX pin */
    gpio_t tx_pin;          /**< TX pin */
#ifndef CPU_FAM_STM32F1
    gpio_af_t rx_af;        /**< alternate function for RX pin */
    gpio_af_t tx_af;        /**< alternate function for TX pin */
#endif
    uint8_t bus;            /**< APB bus */
    uint8_t irqn;           /**< IRQ channel */
#ifdef MODULE_STM32_PERIPH_UART_HW_FC
    gpio_t cts_pin;         /**< CTS pin - set to GPIO_UNDEF when not using HW flow control */
    gpio_t rts_pin;         /**< RTS pin */
#ifndef CPU_FAM_STM32F1
    gpio_af_t cts_af;       /**< alternate function for CTS pin */
    gpio_af_t rts_af;       /**< alternate function for RTS pin */
#endif
#endif
#if defined(CPU_FAM_STM32L0) || defined(CPU_FAM_STM32L4)
    uart_type_t type;       /**< hardware module type (USART or LPUART) */
    uint32_t clk_src;       /**< clock source used for UART */
#endif
#ifdef MODULE_PERIPH_DMA
    dma_t dma;              /**< Logical DMA stream used for TX */
    uint8_t dma_chan;       /**< DMA channel used for TX */
#endif
} uart_conf_t;

```

bluephoton, 2019-01-21:
Yes thanks a lot guys for your help! Closing...

closed 2019-01-21 by bluephoton