RIOT-OS/RIOT #10815: doc: Tracking doxygen 1.8.15 problems

gschorcht, 2019-01-18:
#### Description

Using the current doxygen release 1.8.15 to generate the documentation results into **12700 warnings**!

What are the problems:

1. Setex-style headers in markdown files and comments throw the warning: `multiple use of section label 'xyz' while adding section, (first occurrence: <file>, line <n>).`. It seems to be a bug: https://github.com/doxygen/doxygen/issues/6779.
(20 warnings)

2. doxygen seems to be much more nit-picking. Constructs like the following throw warnings: `Member XYZ (<type>) of <file | group> is not documented.`:
    ```c
    /**
     * @name    UART configuration
     * @{
     */
    #define UART_NUMOF          (1U)
    #define UART_PIN_RX         GPIO_PIN(0,8)
    #define UART_PIN_TX         GPIO_PIN(0,6)
    /** @} */
   ```
   (8960 warnings in files)
   (3720 warnings in groups)

   `<type>` can be function, macro definition, enumeration, typedef

#### Steps to reproduce the issue

Install/download doxygen 1.8.15 or the current master and execute
```
make doc
```

jcarrano, 2019-02-07:
Regarding item (2), it is OK to ignore the "undocumented entity" warnings? In a perfect world, everything is documented, but in reality it is not and I don't think it will ever be even close.

(3) Is there a reason why we have an empty project name?

gschorcht, 2019-02-07:
> (2)  it is OK to ignore the "undocumented entity" warnings?

Our static tests include `make doc` which probably will fail with doxygen 1.8.15 due to these warnings.

> (3) Is there a reason why we have an empty project name?

I guess it was by intention. The behavior of earlier version of doxygen was to handle `mainpage.md` as root in the navigation tree if `PROJECT_NAME` is empty. For example, if you set `PROJECT_NAME = RIOT-OS` you would get a navigation tree as follows

[image]

instead of

[image]

muhammadadeelinfo, 2020-01-08:
Hi, I am just new at this platform.
How about this idea? The project name as "Documentation" by changing it from riot.doxyfile and "Overview" covers the sections: 
RIOT in a nutshell
Contribute to RIOT
The quickest start
Structure
Further information
If you like this idea then I can pull a request with some further modifications.
@gschorcht

[image]

kfessel, 2022-01-25:
maybe this issue should be updated to Doxygen 1.9.1