RIOT-OS/RIOT #10827: boards/nrf51dk: multiple external interrupts not working

thibv, 2019-01-20:
#### Description
On nRF51DK platform, cannot enable multiple external interrupts. Only the latest will work.

#### Steps to reproduce the issue

Use the hello-world program, and add the following lines to enable callback functions cbButton*.
`  gpio_init_int(BTN0_PIN, GPIO_IN_PU, GPIO_FALLING, &cbButton0, NULL);  `
`  gpio_init_int(BTN1_PIN, GPIO_IN_PU, GPIO_FALLING, &cbButton1, NULL);  `
`  gpio_init_int(BTN2_PIN, GPIO_IN_PU, GPIO_FALLING, &cbButton2, NULL);  `
`  gpio_init_int(BTN3_PIN, GPIO_IN_PU, GPIO_FALLING, &cbButton3, NULL);  `

#### Expected results
When one of the four on-board button is pressed, the callback should be called.

#### Actual results
Only the latest callback is working. Here, only cbButton3 is working. If I invert the two last lines (ie cbButton2 and cbButton3), only cbButton2 will work.

#### Versions

Operating System Environment :
       Operating System: "Ubuntu" "18.04.1 LTS (Bionic Beaver)"
                 Kernel: Linux 4.15.0-43-generic x86_64 x86_64

Installed compiler toolchains :
             native gcc: gcc (Ubuntu 7.3.0-27ubuntu1~18.04) 7.3.0
      arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2018-q4-major) 8.2.1 20181213 (release) [gcc-8-branch revision 267074]
                avr-gcc: avr-gcc (GCC) 5.4.0
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
                  clang: missing

Installed compiler libs :
   arm-none-eabi-newlib: "3.0.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
               avr-libc: "2.0.0" ("20150208")

Installed development tools :
                  cmake: cmake version 3.10.2
               cppcheck: missing
                doxygen: 1.8.13
                 flake8: missing
                    git: git version 2.17.1
                   make: GNU Make 4.1
                openocd: Open On-Chip Debugger 0.10.0
                 python: Python 2.7.15rc1
                python2: Python 2.7.15rc1
                python3: Python 3.6.7
             coccinelle: missing

aabadie, 2019-01-20:
I can also reproduce the issue on nrf52dk. Looking at the gpio periph driver implementation, which is the same for all nrf, there's no array for managing multiple callbacks so the behaviour makes sense.
For me it's a bug the implementation, I'll have a look.

aabadie, 2019-01-20:
@thibv, thanks for reporting the issue.
I opened #10828 to fix this issue. Do you confirm that it fixes your problem on nrf51dk ?

closed 2019-01-22 by leandrolanzieri