RIOT-OS/RIOT #10857: frdm-kw41z: error with `SLOT_LEN` handling and flasher not working

cladmi, 2019-01-24:
#### Description

While trying to use the `frdm-kw41z` I noticed an error message when compiling and it cannot be flashed. I am using `openocd 0.10` and also does not work with `USE_OLD_OPENOCD`.

#### Steps to reproduce the issue

Clean shows an error (Maybe `clean` could be used to detect unwanted messages as it should be completely silent).

```
BOARD=frdm-kw41z make --no-print-directory -C examples/default/ clean
/bin/sh: 1: Syntax error: Missing '))'
```

When flashing with `openocd v0.10` from ubuntu `bionic` I get

```
### Flashing Target ###
/home/harter/work/git/RIOT/examples/default/bin/frdm-kw41z/default.elf is fine.
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : add flash_bank kinetis klx.flash
adapter speed: 1000 kHz
none separate
cortex_m reset_config sysresetreq
srst_only separate srst_nogate srst_open_drain connect_deassert_srst
Info : No device selected, using first device.
Info : J-Link OpenSDA 2 compiled May  6 2016 11:04:17
Info : Hardware version: 1.00
Info : VTarget = 3.300 V
Info : clock speed 1000 kHz
Info : SWD DPIDR 0x0bc11477
Error: MDM: failed to read ID register
Info : SWD DPIDR 0x0bc11477
Info : klx.cpu: hardware has 2 breakpoints, 2 watchpoints
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* klx.cpu            cortex_m   little klx.cpu            running
Info : MDM: Chip is unsecured. Continuing.
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000790 msp: 0x1fff8200
auto erase enabled
Error: Unsupported K-series
Error: MCU is unsupported, SDID 0x41591004
Error: auto_probe failed
```

According to the documentation, version `0.10` should be enough. I noticed an option with `USE_OLD_OPENOCD` but it also fails because there is no configuration file for this one:

```
/home/harter/work/git/RIOT/dist/tools/openocd/openocd.sh flash /home/harter/work/git/RIOT/examples/default/bin/frdm-kw41z/default.elf
### Flashing Target ###
Error: Unable to locate OpenOCD configuration file
       (/home/harter/work/git/RIOT/boards/common/frdm/dist/old-openocd-klx.cfg)
/home/harter/work/git/RIOT/examples/default/../../Makefile.include:534: recipe for target 'flash-only' failed
make: *** [flash-only] Error 1
```

#### Expected results

A silent `make clean` and being able to flash.

#### Versions

Ubuntu 18.04 bionic.

```
openocd --version
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
```
```
apt-cache show openocd
Package: openocd
Architecture: amd64
Version: 0.10.0-4
Priority: extra
Section: universe/embedded
Origin: Ubuntu
```

cladmi, 2019-01-28:
Yep, it may only be a documentation issue.

Also having a runtime error checking openocd supports it could be nice. When working on flashers and all I could put it in common in `openocd.inc.mk`.

aabadie, 2019-07-07:
@cladmi, I think the SLOT_LEN handling is fixed now (the board supports riotboot in master). Maybe you could just update the documentation to close this issue ?

cladmi, 2019-07-08:
Indeed, I forgot to update the issues, I will do a cleanup during the release.

closed 2019-10-21 by aabadie