RIOT-OS/RIOT #10875: gnrc_ipv6: crash on heavy network load on native

gschorcht, 2019-01-26:
#### Description

Bombarding `native` with pings of maximum size and an interval of 0 from multiple hosts leads to crash. The following is the backtrace from `gdb`
```
Program received signal SIGSEGV, Segmentation fault.
0x5656cf83 in gnrc_netif_hdr_get_netif (hdr=0x1158) at /home/gs/src/RIOT-Xtensa-ESP.working/sys/include/net/gnrc/netif/hdr.h:291
291	    return gnrc_netif_get_by_pid(hdr->if_pid);
(gdb) bt
#0  0x5656cf83 in gnrc_netif_hdr_get_netif (hdr=0x1158) at /home/gs/src/RIOT-Xtensa-ESP.working/sys/include/net/gnrc/netif/hdr.h:291
#1  0x5656dbc1 in _send (pkt=0x5659d348 <_pktbuf+1704>, prep_hdr=true) at /home/gs/src/RIOT-Xtensa-ESP.working/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c:539
#2  0x5656d385 in _event_loop (args=0x0) at /home/gs/src/RIOT-Xtensa-ESP.working/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c:193
#3  0xf7e0dbdb in makecontext () from /lib/i386-linux-gnu/libc.so.6
#4  0x00000000 in ?? ()
```
#### Steps to reproduce the issue

Compile `examples/gnrc_networking` with `-g` option:
```
gs@gunny8:~/src/RIOT-Xtensa-ESP.working$ CFLAGS="-g3" PORT=tap0 USEMODULE=gnrc_pktbuf_cmd make -C examples/gnrc_networking BOARD=native
```
Start `gdb`:
```
gdb examples/gnrc_networking/bin/native/gnrc_networking.elf
```
Run the RIOT instance in gdb:
```
run tap0
```
Ping from four terms:
```
term1> sudo ping6 fe80::280d:21ff:fed1:c5ed -Itap0 -s1392 -i 0
term2> sudo ping6 fe80::280d:21ff:fed1:c5ed -Itap0 -s1392 -i 0
term3> sudo ping6 fe80::280d:21ff:fed1:c5ed -Itap0 -s1392 -i 0
term4> sudo ping6 fe80::280d:21ff:fed1:c5ed -Itap0 -s1392 -i 0
```
After a while, RIOT instance should crash.

kaspar030, 2019-01-26:
The backtrace looks exactly a #6123, thus I'm closing this as duplicate. re-open if you disgree.

gschorcht, 2019-01-26:
Hm, the crash happens on different calls in the `_send` function. Sure, iIt might have the same cause, an inconsistent memory, but maybe not. The crash as described in this issue crashes reproducable always at the same call.

IMHO it would be reasonable to let @miri64 have a short look before we close it.

miri64, 2019-01-26:
> Hm, the crash happens on different calls in the _send function. Sure, iIt might have the same cause, an inconsistent memory, but maybe not. The crash as described in this issue crashes reproducable always at the same call.

The send function changed significantly since 2015, so I'm not sure that it might be the same GDB dump after all.

miri64, 2019-01-26:
So I think the version of master @kaspar030 reported on in #6123 was 8432d923bec56b0f64e76f38353be46bea8c5d0d. I determined this by running

```sh
git log --merges --before="2016-11-15 17:55"
```

l684 in #6123 seems to me to be the first access to a pointer in the provided `pkt` list

https://github.com/RIOT-OS/RIOT/blob/8432d923bec56b0f64e76f38353be46bea8c5d0d/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c#L657-L684

same goes for l539 in current master 6cd81dbc87c6032a209870707c310903d6cb6c94

https://github.com/RIOT-OS/RIOT/blob/6cd81dbc87c6032a209870707c310903d6cb6c94/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c#L525-L539

I'd say its inconclusive if it is the same error, but in both cases the packet seems to get corrupted while being in `gnrc_ipv6`'s message queue (possibly due to a too early release). All in all it seems to be at least in the same class of issue #6123 and the way to reproduce is also the same, so I'd say we close this one as a duplicate, as @kaspar030 proposed. Any fix should be tested with the steps to reproduce anyway, The testing procedures are better outlined here though, so I will link this issue as a reference in #6123.

miri64, 2019-01-26:
> Start `gdb`:
> 
> ```
> gdb examples/gnrc_networking/bin/native/gnrc_networking.elf
> ```
> Run the RIOT instance in gdb:
> 
> ```
> run tap0
> ```

We have `make debug` for that ;-).

gschorcht, 2019-01-26:
> All in all it seems to be at least in the same class of issue #6123 and the way to reproduce is also the same, so I'd say we close this one as a duplicate, as @kaspar030 proposed.

Thanks. Agreed.

miri64, 2019-01-26:
@gschorcht Why `-s1392` btw?

gschorcht, 2019-01-26:
I didn't try whether it also happens with data sizes less than the maximum size. I just used the same command as for my stress tests of esp8266 `esp_wifi` driver. 

BTW, I still have a packet buffer problem there, issue 4 in https://github.com/RIOT-OS/RIOT/issues/10861. I ran into the problem described here when I was trying whether I can reproduce it on `native`.

miri64, 2019-01-26:
> Probably also because I thought that the crash might be related to the buffer full problem and requires maximum data size to reproduce it.

Since both WiFi and Ethernet have an MTU 1500 that would be `s1452` though ;-).

gschorcht, 2019-01-26:
Yes, but if the router provides the IPv6 MTU option in its RA as mine does, the MTU is downsized to 1440 as in my case :wink: Exactly this question came up also in PR #10792 and PR #10581. The interface starts with MTU 1500 but once the first RA is received and the interface gets its routing prefix, the MTU is also downsized. This happens for Linux boxes in the same way.

miri64, 2019-01-26:
Ok sorry, I forgot about that. On native however, the MTU stays 1500.

miri64, 2019-01-26:
> BTW, I still have a packet buffer problem there, issue 4 in #10861. I ran into the problem described here when I was trying whether I can reproduce it on `native`.

Where you able to?

gschorcht, 2019-01-26:
> Where you able to?

No, I just saw the crash described here. In `esp_wifi` the buffer becomes full and communication stops to work, but it doesn't crash.

gschorcht, 2019-01-26:
> Ok sorry, I forgot about that. On native however, the MTU stays 1500.

Ol, I see. According to description in #6123, the data size seems not matter.

miri64, 2019-01-26:
> Ok, I see. According to the description in #6123, the data size does not seem to matter.

True