RIOT-OS/RIOT #10878: nrfmin can get stuck and never reach RX (while TX works)

pystub, 2019-01-26:
#### Description

nrfmin radio driver can become stuck in a state where it can no longer receive packets over the radio.

This issue triggers sporadically and can happen within seconds or hours of operation. Increased number of nodes concurrently trying to send seems to increase the chance of it.

My original research showed when driver is in this state, there is a packet received, but it appears to be never passed on and cleared, so it prevents the driver from switching periphery into receive mode. I.e. second condition fails:

https://github.com/RIOT-OS/RIOT/blob/3e6336ce89d64d58ab07764ef7f65fc86800cb85/cpu/nrf5x_common/radio/nrfmin/nrfmin.c#L160

For more info please see attempted fixes at the end.

#### Steps to reproduce the issue

1. Use nrfmin, 6LoWPAN, and GNRC in a program that periodically (broad|multi)casts.
2. Leave two or more nodes on.

#### Expected results

Packets should be received until conditions become unsatisfactory for chip operation or radio connection.

#### Actual results

Packets can be sent over the radio, but receiving does not work.

#### Versions

(a few week old version of RIOT; will be filled in later)

My particular NRF51 chip **is not affected by** PAN 20 (which means `STATE` register can be used in my case)

#### Attempted fixes

I attempted to fix this by making certain code run in the interrupt handler (assuming that's the most correct place in effort to avoid blowing something else's stack):

```c
void isr_radio(void)
{
    if (NRF_RADIO->EVENTS_END == 1) {
        NRF_RADIO->EVENTS_END = 0;
        /* did we just send or receive something? */
        if (state == STATE_RX || rx_buf.pkt.hdr.len > 0) {
            /* drop packet on invalid CRC */
            if ((NRF_RADIO->CRCSTATUS != 1) || !(nrfmin_dev.event_callback)) {
                rx_buf.pkt.hdr.len = 0;
                NRF_RADIO->TASKS_START = 1;
            } else {
                rx_lock = 0;
                nrfmin_dev.event_callback(&nrfmin_dev, NETDEV_EVENT_ISR);
            }
        }
        if (state == STATE_TX) {
            goto_target_state();
        }
    }

    cortexm_isr_end();
}
```

This greatly improved reliability, yet now sending function would sometimes spinlock waiting for `state` to change from `STATE_TX`:

https://github.com/RIOT-OS/RIOT/blob/3e6336ce89d64d58ab07764ef7f65fc86800cb85/cpu/nrf5x_common/radio/nrfmin/nrfmin.c#L335

After a few attempts I managed to avoid that by adding code that waits for `EVENT_READY` after triggering `TASK_TXEN` and `TASK_RXEN`:

https://github.com/RIOT-OS/RIOT/blob/3e6336ce89d64d58ab07764ef7f65fc86800cb85/cpu/nrf5x_common/radio/nrfmin/nrfmin.c#L166

https://github.com/RIOT-OS/RIOT/blob/3e6336ce89d64d58ab07764ef7f65fc86800cb85/cpu/nrf5x_common/radio/nrfmin/nrfmin.c#L357

Changed to:

```c
        NRF_RADIO->EVENTS_READY = 0;
        NRF_RADIO->TASKS_RXEN = 1;
        while (NRF_RADIO->EVENTS_READY == 0) {}
```

With these changes, after a lot of time, system sometimes hard faults. I was unable to recover the details because UART disconnected prior to that.

#### Other observations

`cortexm_isr_end()` is docummented as to be called at the end of every interrupt routine, but in this driver, in `isr_radio`, line https://github.com/RIOT-OS/RIOT/blob/3e6336ce89d64d58ab07764ef7f65fc86800cb85/cpu/nrf5x_common/radio/nrfmin/nrfmin.c#L315 can return early and not fufill that specification. Should this be corrected?

Also, when `NRF_RADIO->POWER` is set to `0` (off), it appears to reset some of the registers. When waking periphery up, `nrfmin_init` should probably be called again.

pystub, 2019-01-28:
Spinlock [line 335](https://github.com/RIOT-OS/RIOT/blob/3e6336ce89d64d58ab07764ef7f65fc86800cb85/cpu/nrf5x_common/radio/nrfmin/nrfmin.c#L335) can still happen with mentioned changes.

PeterKietzmann, 2019-01-30:
Maybe something for @SemjonKerner?

SemjonKerner, 2019-04-15:
We could confirm that there is a bug in this ISR. Your approach to remove the return is correct. Obviously the ISR must be terminated with `cortexm_isr_end();`. I will write your code into a fix right away.

Can you please specify the additional problems that arose from the fix  `NRF_RADIO->EVENTS_READY = 0;`? On first glance this seems good to me and is done in other nrf radio drivers as well. I don't see how this might result in a hard fault. Maybe you can reproduce the steps and give more detail?

Also you are most likely right about reinitializing the driver after powering down. Some registers will be cleared and must be initialized again. This is probably not thoroughly tested and I will try to give a fix afterwards.

pystub, 2019-04-16:
Looks great. I saw that miri64 did some testing. Should I try rerunning my case as well and get back with results?

The spin-wait for `EVENTS_READY` is to have a stronger guarantee that `state` agrees with the state that radio peripheral is in (e.g. if by some miracle ISR returns and something manages to send before radio is up). I'll admit that the remark about hard fault was poorly worded and I wanted to say that this particular issue seems sufficiently resolved.

I've faced reinitialization problem while trying to reduce power usage. For now setting network state to `NETOPT_STATE_SLEEP` provides worthwhile savings.

miri64, 2019-04-17:
> Looks great. I saw that miri64 did some testing. Should I try rerunning my case as well and get back with results?

Would be great so we can confirm that your bug is fixed.

SemjonKerner, 2019-04-17:
@pystub, I tried to find the source of your hard fault and suspect the assert in `nrfmin_send`.
https://github.com/RIOT-OS/RIOT/blob/3e6336ce89d64d58ab07764ef7f65fc86800cb85/cpu/nrf5x_common/radio/nrfmin/nrfmin.c#L332
 This might have happened when sending is triggered even though the device is off.
If you can reproduce your findings, please try printing the assertions with `CFLAGS += -DDEBUG_ASSERT_VERBOSE`.
http://doc.riot-os.org/assert_8h.html#a2200149ba880bf26fed140bdcf318113

pystub, 2019-04-18:
I ran some tests with 4 boards sending announcements at 0.1s intervals. I tried to incrementally add changes to in the nrfmin.c file. With the now merged interrupt bugfix applied, **`state` still can be eventually prevented from going into `STATE_RX`**. Line where this happens: https://github.com/RIOT-OS/RIOT/blob/3e6336ce89d64d58ab07764ef7f65fc86800cb85/cpu/nrf5x_common/radio/nrfmin/nrfmin.c#L160
Using gdb I was able to verify that `rx_buf.pkt.hdr.len` is non-zero.

Also, while comparing files of my current working directory I found some other previously unmentioned changes. I'm attaching the full patch file.

[nrfmin-changes.txt](https://github.com/RIOT-OS/RIOT/files/3096028/nrfmin-changes.txt)

```diff
diff --git a/cpu/nrf5x_common/radio/nrfmin/nrfmin.c b/cpu/nrf5x_common/radio/nrfmin/nrfmin.c
index 02d713208..02208dda9 100644
--- a/cpu/nrf5x_common/radio/nrfmin/nrfmin.c
+++ b/cpu/nrf5x_common/radio/nrfmin/nrfmin.c
@@ -163,7 +163,9 @@ static void goto_target_state(void)
         NRF_RADIO->PACKETPTR = (uint32_t)(&rx_buf);
         NRF_RADIO->BASE0 = (CONF_ADDR_BASE | my_addr);
         /* goto RX mode */
+        NRF_RADIO->EVENTS_READY = 0;
         NRF_RADIO->TASKS_RXEN = 1;
+        while (NRF_RADIO->EVENTS_READY == 0) {}
         state = STATE_RX;
     }
 
@@ -269,6 +271,12 @@ int nrfmin_set_state(netopt_state_t val)
     /* make sure radio is turned on and no transmission is in progress */
     NRF_RADIO->POWER = 1;
 
+    DEBUG(
+        "[nrfmin] state: from %d to %d\n"
+        "         mem state %d\n"
+        "         reg STATE %lu\n"
+        "         pkt.hdr.len %u\n", target_state, val, state, NRF_RADIO->STATE, rx_buf.pkt.hdr.len);
+
     switch (val) {
         case NETOPT_STATE_OFF:
             target_state = STATE_OFF;
@@ -297,17 +305,20 @@ void isr_radio(void)
     if (NRF_RADIO->EVENTS_END == 1) {
         NRF_RADIO->EVENTS_END = 0;
         /* did we just send or receive something? */
-        if (state == STATE_RX) {
+        /* HACK: if we have a packet, trigger a callback anyway */
+        state_t orig_state = state;
+        if (orig_state == STATE_RX || rx_buf.pkt.hdr.len > 0) {
             /* drop packet on invalid CRC */
             if ((NRF_RADIO->CRCSTATUS != 1) || !(nrfmin_dev.event_callback)) {
                 rx_buf.pkt.hdr.len = 0;
                 NRF_RADIO->TASKS_START = 1;
-                return;
+                // return;
+            } else {
+                rx_lock = 0;
+                nrfmin_dev.event_callback(&nrfmin_dev, NETDEV_EVENT_ISR);
             }
-            rx_lock = 0;
-            nrfmin_dev.event_callback(&nrfmin_dev, NETDEV_EVENT_ISR);
         }
-        else if (state == STATE_TX) {
+        if (orig_state == STATE_TX) {
             goto_target_state();
         }
     }
@@ -319,10 +330,11 @@ static int nrfmin_send(netdev_t *dev, const iolist_t *iolist)
 {
     (void)dev;
 
-    assert((iolist) && (state != STATE_OFF));
+    assert((iolist) && (((NRF_RADIO->POWER & RADIO_POWER_POWER_Msk) >> RADIO_POWER_POWER_Pos) == RADIO_POWER_POWER_Enabled));
 
     /* wait for any ongoing transmission to finish and go into idle state */
     while (state == STATE_TX) {}
+    // while (NRF_RADIO->STATE != RADIO_STATE_STATE_Disabled && NRF_RADIO->STATE != RADIO_STATE_STATE_Rx) {}
     go_idle();
 
     /* copy packet data into the transmit buffer */
@@ -344,7 +356,9 @@ static int nrfmin_send(netdev_t *dev, const iolist_t *iolist)
     /* trigger the actual transmission */
     DEBUG("[nrfmin] send: putting %i byte into the ether\n", (int)hdr->len);
     state = STATE_TX;
+    NRF_RADIO->EVENTS_READY = 0;
     NRF_RADIO->TASKS_TXEN = 1;
+    while (NRF_RADIO->EVENTS_READY == 0) {}
 
     return (int)pos;
 }
@@ -354,7 +368,7 @@ static int nrfmin_recv(netdev_t *dev, void *buf, size_t len, void *info)
     (void)dev;
     (void)info;
 
-    assert(state != STATE_OFF);
+    assert(((NRF_RADIO->POWER & RADIO_POWER_POWER_Msk) >> RADIO_POWER_POWER_Pos) == RADIO_POWER_POWER_Enabled);
 
     unsigned pktlen = rx_buf.pkt.hdr.len;
 
```

From my testing I'm suspecting that the key to having functional sending is zeroing out `rx_buf.pkt.hdr.len`.

SemjonKerner, 2019-04-18:
So the Interrupt bugfix was not necessarily to fix the states of the driver, but to fix context switch from interrupt to process context.

There seems to be some impreciseness in the implementation of the state machine, but until now I did not find the exact bug(s). Your information is very helpful. I'll have a deeper look asap.

closed 2019-05-15 by SemjonKerner

PeterKietzmann, 2019-08-05:
@pystub from your last comment I did not understand  if  #11395  and  #11507 completely fixed your issues. I think that I just ran into a very similar problem so eventually the issue still exists. Furthermore, do you have your test code lying around somewhere?

pystub, 2019-08-05:
Here's the [test code](https://gist.github.com/pystub/1bbe287f081c9a8d7020e11ad9b43b73) I used.

Right now in production I use my diversion, but I don't recall why I made that choice. I'm still very suspicious of the `status` variable.

PeterKietzmann, 2019-08-06:
@pystub thanks for sharing! Just out of interest, which production is it that you're talking about?

pystub, 2019-08-06:
@PeterKietzmann It's one of those smart monitoring devices, nothing breathtaking. I sometimes check out `master` and flash it on test boards, but for mostly-final hardware I use that old commit (hardware is... *fun* to fix on its own).